wmbusmeters/wmbusmeters #177: Upgrade to 1.0.1 and backward incompatibilities

dwrobel, 2020-10-31:
I'm trying to upgrade wmbusters to 1.0.1 from wmbusmeters-0.9.36-1.fc32.x86_64 and I have a few listed below questions.

Here is the existing configuration _(working for 0.9.36)_:
```
$ cat /etc/wmbusmeters.conf 
loglevel=debug
device=auto
logtelegrams=true
format=json

$ cat /etc/wmbusmeters.d/sensus 
name=sensus
type=iperl
id=00111979
key=NOKEY
shell=/usr/bin/mosquitto_pub -h piwnica -t stat/sensus/dw -m "$METER_JSON"
```

An attempt to run wmbusmeters version 1.0.1 with [fix](https://github.com/weetmuts/wmbusmeters/pull/176) doesn't work:
```
# wmbusmeters --debug
You must supply at least one device (eg auto:c1) to receive wmbus telegrams.
```

Adding `:c1` to existing /etc/wmbusmeters.conf file:
```
$ cat /etc/wmbusmeters.conf 
loglevel=debug
device=auto:c1
logtelegrams=true
format=json
```
also didn't help.
**Q1:** Is it no longer supported to specify `device` in the `/etc/wmbusmeters.conf` file?
If possible, I'd like to ship some kind of `auto` configuration by default in Fedora package.

Official documentation says:
>You can add --device to override the settings in the config. Like this:
>wmbusmeters --useconfig=/home/me/.config/wmbusmeters --device=rtlwmbus

```
# wmbusmeters --debug --device=rtlwmbus
Unknown option "--device=rtlwmbus"
```
**Q2**: Am I specifying something wrong or is it inconsistency with the documentation.

**Q3**: How to avoid accessing any serial ports when using e.g. `rtlwmbus` device type:
```
# gdb -q --args wmbusmeters --debug rtlwmbus
Reading symbols from /home/dw/projects/wmbusmeters/wmbusmeters-1.0.1/build/wmbusmeters...
(gdb) r
Starting program: wmbusmeters --debug rtlwmbus
Missing separate debuginfos, use: dnf debuginfo-install glibc-2.31-4.fc32.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
(wmbusmeters) version: 1.0.1
(config) using device: rtlwmbus 
(config) number of meters: 0
[New Thread 0x7ffff7a13700 (LWP 37200)]
[New Thread 0x7ffff7212700 (LWP 37201)]
(main) device /dev/ttyS0 not currently used, detect contents...
(serial) before config: /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyS0) sent "FF0A02008077"
(serial /dev/ttyS0) sent "FF0A02008077"
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyS0 (detect amb8465)
(amb8465) are you there? no.
(serial) before config: /dev/ttyS0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (detect rc1180)
(serial /dev/ttyS0) sent "00"
(serial) received binary ""
(serial /dev/ttyS0) sent "30"
(serial) received binary ""
(rc1180) config: id=00000000 mfct=0000 (@@@) media=00 version=00 rssi_mode=00 uart_bps=00 uart_flow_ctrl=00 data_interface=00 radio_channel=00 radio_power=00 radio_data_rate=00 preamble_length=00 mbus_mode=00
(serial /dev/ttyS0) sent "58"
(serialtty) closed /dev/ttyS0 (detect rc1180)
(rc1180) are you there? yes 00000000
(main) using globally set link modes c1,t1
Started auto rc1180[00000000] on /dev/ttyS0 listening on c1,t1
(rc1180) on /dev/ttyS0
(serial) before config: /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS0 fd 3 (rc1180)
(rc1180) get config to get device id
(serial /dev/ttyS0) sent "00"
(serial) received binary ""
Not the expected dongle (dongle said ERR, you said 00000000!
(wmbus) deleted RC1180
[New Thread 0x7ffff6a11700 (LWP 37202)]
[Thread 0x7ffff6a11700 (LWP 37202) exited]
[New Thread 0x7ffff6a11700 (LWP 37203)]
(main) rtlsdr device 00000001 not currently used.
[Thread 0x7ffff6a11700 (LWP 37203) exited]
[New Thread 0x7ffff6a11700 (LWP 37204)]
[Thread 0x7ffff6a11700 (LWP 37204) exited]
[New Thread 0x7ffff6a11700 (LWP 37205)]
(main) found specified device (rtlwmbus) that matches detected device (:RTLWMBUS[00000001]:0/0)
(main) using globally set link modes c1,t1
Started auto rtlwmbus[00000001] listening on c1,t1
[Thread 0x7ffff6a11700 (LWP 37205) exited]
[New Thread 0x7ffff6a11700 (LWP 37206)]
[Thread 0x7ffff6a11700 (LWP 37206) exited]
[New Thread 0x7ffff6a11700 (LWP 37207)]
(rtlwmbus) using command: rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
[Thread 0x7ffff6a11700 (LWP 37207) exited]
[Detaching after fork from child process 37208]
(serialcmd) opened /bin/sh pid 37208 fd 4 (rtlwmbus)
(main) regular reset of RTLWMBUS  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for RTLWMBUS
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
```

I was hopping this will not access the `/dev/ttyS0` and send some data over this interface:
```
(serial /dev/ttyS0) sent "FF0A02008077"
```
as it might affect the other device connected to this interface.

I would appreciate if you could comment on those aforementioned questions (Q1, Q2 and Q3).

weetmuts, 2020-11-01:
>>Q1: Is it no longer supported to specify device in the /etc/wmbusmeters.conf file?

Setting device=rtlwmbus in the conf should work! Please test again.
And try device=auto:c1,t1 as well.

I have done some other fixes.

>>Q2: Am I specifying something wrong or is it inconsistency with the documentation.
>>Official documentation says:
>>    You can add --device to override the settings in the config. Like this:
>>    wmbusmeters --useconfig=/home/me/.config/wmbusmeters --device=rtlwmbus

I have improved the README. The --device was initially added for the multiple udev
rules, where the udev rule decided which device to use. Now with auto detection the
original reason for the --device= is gone, however there is still a value.
Assume that you could have a running daemon with device=rtlwmbus and several configured 
meters. Now you want to use the config but with a different device:
wmbusmeters --useconfig=/ --device=amb8465:c1,t1
So you want to use --device= then you have to have --useconfig= 

>>Q3: How to avoid accessing any serial ports when using e.g. rtlwmbus device type:

The current version pushed to github should avoid proving serial ports if you specify:
wmbusmeters rtlwmbus
or if you specify a tty:
wmbusmeters /dev/ttyUSB0:im871a:c1
or if you add:
wmbusmeters --donotprobe=all auto:c1,t1
or more specific
wmbusmeters --donotprobe=/dev/ttyUSB0 auto:c1,t1

wmbusmeters will probe serial ports if you type:
wmbusmeters auto:c1
wmbusmeters im871a:c1

auto:c1 obviously should probe and im871a:c1 must probe since
it does not know which tty has the im871a.
 
I am still working on the logic for detection and setting of linkmodes/frequencies.
It was all so easy in the beginning, one device one meter. :-)

Expect some changes next week.

dwrobel, 2020-11-02:
> > > Q1: Is it no longer supported to specify device in the /etc/wmbusmeters.conf file?
> 
> Setting device=rtlwmbus in the conf should work! Please test again.
> And try device=auto:c1,t1 as well.

Indeed, using 2c19a838c493474ad381a7cf7e0edf20ab47adff it started to work having:
```
$ cat /etc/wmbusmeters.conf 
loglevel=debug
device=auto:c1,t1
logtelegrams=true
format=json
```
and launching via:
```
sudo ./wmbusmetersd  /run/wmbusmeters/wmbusmeters-rtlwmbus.pid
```

However, if I only replace `device=auto:c1,t1` with `device=rtlwmbus` in the `/etc/wmbusmeters.conf` then it no longer works:
```
wmbusmetersd[240980]: (wmbusmeters) started /run/wmbusmeters/wmbusmeters-rtlwmbus.pid
wmbusmetersd[240981]: (config) "device" "rtlwmbus"
wmbusmetersd[240981]: (config) "logtelegrams" "true"
wmbusmetersd[240981]: (config) "format" "json"
wmbusmetersd[240981]: (config) "" ""
wmbusmetersd[240981]: (config) loading meter file /etc/wmbusmeters.d/sensus
wmbusmetersd[240981]: (config) name=sensus
wmbusmetersd[240981]: (config) type=iperl
wmbusmetersd[240981]: (config) id=00111979
wmbusmetersd[240981]: (config) key=<notprinted>
wmbusmetersd[240981]: (config) shell=/usr/bin/mosquitto_pub -h piwnica -t stat/sensus/dw -m "$METER_JSON"
wmbusmetersd[240981]: (wmbusmeters) version: 1.0.1
wmbusmetersd[240981]: (config) using device: rtlwmbus:any
wmbusmetersd[240981]: (config) number of meters: 1
sudo[240978]: pam_unix(sudo:session): session closed for user root
wmbusmetersd[240981]: (main) configured "sensus" "iperl" "00111979" not-encrypted
wmbusmetersd[240981]: (serial) expecting devices to work
wmbusmetersd[240981]: (lookup) with file "rtlwmbus:any"
wmbusmetersd[240981]: (lookup) driver: raw file
wmbusmetersd[240981]: Started config rawtty on rtlwmbus listening on any
wmbusmetersd[240981]: (serial) override with devicefile: rtlwmbus
wmbusmetersd[240981]: (rawtty) on rtlwmbus
wmbusmetersd[240981]: (serialfile) reading from file rtlwmbus (override rtlwmbus)
wmbusmetersd[240981]: (main) regular reset of rawtty  on rtlwmbus will happen every 82800 seconds
wmbusmetersd[240981]: (wmbus) no alarm (expected activity) for rawtty
wmbusmetersd[240981]: (serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
wmbusmetersd[240981]: (wmbusmeters) waiting for telegrams
wmbusmetersd[240981]: (serial) waiting for stop
wmbusmetersd[240981]: (serial) EVENT thread interrupted
wmbusmetersd[240981]: (serial) no more data on fd=4
wmbusmetersd[240981]: (serial) received binary "3233363932350A"
wmbusmetersd[240981]: (serialfile) closed rtlwmbus -1 (override rtlwmbus)
wmbusmetersd[240981]: (wmbus) checkWMBUSFrame
                       "3233363932350A"
wmbusmetersd[240981]: (wmbus) less than 11 bytes, partial frame
wmbusmetersd[240981]: (serial) no devices working emergency exit!
wmbusmetersd[240981]: (serial) stopping manager
wmbusmetersd[240981]: (serial) event loop stopped!
wmbusmetersd[240981]: (serial) MAIN thread interrupted
wmbusmetersd[240981]: (serial) TIMER thread interrupted
wmbusmetersd[240981]: (wmbusmeters) shutting down
wmbusmetersd[240981]: (wmbus) deleted rawtty
wmbusmetersd[240981]: (serial) no devices working emergency exit!
```

weetmuts, 2020-11-12:
All bugs should be fixed! Fingers crossed.
Please test!

dwrobel, 2020-11-16:
I tested the tagged `1.0.3` version with the following configuration:
```
$ cat /usr/lib/systemd/system/wmbusmeters@.service 
[Unit]
Description="wmbusmeters service on %I device"
Documentation=https://github.com/weetmuts/wmbusmeters
Documentation=man:wmbusmeters(1)
After=network.target

[Service]
Type=forking
ExecStartPre=/bin/mkdir -p /run/wmbusmeters
ExecStart=/usr/sbin/wmbusmetersd --device='%I' /run/wmbusmeters/wmbusmeters-%i.pid
PIDFile=/run/wmbusmeters/wmbusmeters-%i.pid

[Install]
WantedBy=multi-user.target
```

```
[dw@t630 etc]$ cat /etc/wmbusmeters.conf 
loglevel=debug
#device=rtlwmbus
device=auto:c1,t1
logtelegrams=true
format=json
```

```
[dw@t630 etc]$ cat /etc/wmbusmeters.d/sensus 
name=sensus
type=iperl
id=00111979
key=NOKEY
shell=/usr/bin/mosquitto_pub -h piwnica -t stat/sensus/dw -m "$METER_JSON"
```

Running with `auto:c1,t1` mode seems to work fine:
```
[dw@t630 etc]$ sudo systemctl start wmbusmeters@auto:c1\\x2ct1.service
[dw@t630 etc]$ sudo systemctl status wmbusmeters@auto:c1\\x2ct1.service
● wmbusmeters@auto:c1\x2ct1.service - "wmbusmeters service on auto:c1,t1 device"
     Loaded: loaded (/usr/lib/systemd/system/wmbusmeters@.service; disabled; vendor preset: disabled)
     Active: active (running) since Mon 2020-11-16 22:46:56 CET; 8s ago
       Docs: https://github.com/weetmuts/wmbusmeters
             man:wmbusmeters(1)
    Process: 5353 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 5354 ExecStart=/usr/sbin/wmbusmetersd --device=auto:c1,t1 /run/wmbusmeters/wmbusmeters-auto:c1\x2ct1.pid (code=exited, status=0/SUCCESS)
   Main PID: 5355 (wmbusmetersd)
      Tasks: 7 (limit: 17888)
     Memory: 1.8M
        CPU: 1.423s
     CGroup: /system.slice/system-wmbusmeters.slice/wmbusmeters@auto:c1\x2ct1.service
             ├─5355 /usr/sbin/wmbusmetersd --device=auto:c1,t1 /run/wmbusmeters/wmbusmeters-auto:c1\x2ct1.pid
             ├─5364 /bin/sh -c /usr/bin/rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
             ├─5365 /usr/bin/rtl_sdr -d 0 -f 868.95M -s 1.6e6 -
             └─5366 /usr/bin/rtl_wmbus

Nov 16 22:46:59 t630 wmbusmetersd[5355]: (wmbusmeters) waiting for telegrams
Nov 16 22:46:59 t630 wmbusmetersd[5355]: (serial) waiting for stop
Nov 16 22:47:01 t630 wmbusmetersd[5355]: (memory) rss 4349952 peak 4.14 MiB
Nov 16 22:47:03 t630 wmbusmetersd[5355]: (serial) received ascii "T1;1;1;2020-11-16 21:47:03.000;16;16;80037825;0x2e4401062578038001167ab20020051252526d5c26de243612194496d30efaff7c85eb58c16f72713ad5f1a0a9605f<0A>"
Nov 16 22:47:03 t630 wmbusmetersd[5355]: (rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-11-16 21:47:03.000;16;16;80037825;0x2e4401062578038001167ab20020051252526d5c26de243612194496d30efaff7c85eb58c16f72713ad5f1a0a9605f<0A>"
Nov 16 22:47:03 t630 wmbusmetersd[5355]: (rtlwmbus) received full frame
Nov 16 22:47:03 t630 wmbusmetersd[5355]: (wmbus) parseDLL @0 47
Nov 16 22:47:03 t630 wmbusmetersd[5355]: (meter) sensus: for me? 80037825
Nov 16 22:47:03 t630 wmbusmetersd[5355]: (meter) sensus: not for me: not my id
Nov 16 22:47:03 t630 wmbusmetersd[5355]: (wmbus) telegram from 80037825 ignored by all configured meters!
[dw@t630 etc]$ sudo systemctl stop wmbusmeters@auto:c1\\x2ct1.service
```

However, an attempt to run using `rtlwmbus` device fails as following:
```
[dw@t630 etc]$ sudo systemctl start wmbusmeters@rtlwmbus.service
Job for wmbusmeters@rtlwmbus.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status wmbusmeters@rtlwmbus.service" and "journalctl -xe" for details.
[dw@t630 etc]$ sudo systemctl status wmbusmeters@rtlwmbus.service
● wmbusmeters@rtlwmbus.service - "wmbusmeters service on rtlwmbus device"
     Loaded: loaded (/usr/lib/systemd/system/wmbusmeters@.service; disabled; vendor preset: disabled)
     Active: failed (Result: protocol) since Mon 2020-11-16 22:48:26 CET; 12s ago
       Docs: https://github.com/weetmuts/wmbusmeters
             man:wmbusmeters(1)
    Process: 5448 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
    Process: 5449 ExecStart=/usr/sbin/wmbusmetersd --device=rtlwmbus /run/wmbusmeters/wmbusmeters-rtlwmbus.pid (code=exited, status=0/SUCCESS)
        CPU: 21ms

Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) "" ""
Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) loading meter file /etc/wmbusmeters.d/sensus
Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) name=sensus
Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) type=iperl
Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) id=00111979
Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) key=<notprinted>
Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) shell=/usr/bin/mosquitto_pub -h piwnica -t stat/sensus/dw -m "$METER_JSON"
Nov 16 22:48:26 t630 wmbusmetersd[5450]: (config) overriding device with "rtlwmbus"
Nov 16 22:48:26 t630 wmbusmetersd[5450]: You cannot mix auto with stdin or a file.
Nov 16 22:48:26 t630 systemd[1]: Failed to start "wmbusmeters service on rtlwmbus device".
```

weetmuts, 2020-11-17:
Ah, @dwrobel , you are using the old service file. Check the install script for the new version.
The new style is simpler, no use of  @.service just a single wmbusmeters.service
No udev rules, they myst be removed.

The daemon stays running all the time, it checks for wmbus dongles every 2 seconds
and starts/stops them whenever they come and go. Therefore wmbusmeters must
be a member of plugdev to be able to grab the inserted devices.

dwrobel, 2020-11-17:
> Ah, @dwrobel , you are using the old service file. Check the install script for the new version.
> The new style is simpler, no use of @.service just a single wmbusmeters.service

I understand you, but the problem doesn't seem to be with the service file. It looks that you cannot specify `--device=` option if it already exists in the `wmbusmeters.conf` file _(especially if the option from command line differs from what you have in the config file)_.

> No udev rules, they myst be removed.

In Fedora I never installed any udev rules specifically for wmbusmeters.

dwrobel, 2020-11-17:
There seems to be also inconsistency w.r.t `rtlwmbus` in the documentation:

>auto:c1, to have wmbusmeters probe for devices: im871a, amb8465, cul, rc1180 or **rtlsdr** (spawns rtlwmbus).

>**rtlwmbus**, to spawn the background process: "rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus" for each attached rtlsdr dongle.
>**rtlwmbus**:868.9M, to tune to this fq instead.
>**rtlwmbus**:CMD(), to specify the entire background process command line that is expected to produce rtlwbus compatible output. Likewise for rtl433.

Is there any intentional difference between `rtlsdr` vs `rtlwmbus`? The former seems to work but the latter doesn't in `>0.9.36`.

weetmuts, 2020-11-17:
>> Is there any intentional difference between rtlsdr vs rtlwmbus
The device that can be probed is the rtl-sdr dongle. Then you have to start a decoder.
There are two possible decoders: rtl-wmbus and rtl-433.
By default, if an rtlsdr dongle is probed and detected, then wmbusmeters will start rtl_sdr | rtl_wmbus.

It should never have worked to specify just rtlsdr. :-) It was always rtlwmbus, which implies there
is an rtlsdr dongle somewhere.

weetmuts, 2020-11-17:
> It looks that you cannot specify `--device=` option if it already exists in the `wmbusmeters.conf` file 

I will take a look. Since the --device override is no longer necessary for daemon mode, I have not
tested it that much. Perhaps what you are experiencing is that it just adds another device 
to the devices that should be used, not overriding any existing ones. That could be a bug. I will check.

weetmuts, 2020-11-25:
I fixed the --device= override bug. The bug was that it added another device, without remove the existing devices.
The option now clears out any existing devices loaded from the conf file before adding 
the supplied override device.

Please test!
@dwrobel

dwrobel, 2020-12-07:
Tested on v1.0.4 tagged version.

Overrides from `auto:c1,t1` -> `rtlwmbus` doesn't seem to work:
```
$ cat /etc/wmbusmeters.conf 
loglevel=debug
device=auto:c1,t1
logtelegrams=true
format=json
```

```
$ cat /etc/wmbusmeters.d/sensus 
name=sensus
type=iperl
id=00111979
key=NOKEY
shell=/usr/bin/mosquitto_pub -h piwnica -t stat/sensus/dw -m "$METER_JSON"
```

```
# /usr/sbin/wmbusmetersd --device=rtlwmbus /run/wmbusmeters/wmbusmeters-rtlwmbus.pid

(wmbusmeters) started /run/wmbusmeters/wmbusmeters-rtlwmbus.pid
(config) "device" "auto:c1,t1"
(config) "logtelegrams" "true"
(config) "format" "json"
(config) "" ""
D pid=12172 uid=0 auid=1000 ses=4 msg='op=PAM:session_close grantors=pam_keyinit,pam_limits,pam_keyinit,pam_limits,pam_systemd,pam_unix acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
SP pid=12172 uid=0 auid=1000 ses=4 msg='op=PAM:setcred grantors=pam_env,pam_fprintd acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
(config) loading meter file /etc/wmbusmeters.d/sensus
(config) name=sensus
(config) type=iperl
(config) id=00111979
(config) key=<notprinted>
(config) shell=/usr/bin/mosquitto_pub -h piwnica -t stat/sensus/dw -m "$METER_JSON"
(config) overriding device with "rtlwmbus"
(wmbusmeters) version: 1.0.4
(config) using device: rtlwmbus:any
(config) number of meters: 1
(main) configured "sensus" "iperl" "00111979" not-encrypted
(serial) expecting devices to work
(lookup) with file "rtlwmbus"
(lookup) driver: raw file
(main) opening rtlwmbus:any
Started config rawtty on rtlwmbus listening on any
(serial) override with devicefile: rtlwmbus
(rawtty) on rtlwmbus
(serialfile) reading from file rtlwmbus (override rtlwmbus)
(main) regular reset of rawtty  on rtlwmbus will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rawtty
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
(serial) waiting for stop
(serial) no more data on file fd=4
(serial) received binary "3233363932350A"
(serialfile) closed rtlwmbus -1 (override rtlwmbus)
(wmbus) checkWMBUSFrame
 "3233363932350A"
(wmbus) less than 11 bytes, partial frame
(serial) no devices working emergency exit!
(serial) stopping manager
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbusmeters) shutting down
(wmbus) deleted rawtty
(serial) no devices working emergency exit!
```

On the other hand overriding from `rtlwmbus` -> `rtlwmbus` also doesn't work:

```
$ cat /etc/wmbusmeters.conf 
loglevel=debug
#device=auto:c1,t1
device=rtlwmbus
logtelegrams=true
format=json
```

```
# /usr/sbin/wmbusmetersd --device=rtlwmbus /run/wmbusmeters/wmbusmeters-rtlwmbus.pid

(config) "#device=auto:c1,t1" ""
(config) "device" "rtlwmbus"
(config) "logtelegrams" "true"
SP pid=12286 uid=0 auid=1000 ses=4 msg='op=PAM:setcred grantors=pam_env,pam_fprintd acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'
(config) "format" "json"
(config) "" ""
(config) loading meter file /etc/wmbusmeters.d/sensus
(config) name=sensus
(config) type=iperl
(config) id=00111979
(config) key=<notprinted>
(config) shell=/usr/bin/mosquitto_pub -h piwnica -t stat/sensus/dw -m "$METER_JSON"
(config) overriding device with "rtlwmbus"
You can only specify one stdin or one file or one command!
```

weetmuts, 2020-12-31:
I see from the last log that you probably have a file called rtlwmbus in your CWD! :-)
So wmbusmeters sees this file and thinks that it should load from it instead... thus
causing very confusing error messages.
 
I have just pushed a fix for this. It will now ignore any rtlwmbus/rtl433 named files....

Also I have tested it now with auto:c1,t1 in the conf file and
--device=rtlwmbus:t1 on the daemon command line.
It seems to be working, ie it says it uses only t1 instead of c1,t1.

So I think your problems are fixed! Please test!

dwrobel, 2021-01-08:
>So I think your problems are fixed! Please test!

I tested it using cff3eb5b30015a5e7dc54f99623dc404b62d2bae and it seems to work fine. IMHO the issue can be closed.

Thank you for the support!.

closed 2021-01-11 by weetmuts