wmbusmeters/wmbusmeters #178: rtlwmbus ignored after upgrade to 1.0.1-1-g3bd42bb

pswiatki, 2020-11-01:
Just switched to `master` branch rebased to the latest commit and built the app. It starts, but seems to ignore the RTL-SDR stick (and `rtlwmbus` device). I have the following setup in `/etc/wmbusemeters.conf`:

```
loglevel=debug
device=rtlwmbus[0000616]
logtelegrams=true
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
meterfilesnaming=name-id
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log
```
However, when the service is started I find the following in `/var/log/wmbusmeters/wmbusmeters.log`:
```
(wmbusmeters) logging started 2020-11-01 17:13:15
(wmbusmeters) version: 1.0.1-7-g5e273dd
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
*(config) using device: .[0000616]*
(config) number of meters: 2
(main) configured "P15_Prund" "amiplus" "00109314" encrypted
(main) configured "P15_Woda" "izar" "4122bcb2" not-encrypted
```

weetmuts, 2020-11-01:
You are quite right. The rtlwmbus[id] detection is currently broken.
Can you temporarily remove [0000616] and use just rtlwmbus.
With the fix I just pushed that should work.

pswiatki, 2020-11-01:
OK, I will fetch and recompile. Will let you know how it behaves.

pswiatki, 2020-11-01:
It looks much better now. I forgot to change the setup and left as it was (with serial). This is the result:

```
wmbusmeters) logging started 2020-11-01 23:36:28
(wmbusmeters) version: 1.0.1-10-g2c19a83
(config) store meter files in: "/var/log/wmbusmeters/meter_readings"
(config) using device: rtlwmbus[0000616]
(config) number of meters: 2
(main) configured "P15_Prund" "amiplus" "00109314" encrypted
(main) configured "P15_Woda" "izar" "4122bcb2" not-encrypted
(main) rtlsdr device 0000616 not currently used.
(main) found specified device (rtlwmbus[0000616]) that matches detected device (:rtlwmbus[0000616]:0/0)
Started auto rtlwmbus[0000616] listening on none
(rtlwmbus) using command: /usr/bin/rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus
(rtlwmbus) opening 0000616
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 29656 fd 4 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(wmbusmeters) waiting for telegrams
(serial) waiting for stop
(memory) rss 3813376 peak 3.63 MiB
(serial) received ascii "T1;1;1;2020-11-01 22:36:30.000;6;14;2131C756;0x1944304c56c73121d401a2311d00130c5509e39d4572981247cb<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-11-01 22:36:30.000;6;14;2131C756;0x1944304c56c73121d401a2311d00130c5509e39d4572981247cb<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 26
```

weetmuts, 2020-11-12:
Detection and specification of device now work!
You can say:
wmbusmeters rtlwmbus
and specifying an id:
wmbusmeters rtlwmbus[1234]:434M

Please test!

weetmuts, 2020-12-01:
It seems to be working for me, so closing this for now.

closed 2020-12-01 by weetmuts