wmbusmeters/wmbusmeters #181: Support for Multical 803 heat meter

Nikodemski2002, 2020-11-04:
Hi Weetmuts. 

As you can guess, I'm trying to add support for Kamstrup Multical 803 heat meter. I've read through the #165 issue, but my inexperience in terms of m-bus protocol, or overlooking something left me stuck. 

1. So as in #165 I've made a copy of the meter_multical403.cc file, and replaced all instances of 403 with 803.
2. I've added the lines for multical 803 in meters.h:

LIST_OF_METERS:
_X(multical403,C1_bit, Heat, MULTICAL403, Multical403)
X(multical603,C1_bit, Heat, MULTICAL603, Multical603)
X(multical603,C1_bit, Heat, MULTICAL803, Multical803)_

_METER_DETECTION:
X(MULTICAL403,MANUFACTURER_KAM, 0x0d, 0x34)
X(MULTICAL603,MANUFACTURER_KAM, 0x04, 0x35)
X(MULTICAL803,MANUFACTURER_KAM, 0x04, 0x39)_

3. I've set the meter file to:
_name=79806477
type=multical803:c1
id=79806477
key=00112233445566778899AABBCCDDEEFF_
4. When I used the `wmbusmeters /dev/ttyUSB0:im871a:c1 79806477 multical803:c1 79806477 00112233445566778899AABBCCDDEEFF` 
I didn't receive any data, also when I tried resetting the daemon or the whole RPi, the same scenario when I changed the meter type in the meter file to multical603:c1 (and of course in the listening command too).
5. When I changed the meter type in the meter file, and command to multical403, I received mixed data, as shown below:
```
Started config im871a on /dev/ttyUSBO listening on c1
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmet ers/
(meter) to add support for this unknown mfct,media,version combination
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
79806477	79806477	0 kWh	0 m3	0 m3/h	0 °C	0 °C	SENSOR_Tl_BELOW_MEASURING_RANGE SENSOR_T2_BELOW_MEASURING_RANGE 2020-10-27 22:10.14
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
79806477	79806477	0 kWh	0 m3	0 m3/h	0 °C	0 °C	SENSOR_Tl_BELOW_MEASURING_RANGE SENSOR_T2_BELOW_MEASURING_RANGE 2020-10-27 22:10.30
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
79806477	79806477	0	kWh	0	m3	0	m3/h	0	°C	0	°C	SENSOR_Tl_BELOW_MEASURING_RANGE	SENSOR_T2_BELOW_MEASURING_RANGE	2020-10-27	22:10.46
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
79806477	79806477	0	kWh	0	m3	0	m3/h	0	°C	0	°C	SENSOR_Tl_BELOW_MEASURING_RANGE	SENSOR_T2_BELOW_MEASURING_RANGE	2020-10-27	22:11.02
(meter) 79806477: meter detection did not match the selected driver multical403! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
79806477	79806477	0	kWh	0	m3	0	m3/h	0	°C	0	°C	SENSOR_Tl_BELOW_MEASURING_RANGE	SENSOR_T2_BELOW_MEASURING_RANGE	2020-10-27	22:11.18
```
Best regards.
Nikodem.

weetmuts, 2020-11-04:
Great! I updated the HowToAddaNewMeter.txt with a lot more info.
Most important, build/run/test should not involve updating the daemon, that
is just a difficult way to develop.
Run from the command line instead! Read the Howto and please ask me any questions
here so that I can improve the HowTo.

Nikodemski2002, 2020-11-06:
Hi Weetmuts. Thanks for your support.
I have a problem with: 
` wmbusmeters simulation_multical803.txt` and  
`wmbusmeters --debug simulation_multical803.txt Wasser supercom587 12345678 <my key>`
when I execute these commands, I receive: 
`You must supply at least one device (eg auto:c1) to receive wmbus telegrams.`  

I tried using `device=auto:c1`, `device=auto:c1,t1` and `/dev/ttyUSB0:im871a:c1`,
And neither of these options worked.

Best regards.
Nikodem.

weetmuts, 2020-11-07:
It means that you do not have a file in your directory called simulation_multical803.txt 
Sorry for the bad error message, I will see if can try to fix that.
The file name must begin with simulation_ for wmbusmeters to know that it is a simulation
file with telegram=|..| inside.

If the name does not begin with simulation_ then wmbusmeters assumes it is file containing raw wmbus telegram bytes.
Or you can type: "wmbusmeters myfile.txt:rtlwmbus" which expects myfile.txt to contain rtlwmbus formatted telegrams.

Nikodemski2002, 2020-11-07:
Hi. But I have already created a file called simulation_multical803.txt. It also contains the `telegram=|...|` inside.
I've also tested `wmbusmeters simulation_multical803.txt` and the result was the same.
```
pi@raspberrypi:~/wmbusmeters $ cd simulations
pi@raspberrypi:~/wmbusmeters/simulations $ ls
serial_rawtty_bad.hex  serial_rtlwmbus_ok.msg          simulation_alarm.txt  simulation_conversionsadded.txt  simulation_multical803.txt         simulation_t1_and_c1.txt
serial_rawtty_ok.hex   simulation_additional_json.txt  simulation_apas.txt   simulation_duplicates.txt        simulation_multiple_qcalorics.txt  simulation_t1.txt
serial_rtl433_ok.msg   simulation_aes.msg              simulation_c1.txt     simulation_multical603.txt       simulation_shell.txt               simulation_unknown.txt
```

weetmuts, 2020-11-07:
You have to execute wmbusmeters in the same dir as your sim file, or give a full path to it.

weetmuts, 2020-11-10:
wmbusmeters /home/me/simulation_foo.txt
or if pwd is /home/me
wmbusmeters ./simulation_foo.txt
or 
wmbusmeters simulation_foo.txt

Nikodemski2002, 2020-11-10:
Hi. Sorry for the inactivity, I've already copied 8 telegrams to the simulation file, and run the simulation properly (I thought it will automatically pick the path to the simulations folder). Now I'm analyzing it. This is what I've got from the simulation:
```
(wmbusmeters) version: 1.0.1-19-g1df5af4
(config) using device: wmbusmeters/simulations/simulation_multical803.txt:any
(config) number of meters: 1
(main) configured "Wasser" "supercom587" "79806477" encrypted
(lookup) with file "wmbusmeters/simulations/simulation_multical803.txt:any"
(lookup) driver: simulation file
Started config simulation on wmbusmeters/simulations/simulation_multical803.txt listening on any
(simulation) in wmbusmeters/simulations/simulation_multical803.txt
(main) regular reset of simulation  on wmbusmeters/simulations/simulation_multical803.txt will happen every 82800 seconds
(main) added wmbusmeters/simulations/simulation_multical803.txt to files
(wmbus) no alarm (expected activity) for simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "09472D2C776480793904"
(wmbus) parseDLL @0 10
(meter) Wasser: for me? 79806477
(meter) Wasser: meter detection did not match the selected driver supercom587! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Wasser: yes for me
(meter) Wasser supercom587 handling telegram from 79806477
(meter) Wasser 79806477 "09472D2C776480793904"
(wmbus) parseDLL @0 10
(telegram) DLL L=09 C=47 (from meter ACC_NR) M=2c2d (KAM) A=79806477 VER=39 TYPE=04 (Heat meter) (driver unknown!) DEV= RSSI=0
(wmbus) telegram from 79806477 ignored by all configured meters!
(simulation) from file "88442D2C7764807939048D2086405132200A32EE97FCC2E664F8FCD1E94C67AABD0482DC579DA2094DEE9A2C29263F4FF7C67A88DC359DDDA76FA020E9761BB08B20496AD1D78556513617554BB1246F1FFCB71C93164E66FF680D455A906E391C4B0E49C7D5F196A6E351BC0FF039C0E5757DFF95ED6B9A39CEF2B5D5B93255456677D4E53E05C170"
(wmbus) parseDLL @0 137
(meter) Wasser: for me? 79806477
(meter) Wasser: meter detection did not match the selected driver supercom587! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Wasser: yes for me
(meter) Wasser supercom587 handling telegram from 79806477
(meter) Wasser 79806477 "88442D2C7764807939048D2086405132200A32EE97FCC2E664F8FCD1E94C67AABD0482DC579DA2094DEE9A2C29263F4FF7C67A88DC359DDDA76FA020E9761BB08B20496AD1D78556513617554BB1246F1FFCB71C93164E66FF680D455A906E391C4B0E49C7D5F196A6E351BC0FF039C0E5757DFF95ED6B9A39CEF2B5D5B93255456677D4E53E05C170"
(wmbus) parseDLL @0 137
(telegram) DLL L=88 C=44 (from meter SND_NR) M=2c2d (KAM) A=79806477 VER=39 TYPE=04 (Heat meter) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 127
(ELL) decrypting "0A32EE97FCC2E664F8FCD1E94C67AABD0482DC579DA2094DEE9A2C29263F4FF7C67A88DC359DDDA76FA020E9761BB08B20496AD1D78556513617554BB1246F1FFCB71C93164E66FF680D455A906E391C4B0E49C7D5F196A6E351BC0FF039C0E5757DFF95ED6B9A39CEF2B5D5B93255456677D4E53E05C170"
(ELL) IV 2D2C7764807939042040513220000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "84C178040F0000000004FF0700000000"
(ELL) block 1 block_size 16 offset 16
(ELL) decrypted "04FF0800000000041400000000844014"
(ELL) block 2 block_size 16 offset 32
(ELL) decrypted "000000008480401400000000043B0000"
(ELL) block 3 block_size 16 offset 48
(ELL) decrypted "000002590000025D0000142D00000000"
(ELL) block 4 block_size 16 offset 64
(ELL) decrypted "84100F0000000084200F0000000004FF"
(ELL) block 5 block_size 16 offset 80
(ELL) decrypted "2260000100026C892B440F0000000044"
(ELL) block 6 block_size 16 offset 96
(ELL) decrypted "1400000000C4401400000000C4804014"
(ELL) block 7 block_size 8 offset 112
(ELL) decrypted "00000000426C812B"
(ELL) decrypted "84C178040F0000000004FF070000000004FF0800000000041400000000844014000000008480401400000000043B0000000002590000025D0000142D0000000084100F0000000084200F0000000004FF2260000100026C892B440F00000000441400000000C4401400000000C480401400000000426C812B"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=86 SN=40513220 (AES_CTR session=0 time=206100) CRC=84c1
(wmbus) parseAFL @19 118
(wmbus) parseTPL @19 118
(dvparser) found new format "040F04FF0704FF08041484401484804014043B0259025D142D84100F84200F04FF22026C440F4414C44014C4804014426C" with hash e7a2, remembering!
(telegram) TPL CI=78
telegram=|88442D2C7764807939048D20864051322084C178|040F0000000004FF070000000004FF0800000000041400000000844014000000008480401400000000043B0000000002590000025D0000142D0000000084100F0000000084200F0000000004FF2260000100026C892B440F00000000441400000000C4401400000000C480401400000000426C812B|+0
(supercom587) 00: 88 length (136 bytes)
(supercom587) 01: 44 dll-c (from meter SND_NR)
(supercom587) 02: 2d2c dll-mfct (KAM)
(supercom587) 04: 77648079 dll-id (79806477)
(supercom587) 08: 39 dll-version
(supercom587) 09: 04 dll-type (Heat meter)
(supercom587) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
[... 398 lines cut ...]
(supercom587) 4a: 00000000
(supercom587) 4e: 00000000
(supercom587) 52: 00000000
(supercom587) 56: 00000000
(supercom587) 5a: 812B
Wasser  79806477        0 m3    2020-11-09 15:46.28
(simulation) from file "09472D2C776480793904"
(wmbus) parseDLL @0 10
(meter) Wasser: for me? 79806477
[... cut ...]
(meter) Wasser supercom587 handling telegram from 79806477
(meter) Wasser 79806477 "09472D2C776480793904"
(wmbus) parseDLL @0 10
(telegram) DLL L=09 C=47 (from meter ACC_NR) M=2c2d (KAM) A=79806477 VER=39 TYPE=04 (Heat meter) (driver unknown!) DEV= RSSI=0
(wmbus) telegram from 79806477 ignored by all configured meters!
(serial) stopping manager
(serial) waiting for stop
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted simulation
```

Nikodemski2002, 2020-11-18:
Hi. I do things as you and HowToAddNewMeter.txt said. This is what I recive after executing the `sudo make DEBUG=true` command:
```
Building 1.0.1-19-g1df5af4 with local changes
g++ -O0 -ggdb -fsanitize=address -fno-omit-frame-pointer -fprofile-arcs -ftest-c                                                                                                             overage -fPIC -std=c++11 -Wall -Werror=format-security -Ibuild_debug src/main.cc                                                                                                              -c -E > build_debug/main.o.src
g++ -O0 -ggdb -fsanitize=address -fno-omit-frame-pointer -fprofile-arcs -ftest-c                                                                                                             overage -fPIC -std=c++11 -Wall -Werror=format-security -Ibuild_debug src/main.cc                                                                                                              -MMD -c -o build_debug/main.o
src/main.cc: In function ‘std::shared_ptr<Meter> create_meter(Configuration*, Me                                                                                                             terType, MeterInfo*, const char*)’:
src/main.cc:297:20: error: ‘createMultical803’ was not declared in this scope
             newm = create##cname(*mi);                      \
                    ^~~~~~
src/meters.h:57:5: note: in expansion of macro ‘X’
     X(multical803,C1_bit, Heat,        MULTICAL803, Multical803)  \
     ^
src/main.cc:304:1: note: in expansion of macro ‘LIST_OF_METERS’
 LIST_OF_METERS
 ^~~~~~~~~~~~~~
src/main.cc:297:20: note: suggested alternative: ‘createMultical603’
             newm = create##cname(*mi);                      \
                    ^~~~~~
src/meters.h:57:5: note: in expansion of macro ‘X’
     X(multical803,C1_bit, Heat,        MULTICAL803, Multical803)  \
     ^
src/main.cc:304:1: note: in expansion of macro ‘LIST_OF_METERS’
 LIST_OF_METERS
 ^~~~~~~~~~~~~~
make: *** [Makefile:105: build_debug/main.o] Error 1
```

weetmuts, 2020-11-19:
Can you do a "git pull" to get the latest source?
I made a small change to the declaration of createXXXX()
previously they were manually added to the end of meters.h

Now they are automatically added with this:
#define X(mname,linkmode,info,type,cname) shared_ptr<info> create##cname(MeterInfo &m);
LIST_OF_METERS
#undef X

Do you have this X macro at line 370 in meters.h?

Nikodemski2002, 2020-11-20:
Hi. After `git pull`, I've passed the `sudo make DEBUG=true`. I wanna try to execute the JSON simulation, assuming that at least 75% of the data from multical803, will match the things copied (and changed ofc.) from meter_multical603.cc to my meter_multical803.cc file.
 But after trying it, I receive information, as bellow:
```
pi@raspberrypi:~/wmbusmeters/build_debug $ wmbusmeters --format=json simulation_multical803.txt Heat multical803 79806477 <key>
Not a valid meter type "multical803"
```
I've also tried this.
```
pi@raspberrypi:~/wmbusmeters/build_debug $ wmbusmeters --format=json simulation_multical803.txt Heat supercom587 79806477 <KEY>
(meter) Heat: meter detection did not match the selected driver supercom587! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Heat: meter detection did not match the selected driver supercom587! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```

weetmuts, 2020-11-22:
Try ./build_debug/wmbusmeters --format......etc
So that you are sure that it is the right wmbusmeters binary that is actually executed.

Nikodemski2002, 2020-11-22:
I'm receiving `Unknown option "--format"`. 
I've reinstalled everything (and redo every step of adding new meter) because I was thinking, that I messed up sth when trying to repair errors, but after all when I try to execute the:
```
pi@raspberrypi:~/wmbusmeters/build_debug $ wmbusmeters --format=json simulation_multical803.txt Heat multical803 79806477 <key>
```
I've received
```
Not a valid meter type "multical803"

```

weetmuts, 2020-11-23:
Looks odd. How about you push your code to a fork, so that I can take a look!

Nikodemski2002, 2020-11-23:
Hi. This link to my fork:

 [https://github.com/Nikodemski2002/wmbusmeters](https://github.com/Nikodemski2002/wmbusmeters)

Nikodemski2002, 2020-11-25:
Hi @weetmuts, so did you found out sth?

Nikodemski2002, 2020-11-25:
Hi. I missed, that the build_debug folder was included in the .gitignore files. Now everything is in place, as in my local changes.

Nikodemski2002, 2020-11-25:
The one thing that is most curious for me, is why the driver is still: unknown! . Everything in the meters.h file is fine and it corresponds to values, which are printed when using the `wmbusmeters --logtelegrams /dev/ttyUSB0:im871a:c1` command.

weetmuts, 2020-11-26:
Do git pull; make clean; make
then run: ./build/wmbusmeters --version
it should say:
```
wmbusmeters: 1.0.3-12-g2b39daa
2b39daab39b5744eac5d32950e12e1232e4923f5
```

weetmuts, 2020-11-26:
I tested your fork! It works great! I think you are just running an old wmbusmeters binary somewhere.
If you do: "make"
then you will find the binary to run here:
./build/wmbusmeters

If you do: "make DEBUG=true"
then you will find the binary to run here:
./build_debug/wmbusmeters

When you do "sudo make install"
it will take the binary ./build/wmbusmeters
and install that into /usr/bin

Do a "make clean; make; sudo make install"
check ./build/wmbusmeters --version
and 
wmbusmeters --version

so that they are the same.

I can easily bring in your code into wmbusmeters right now if you like.
You do not have to create a pull-request.

I tested on one of your old telegrams:

./build/wmbusmeters simulation_multical803.txt 
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 79806477
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Heat meter (0x04)
                   ver: 0x39
                driver: multical803

Nikodemski2002, 2020-11-28:
Hi. Now everything works great! Thank you very much! I've tested the:
```
wmbusmeters /dev/ttyUSB0:im871a:c1 79806477 multical803:c1 79806477 00112233445566778899AABBCCDDEEFF
```
And the results were as expected.

But, there's an one problem more, namely after adding my meters (multical21 and multical803) in /etc/wmbusmeters.d/ directory, and restarting daemon, reloading configure file and even rebooting whole raspberry pi, I don't see any readings in var/log/wmbusmeters/meter_readings directory. If I use `wmbusmeters auto:c1` I see the results normally.

If you want me to add this as another issue, let me know.

Best regards.
Nikodem.

weetmuts, 2020-11-28:
sudo rm /var/log/wmbusmeters/wmbusmeters.log
sudo systemctl restart wmbusmeters.service
Then post /var/log/wmbusmeters/wmbusmeters.log here

Nikodemski2002, 2020-12-01:
This is what I've got:
(wmbusmeters) logging started 2020-12-01 21:29:30

And this is what I have in my wmbusmeters.conf file:
```
loglevel=normal
device=auto:c1
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
meterfilesnaming=name
meterfilestimestamp=day
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=echo curl -i -XPOST 'http://localhost:8086/write?db=multical21' --data-binary 'm3,meter_id='$METER_ID' value='$METER_TOTAL_M3''

```

Nikodemski2002, 2020-12-03:
Hi. I also tested the wmbusmeters command with shell added, and I recieved these results:
```
wmbusmeters --shell="echo curl -i -XPOST 'http://localhost:8086/write?db=multical21' --data-binary 'm3,meter_id='$METER_ID' value='$METER_TOTAL_M3''" /dev/ttyUSB0:im871a:c1 74887686 multical21 74887686 <key>
Started config im871a[00101983] on /dev/ttyUSB0 listening on c1
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
curl -i -XPOST http://localhost:8086/write?db=multical21 --data-binary m3,meter_id= value=
```
With the config file set to:
```
loglevel=normal
device=auto:c1
logtelegrams=false
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
meterfilesnaming=name
meterfilestimestamp=micros
logfile=/var/log/wmbusmeters/wmbusmeters.log
shell=echo curl -i -XPOST 'http://localhost:8086/write?db=multical21' --data-binary 'm3,meter_id='$METER_ID' value='$METER_TOTAL_M3''
ignoreduplicates=false
``` 
And the logs in a log file
```
(wmbusmeters) logging started 2020-12-03 12:30:00
```

weetmuts, 2020-12-03:
Its the quotes! :-) 

For the conf file:
When you place the $METER_ID etc betwen ' single quotes, the shell (which is invoked by wmbusmeters) will not expand the variable $METER_ID. So use " double quotes instead!
This should work 
shell=echo curlxxx "meter_id=$METER_ID" "value=$METER_TOTAL_M3"

For the command line: 
It gets complicated because the shell is going to look for variables before wmbusmeters is started 
so the best is to quote the whole command in single quotes, like this:
--shel='curl .... "$METER_ID" ..... '
Your current problem is that the shell will replace $METER_ID with nothing even before wmbusmeters is started.

This should work:
wmbusmeters --shell='echo curlxxx "m3,meter_id=$METER_ID" "value=$METER_TOTAL_M3" ' auto:c1 ...

Nikodemski2002, 2020-12-03:
Hi. I will test the curl changes that you have mentioned. But now I have reinstalled everything, without my mc803 changes just to check if it is working with mc21, and I'm still don't have any readings in var/log/wmbusmeters/meter_readings.
The config I'm using right now is:
```
loglevel=normal
device=auto:c1
logtelegrams=false
metefilesaction=append
meterfilestimestamp=day
format=json
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=overwrite
logfile=/var/log/wmbusmeters/wmbusmeters.log
ignoreduplicates=false
```

weetmuts, 2020-12-03:
set loglevel=debug 
and post wmbusmeters.log here.

weetmuts, 2020-12-05:
I pushed your work into the main wmbusmters. Please use the master branch to test now! :-)

Slight adjustment to the json, wmbusmeters should always normalize values into kWh for energy.
So it reports "energy_forward_kwh":0
If you want to GJ then add the option
--addconversions=GJ
and you will also have:
"energy_forward_gj":0

Nikodemski2002, 2020-12-08:
Hi sorry for my inactivity, I was away from home for a few days. The problem was that the auto:c1 detected my device as amb8465 instead of im871a, but I changed it to static im871a and now it works. Bu still the curl doesn't work even with the quotes changes you mentioned. There also another problem. When I use command `sudo systemctl start wmbusmeters` I receive ` System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
` Do you know is there else option of starting wmbusmteres service? Or how I can repair this. `Sudo service wmbusmeters start` didn't work.

weetmuts, 2020-12-10:
Hmm, these seems like more generic GNU/Linux  problems. I really do not have time to follow up on such things, sorry.
But it seems like your system does not have systemd installled? The current install script assumes GNU/Linux system
that uses systemd for its daemons. What is the name of your GNU/Linux distribution?

You can start your daemon manually if you like:
sudo runuser -u wmbusmeters wmbusmetersd /home/wmbusmeters/wmbusmeters.pid

Make sure that /home/wmbusmeters/wmbusmeters.pid is owned and writeable by the wmbusmeters user.

weetmuts, 2020-12-31:
"The problem was that the auto:c1 detected my device as amb8465 instead of im871a, "
If this really happens, then it is a bug that should be fixed. :-) Please open a new issue if it happens again!

Can we close this particular issue now?

weetmuts, 2020-12-31:
I'll close it for now! Please reopen a new ticket with any follow up problems that you have! I hope the new multical 803 driver works for you!

closed 2020-12-31 by weetmuts

Nikodemski2002, 2021-01-03:
Hi @weetmuts. Thanks for your help. After connecting the meter I saw that the total energy consumption is still displaying as 0, but it should be sth around 30. The meter wMBUS module is set for 30-10-102 configuration (Image 1). So the total energy consumption stands for E1 in the table. The meter is sending the E1 in GJ (Image 2). So do you have any idea, what I should change in the meter_multical803.cc file or somewhere else to see the total energy consumption. I also want to see the actual power in kWh, but I have to wait for the optical eye to change the wMBUS module configuration, to make it send this particular data.

This is what I've got from the ` wmbusmeters --format=json --debug auto:c1 79806477 multical803:c1 79806477 [KEY]`
```
(multical803) 00: 8c length (140 bytes)
(multical803) 01: 44 dll-c (from meter SND_NR)
(multical803) 02: 2d2c dll-mfct (KAM)
(multical803) 04: 77648079 dll-id (79806477)
(multical803) 08: 39 dll-version
(multical803) 09: 04 dll-type (Heat meter)
(multical803) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical803) 0b: 20 ell-cc (slow_resp sync)
(multical803) 0c: 59 ell-acc
(multical803) 0d: 93b84520 sn (AES_CTR)
(multical803) 11: 4770 payload crc (calculated 4770 OK)
(multical803) 13: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(multical803) 14: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 15: FB vif (First extension of VIF-codes)
(multical803) 16: 08 vife (10^-1 GJ)
(multical803) 17: A5010000
(multical803) 1b: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 1c: FF vif (Vendor extension)
(multical803) 1d: 07 vife (?)
(multical803) 1e: * 24120000 something A (0.000000)
(multical803) 22: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 23: FF vif (Vendor extension)
(multical803) 24: 08 vife (?)
(multical803) 25: * 270E0000 something B (0.000000)
(multical803) 29: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 2a: 15 vif (Volume 10⁻¹ m³)
(multical803) 2b: * 831B0000 total volume (704.300000 m3)
(multical803) 2f: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 30: 40 dife (subunit=1 tariff=0 storagenr=0)
(multical803) 31: 14 vif (Volume 10⁻² m³)
(multical803) 32: 00000000
(multical803) 36: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 37: 80 dife (subunit=0 tariff=0 storagenr=0)
(multical803) 38: 40 dife (subunit=2 tariff=0 storagenr=0)
(multical803) 39: 14 vif (Volume 10⁻² m³)
(multical803) 3a: 00000000
(multical803) 3e: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 3f: 3C vif (Volume flow 10⁻² m³/h)
(multical803) 40: * 47060000 volume flow (16.070000 m3/h)
(multical803) 44: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical803) 45: 59 vif (Flow temperature 10⁻² °C)
(multical803) 46: * AD1A T1 flow temperature (68.290000 °C)
(multical803) 48: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical803) 49: 5D vif (Return temperature 10⁻² °C)
(multical803) 4a: * B614 T2 flow temperature (53.020000 °C)
(multical803) 4c: 14 dif (32 Bit Integer/Binary Maximum value)
(multical803) 4d: 2E vif (Power kW)
(multical803) 4e: 04010000
(multical803) 52: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 53: 10 dife (subunit=0 tariff=1 storagenr=0)
(multical803) 54: FB vif (First extension of VIF-codes)
(multical803) 55: 08 vife (10^-1 GJ)
(multical803) 56: 00000000
(multical803) 5a: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 5b: 20 dife (subunit=0 tariff=2 storagenr=0)
(multical803) 5c: FB vif (First extension of VIF-codes)
(multical803) 5d: 08 vife (10^-1 GJ)
(multical803) 5e: 00000000
(multical803) 62: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 63: FF vif (Vendor extension)
(multical803) 64: 22 vife (per hour)
(multical803) 65: * 00000100 info codes ()
(multical803) 69: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical803) 6a: 6C vif (Date type G)
(multical803) 6b: * A321 target date (2021-01-03 00:00)
(multical803) 6d: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 6e: FB vif (First extension of VIF-codes)
(multical803) 6f: 08 vife (10^-1 GJ)
(multical803) 70: 00000000
(multical803) 74: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 75: 15 vif (Volume 10⁻¹ m³)
(multical803) 76: 00000000
(multical803) 7a: C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 7b: 40 dife (subunit=1 tariff=0 storagenr=1)
(multical803) 7c: 14 vif (Volume 10⁻² m³)
(multical803) 7d: 00000000
(multical803) 81: C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 82: 80 dife (subunit=0 tariff=0 storagenr=1)
(multical803) 83: 40 dife (subunit=2 tariff=0 storagenr=1)
(multical803) 84: 14 vif (Volume 10⁻² m³)
(multical803) 85: 00000000
(multical803) 89: 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 8a: 6C vif (Date type G)
(multical803) 8b: 0000
{"media":"heat","meter":"multical803","name":"79806477","id":"79806477","total_energy_consumption_kwh":0,"total_volume_m3":704.3,"volume_flow_m3h":16.07,"t1_temperature_c":68.29,"t2_temperature_c":53.02,"at_date":"2021-01-03 00:00","current_status":"","energy_forward_gj":"4644","energy_returned_gj":"3623","timestamp":"2021-01-03T15:31:37Z","device":"im871a[00101983]","rssi_dbm":-58}
```
**img1**
[image]

**img2**
[image]