wmbusmeters/wmbusmeters #182: IZAR RC 868 I did not match the selected driver izar

Delphius7, 2020-11-08:
Hi Weetmuts,

I have a "DIEHL IZAR RC 868 I" water meter and used the izar driver.
All values look good but I receive the following messages:
```
(meter) watermeter2: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```
./wmbusmeters rtlwmbus
```
Received telegram from: 12345678
manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
type: A/D converter (0x19)
ver: 0x00
device: rtlwmbus[00000001]
rssi: 47 dBm
driver: unknown!
```

Maybe it is similar to [#54 ](https://github.com/weetmuts/wmbusmeters/issues/54).

Please find debug logs attached below and let me know if you need any other details. I am new to this topic so I am not sure what is needed for this.

Thanks very much!

```
pi@raspberrypi:~/wmbusmeters/build $ ./wmbusmeters --debug rtlwmbus
(wmbusmeters) version: 1.0.1-19-g1df5af4
(config) using device: rtlwmbus
(config) number of meters: 0
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
Started auto rtlwmbus[00000001] listening on none
(rtlwmbus) using command: rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh pid 3914 fd 3 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "T1;0;0;2020-11-08 00:07:25.000;46;179;12345678;0x19ffa51178075828ff19a2310f001328e755ff89eb1d924b703e<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-11-08 00:07:25.000;46;179;12345678;0x19ffa51178075828ff19a2310f001328e755ff89eb1d924b703e<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-11-08 00:07:33.000;44;104;285807FF;0x1944a511ff0758280019a2410f001328829960d80e269fff3f33<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-11-08 00:07:33.000;44;104;285807FF;0x1944a511ff0758280019a2410f001328829960d80e269fff3f33<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-11-08 00:07:42.000;53;85;285807FF;0x1944a511ff0758280019a2ff0fffff2895b629732ff546912631<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-11-08 00:07:42.000;53;85;285807FF;0x1944a511ff0758280019a2ff0fffff2895b629732ff546912631<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;1;1;2020-11-08 00:07:51.000;47;70;12345678;0x1944a511780758280019a2610f001328acc7f38e4d812cfc0d36<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-11-08 00:07:51.000;47;70;12345678;0x1944a511780758280019a2610f001328acc7f38e4d812cfc0d36<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 26
(telegram) DLL L=19 C=44 (from meter SND_NR) M=11a5 (DME) A=12345678 VER=00 TYPE=19 (A/D converter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=47
(wmbus) parseELL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) TPL CI=a2
Received telegram from: 12345678
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: A/D converter (0x19)
                   ver: 0x00
                device: rtlwmbus[00000001]
                  rssi: 47 dBm
                driver: unknown!
(wmbus) 00: 19 length (25 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: a511 dll-mfct (DME)
(wmbus) 04: 78075828 dll-id (12345678)
(wmbus) 08: 00 dll-version
(wmbus) 09: 19 dll-type (A/D converter)
(wmbus) 0a: a2 tpl-ci-field (Mfct specific)
telegram=||1944A511780758280019A2610F001328ACC7F38E4D812CFC0D36|+29
```

```
pi@raspberrypi:~/wmbusmeters/build $ ./wmbusmeters --debug --format=json rtlwmbus watermeter2 izar 12345678 NOKEY
(wmbusmeters) version: 1.0.1-19-g1df5af4
(config) using device: rtlwmbus
(config) number of meters: 1
(main) configured "watermeter2" "izar" "12345678" not-encrypted
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
Started auto rtlwmbus[00000001] listening on none
(rtlwmbus) using command: rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 3827 fd 3 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received ascii "T1;0;0;2020-11-07 23:56:15.000;51;98;12345678;0x1944a511780758280019a2ffffff132893b629732ff546912631<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-11-07 23:56:15.000;51;98;12345678;0x1944a511780758280019a2ffffff132893b629732ff546912631<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2020-11-07 23:56:33.000;49;84;12345678;0x1944a5ff780758280019a271ff001328bde8ba256c52f5271434<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2020-11-07 23:56:33.000;49;84;12345678;0x1944a5ff780758280019a271ff001328bde8ba256c52f5271434<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;1;1;2020-11-07 23:56:41.000;51;180;12345678;0x1944a511780758280019a2010f001328d82446748969f8265b39<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2020-11-07 23:56:41.000;51;180;12345678;0x1944a511780758280019a2010f001328d82446748969f8265b39<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 26
(meter) watermeter2: for me? 12345678
(meter) watermeter2: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) watermeter2: yes for me
(meter) watermeter2 izar handling telegram from 12345678
(meter) watermeter2 12345678 "1944A511780758280019A2010F001328D82446748969F8265B39"
(wmbus) parseDLL @0 26
(telegram) DLL L=19 C=44 (from meter SND_NR) M=11a5 (DME) A=12345678 VER=00 TYPE=19 (A/D converter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=51
(wmbus) parseELL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) TPL CI=a2
telegram=|1944A511780758280019A2|010F001328D82446748969F8265B39|+28
(izar) Decoded PRIOS data: 4B129C0200869002009F2A
(izar) 00: 19 length (25 bytes)
(izar) 01: 44 dll-c (from meter SND_NR)
(izar) 02: a511 dll-mfct (DME)
(izar) 04: 78075828 dll-id (12345678)
(izar) 08: 00 dll-version
(izar) 09: 19 dll-type (A/D converter)
(izar) 0a: a2 tpl-ci-field (Mfct specific)
{"media":"water","meter":"izar","name":"watermeter2","id":"12345678","total_m3":171.026,"last_month_total_m3":168.07,"last_month_measure_date":"2020-10-31","remaining_battery_life_y":7.5,"current_alarms":"no_alarm","previous_alarms":"no_alarm","timestamp":"2020-11-07T23:56:41Z","device":"rtlwmbus[00000001]","rssi_dbm":51}
```

weetmuts, 2020-11-11:
Thanks! Fixed and I pushed the update where this mfct,type,ver combo is detected as izar.
Please test!

Why on earth do  they not even comply with the standard in the header.
A/D converter.... meh. Oh well. Lets hope that they at least take note within
their own company so that they do not release another kind of meter with 
the same combo.

Delphius7, 2020-11-12:
Works like a charm!
Thanks for your help and in general for all your effort.

```
Received telegram from: 28580778
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: A/D converter (0x19)
                   ver: 0x00
                device: rtlwmbus[00000001]
                  rssi: 68 dBm
                driver: izar
```

closed 2020-11-12 by Delphius7