wmbusmeters/wmbusmeters #184: CC110 M-Bus chip support

HSPilot2008, 2020-11-19:
The CC1101 chip is commonly used for M-Bus and various add-on cards for Arduino and RPI is on the market. It interfaces via SPI to the Arduino/RPI and has a set message format. Would appreciate if support for the CC1101 can be investigated/added.

See https://github.com/SpaceTeddy/CC1101 for wrapper function to connect to Arduino/RPI

weetmuts, 2020-11-19:
wmbusmeters supports the nanoCUL which contains a CC1101 chip.

Do you want to talk SPI directly to the CC1101 chip from a Raspberry PI
using its GPIO pins?

HSPilot2008, 2020-11-20:
Hi weetmuts. Yes I would like to use an CC1101 add-on card connected straight to the GPIO SPI interface.

weetmuts, 2020-11-20:
Do you have any links to where one can buy such add-on cards?

wmbusmeters currently supports either direct control of serial ports and talking to the dongle behind.
or starting a background process, that will output the telegrams (in some format, for example the rtlwmbus format)
and reading from the background process.

So if you can write a program (lets call it readcc1101, that when started configures the CC1101 and then starts listening,
whenever a telegram arrives, it prints it on stdout, in this rtlwmbus format:
T1;1;1;2019-04-03 19:00:42.000;97;148;88888888;0x2e44333003020100071b7a634820252f2f0265840842658308820165950802fb1aae0142fb1aae018201fb1aa9012f

then you can start wmbusmeters like this:

wmbusmeters rtlwmbus:CMD(readcc1101) 

and its good to go. :-)

HSPilot2008, 2020-11-20:
Hi weetmuts.
Thanks for this. 

My idea is to use this board
http://www.giga.co.za/ocart/index.php?route=product/product&product_id=330

Connected as described
https://salmg.net/2017/09/20/cc1101-transceiver-raspberry-pi/

There is a driver lib for the CC1101 on which can probably be used to print to stdout
https://github.com/SpaceTeddy/CC1101

weetmuts, 2021-01-24:
Closing for now. Please let us know if you manage to create a working readcc1101.

closed 2021-01-24 by weetmuts