wmbusmeters/wmbusmeters #185: use wmbusmeters and rtl_433 at the same time

Delphius7, 2020-11-20:
Hi,
I have a general question and not really an issue with wmbusmeters.
I hope it is ok that I ask this here anyways as I don't know where else I could ask it.

Currently I use a RTL2838 DVB-T stick with rtl_433 to collect some temperature sensor data. e.g.
`rtl_433 -F json -M utc | mosquitto_pub -h 127.0.0.1 -t sensors/rtl_433 -l`
Now I plan to also get data from my watermeter, which also works perfect with wmbusmeters.

However I cannot do both at the same time because both block the device:
```
test@raspberrypi:~/wmbusmeters/build $ ./wmbusmeters rtlwmbus
Started config rtlwmbus[00000001] listening on none
No meters configured. Printing id:s of all telegrams heard!
Lost 00000001 closing rtlwmbus[00000001]
Started config rtlwmbus[00000001] listening on none
Lost 00000001 closing rtlwmbus[00000001]
Started config rtlwmbus[00000001] listening on none
```

Is there any way to get both working without buying a second stick?
Does anyone here do this already?

Sorry, I am an absolute beginner and was searching already for days and cannot find an answer anywhere else.
Thanks!

BIBOLV, 2020-11-21:
Hi,
Indeed you will not be able to run two processes against one dongle because of locking and even if you would edit rtl* source to disable locking, it will not help because each time rtl* is executed they connect to dongle and open for listening with specific parameters and frequencies for current session. In case you would only listen to some sockets output than socat could help, but this is not the case. You can try to run one rtl_433 process and pass received data to two processes, one for wmbusmeters and other for rtl_433 for sensor data parsing, rough example - `rtl_433 -F csv -f 868.95M -w - | tee >(wmbusmeters rtl_433:CMD(rtl_433 -r -)) >(rtl_433 -F "mqtt://localhost:1883devices=sensors/rtl_433" -r -)`, I haven't tested this so this is just idea. If it does not work, second option you could try is to run rtl_433 and write output to file with -S or -W and then write second wrapper script that would read file and pass values wmbusmeters and/or rtl_433 for further processing. 
One more thing to consider - looking at cmd's you are running, rtl_433 by default listens to 433Mhz but rtl_wmbus to 868Hhz, so you need to understand at what frequencies your sensors and meters are transmitting, and if they are at different frequencies then you need to calculate intervals for data transmutations and configure main rtl_433 process to hop between frequencies to receive data from both sensors and meters. 
Carefully read manuals for both rtl_433 and wmbusmeters.

Delphius7, 2020-11-23:
Thanks for the detailed explanation and answers, even they are not the answers I was hoping for.

closed 2020-12-01 by weetmuts