wmbusmeters/wmbusmeters #188: contiiously losing connection with rtl-sdr

henkiejan1, 2020-11-28:
Hi,
I got contineusly errors like these:

`Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on t1
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on t1
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on t1
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on t1
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on t1
Lost 00000001 closing rtlwmbus[00000001]
Started auto rtlwmbus[00000001] listening on t1
`

Every 2 a 3 seconds. At the moment i did not use another rtl_sdr tool. Even when i start rtl_433 it works without losing the connection with the RTL_sdr stick. 

Does someone have some ideas??

weetmuts, 2020-11-28:
Run with --debug and post the log here.

henkiejan1, 2020-11-28:
`(wmbusmeters) version: 1.0.3-16-g357918a
(config) number of meters: 0
(main) device /dev/ttyACM0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyACM0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial /dev/ttyACM0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyACM0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyACM0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a) are you there? no.
(serialtty) closed /dev/ttyACM0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect rc1180)
(serial /dev/ttyACM0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyACM0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyACM0 (detect rc1180)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect cul)
(cul) get version
(serial /dev/ttyACM0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(cul) get version
(serial /dev/ttyACM0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(cul) get version
(serial /dev/ttyACM0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyACM0 (detect cul)
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyACM0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyS4 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyS4 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyS4 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyS4 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyS4) sent "FF0A02008077"
(serial /dev/ttyS4) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
[... 141 lines cut ...]
(serialcmd) closed /bin/sh pid=112550 fd=-1 (rtlwmbus)
(serial) EVENT thread interrupted
(main) rtlsdr device 00000001 not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus[00000001] listening on t1
(rtlwmbus) using command: rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serial) EVENT thread interrupted
(serialcmd) opened /bin/sh pid 112574 fd 7 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
^C(serial) stopping manager
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted rtlwmbus
(serial) closing 1 devices
(shell) sending SIGTERM to process group 112574
(bgshell) sent sigterm, now waiting for child 112574 to exit.
(bgshell) 112574 terminated due to signal 15
(serialcmd) closed /bin/sh pid=112574 fd=7 (rtlwmbus)
`

weetmuts, 2020-11-28:
It does not find rtl_wmbus in the path.
Have you installed it?

henkiejan1, 2020-11-28:
Oops that was not really clear for me. I found this version https://github.com/xaelsouth/rtl-wmbus and cloned and run ''make release'' but i have the idea that's not enough. Sorry for my maybe simple questions...

henkiejan1, 2020-11-28:
I got more luck with my Nanocul

eceived telegram from: 67270510
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0xc3)
                   ver: 0x39
                device: cul
                  rssi: 0 dBm
                driver: unknown!
Received telegram from: 67270558
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0xc3)
                   ver: 0x39
                device: cul
                  rssi: 0 dBm
                driver: unknown!
Received telegram from: 67270524
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0xc3)
                   ver: 0x39
                device: cul
                  rssi: 0 dBm
                driver: unknown!
Received telegram from: 67270513
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0xc3)
                   ver: 0x39
                device: cul
                  rssi: 0 dBm
                driver: unknown!

weetmuts, 2020-12-01:
Nice, that it works with the nanocul!

Clone this instead:
https://github.com/weetmuts/rtl-wmbus
make release
and then "sudo make install"
this will install rtl_wmbus into /usr/bin

weetmuts, 2020-12-31:
Did it work? Should I close this issue?

weetmuts, 2021-01-24:
Closing for now. Please reopen if problem persists.

closed 2021-01-24 by weetmuts