wmbusmeters/wmbusmeters #191: Driver unknown for Techem meters (heat allocators and water meters)

Marooned-MB, 2020-12-06:
Hi
In my flat every heater has attached Techem heat cost allocator (like [this one](https://www.techem.com/fileadmin/_processed_/5/d/csm_1740_FHKV_f631363c05.jpg)). I also have cold/hot water meters (like [this one](https://i.wpimg.pl/2000x0/filerepo.grupawp.pl/api/v1/display/embed/0cc2a880-cea9-4e9f-92af-a0aab9197761)).
I've run wmbusmeters for ~9h and gathered ~70 unique meters ID. Some of them catched only once, some of them over 100. I assume the more packed I got, the closer they are to my SDR (so more likely they belong to me).

For heat cost allocator I managed to find matched ID from display (last 4 digits) in my logs. For water there is no match but I found that ID printed on the meter is not the one send over the radio so that's problematic.

Anyway, 100% of catched telegrams rise an error `driver unknown! please consider opening an issue` so here I am.

As I have hundreds of grabbed telegrams, let me paste a small excerpt and I can post more if needed.
Here is for the heat meter:
```
Received telegram from: 03065716
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x80)
                   ver: 0x94
                device: rtlwmbus[00000001]
                  rssi: 39 dBm
                driver: unknown!
telegram=||33446850165706039480A20F9E28F2003018370011EA07D5070018001800160900000000000000000000000009190A22322C1215|+12
Received telegram from: 03065716
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x94
                device: rtlwmbus[00000001]
                  rssi: 78 dBm
                driver: unknown!
telegram=||294468501657060394087A4C001025A11E4EFF816C2F8D147ED0C1F959F55E0F0FF724805FCC33D19A19|+32
Received telegram from: 03065716
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x94
                device: rtlwmbus[00000001]
                  rssi: 59 dBm
                driver: unknown!
telegram=||294468501657060394087A4D001025432D9F0CCEA4F4230D9346F35865EEE40F0FF730A3B6D3D52C7730|+270
Received telegram from: 03065716
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x94
                device: rtlwmbus[00000001]
                  rssi: 47 dBm
                driver: unknown!
telegram=||294468501657060394087AA7001025F7F36E7F209FE41FD358018384F263AB0F0FF7685675183EF0D8D8|+21438
Received telegram from: 03065716
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x94
                device: rtlwmbus[00000001]
                  rssi: 56 dBm
                driver: unknown!
telegram=||294468501657060394087AA9001025E8B27AA7D06F2C7F398DEAF1877EED820F0FF773AFC52445069F4D|+21912
Received telegram from: 03065716
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x94
                device: rtlwmbus[00000001]
                  rssi: 67 dBm
                driver: unknown!
telegram=||294468501657060394087AAF001025C581533DA2E8518D2B4289A1504F1CB60F0FF70C37BB33ED8BB71C|+23337
Received telegram from: 03065716
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x94
                device: rtlwmbus[00000001]
                  rssi: 49 dBm
                driver: unknown!
telegram=||294468501657060394087AB10010256AABC01497883A1E7817D48BC72F3CAB0F0FF766E329FB1B0AEE1C|+2381
```

Here is for the warm water:
```
Received telegram from: 20655792
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Warm water (0x62)
                   ver: 0x70
                device: rtlwmbus[00000001]
                  rssi: 97 dBm
                driver: unknown!
telegram=||2E446850925765207062A0009E287C0030182C00000103030205020202030202050403050304050403020202030303|+21
Received telegram from: 20655792
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Warm water (0x62)
                   ver: 0x70
                device: rtlwmbus[00000001]
                  rssi: 98 dBm
                driver: unknown!
telegram=||2E446850925765207062A0009E287C0030182C00000103030205020202030202050403050304050403020202030303|+117
Received telegram from: 20655792
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Warm water (0x62)
                   ver: 0x70
                device: rtlwmbus[00000001]
                  rssi: 78 dBm
                driver: unknown!
telegram=||2E446850925765207062A0009E287C0030182C00000103030205020202030202050403050304050403020202030303|+321
```

and here is for the cold water:
```
Received telegram from: 20755140
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Cold water (0x72)
                   ver: 0x70
                device: rtlwmbus[00000001]
                  rssi: 79 dBm
                driver: unknown!
telegram=||2E446850405175207072A0009E28C20030182600000004040203030303020202030302040203040402020102020102|+7
Received telegram from: 20755140
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Cold water (0x72)
                   ver: 0x70
                device: rtlwmbus[00000001]
                  rssi: 88 dBm
                driver: unknown!
telegram=||2E446850405175207072A0009E28C20030182600000004040203030303020202030302040203040402020102020102|+100
Received telegram from: 20755140
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Cold water (0x72)
                   ver: 0x70
                device: rtlwmbus[00000001]
                  rssi: 97 dBm
                driver: unknown!
telegram=||2E446850405175207072A0009E28C20030182600000004040203030303020202030302040203040402020102020102|+169
```

I've tried different meters available in wmbusmeters, some of them printed some data for some of telegrams (even for one particular ID, so either no consistency or different types of telegrams sent by one device). I assume heat allocators might be encrypted (32 zeros didn't work) and I doubt I would get the key from the company. I'm more interested in water meters but couldn't crack that either.

Not sure if the data provided above would be enough to add support for those Techem devices. I have them available for testing so just ask for more data and I'll provide.
Thanks

MatthewGy, 2020-12-06:
Hi, 

I think I do have the same issue. The heatmeters are working fine (driver: fhkvdataiii) while I'm seeing also a lot of datagrams like this:

Received telegram from: 0099xxxx
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x80)
                   ver: 0x94
                device: im871a[00102456]
                  rssi: -88 dBm
                driver: unknown! 

I didn't see any telegram for the water meters (radio 4). There should be a dozen around me... I only found an old one (still radio 3). When do they send data? Only once every week/month?

Any help is appreciated!

THANK YOU!

weetmuts, 2020-12-08:
Thanks for the information! I have added the new version numbers that should be mapped to the
fhkvdataiii and the mkradio4 drivers! Pushed to master, please test!

Marooned-MB, 2020-12-08:
I did `git fetch/pull`, then `make` and `sudo make install`.
Type `x80` was recognized (although not properly read?):
```
(meter) zielony fhkvdataiii handling telegram from 03065716
(meter) zielony 03065716 "33446850165706039480A20F9E28F2003018370011EA07D5070018001800160900000000000000000000000009190A22322C1215"
(wmbus) parseDLL @0 52
(telegram) DLL L=33 C=44 (from meter SND_NR) M=5068 (TCH) A=03065716 VER=94 TYPE=80 (Heat Cost Allocator) (driver fhkvdataiii) DEV= RSSI=0
(wmbus) parseELL @10 42
(wmbus) parseAFL @10 42
(wmbus) parseTPL @10 42
(telegram) TPL CI=a2
telegram=|33446850165706039480A2|0F9E28F2003018370011EA07D5070018001800160900000000000000000000000009190A22322C1215|+1
(fhkvdataiii) 00: 33 length (51 bytes)
(fhkvdataiii) 01: 44 dll-c (from meter SND_NR)
(fhkvdataiii) 02: 6850 dll-mfct (TCH)
(fhkvdataiii) 04: 16570603 dll-id (03065716)
(fhkvdataiii) 08: 94 dll-version
(fhkvdataiii) 09: 80 dll-type (Heat Cost Allocator)
(fhkvdataiii) 0a: a2 tpl-ci-field (Mfct specific)
zielony 03065716        5632 hca        2020-12-01T02:00:00Z    6144 hca        2003-14-21T02:00:00Z    0.09 °C 0 °C    2020-12-08 12:00.38
```
however, type `x08` was not:
```
(meter) zielony: meter detection did not match the selected driver fhkvdataiii! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) zielony: yes for me
(meter) zielony fhkvdataiii handling telegram from 03065716
(meter) zielony 03065716 "294468501657060394087A4C001025A11E4EFF816C2F8D147ED0C1F959F55E0F0FF724805FCC33D19A19"
(wmbus) parseDLL @0 42
(telegram) DLL L=29 C=44 (from meter SND_NR) M=5068 (TCH) A=03065716 VER=94 TYPE=08 (Heat Cost Allocator) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(wmbus) decrypted content failed check, did you use the correct decryption key? Ignoring telegram.
(telegram) TPL CI=7a ACC=4c STS=00 CFG=2510 (synchronous AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
```

Water meeters seems to be working (need to find a way to guess which are mine and how to understand prev/curr values)
```
(meter) woda mkradio4 handling telegram from 20754986
(meter) woda 20754986 "2E446850864975207062A0009E28810120183C00000105050407040505030202040404050506060803050204040704"
(wmbus) parseDLL @0 47
(telegram) DLL L=2e C=44 (from meter SND_NR) M=5068 (TCH) A=20754986 VER=70 TYPE=62 (Warm water) (driver mkradio4) DEV= RSSI=0
(wmbus) parseELL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(telegram) TPL CI=a0
telegram=|2E446850864975207062A0|009E28810120183C00000105050407040505030202040404050506060803050204040704|+5
(mkradio4) 00: 2e length (46 bytes)
(mkradio4) 01: 44 dll-c (from meter SND_NR)
(mkradio4) 02: 6850 dll-mfct (TCH)
(mkradio4) 04: 86497520 dll-id (20754986)
(mkradio4) 08: 70 dll-version
(mkradio4) 09: 62 dll-type (Warm water)
(mkradio4) 0a: a0 tpl-ci-field (Mfct specific)
(mkradio4) 0e: * 8101 prev consumption (38.500000 m3)
(mkradio4) 12: * 3c00 curr consumption (6.000000 m3)
woda    20754986        44.5 m3 38.5 m3 2020-12-08 12:07.05
```

weetmuts, 2020-12-08:
Ok, added the 0x08 type as well. This is actually the real type for heat cost allocators,
but the other techem heat cost allocators use 0x80 which is non-standard. Perhaps
they have realized in later versions of their product to actually follow the standard,
or they simply made a bug, by flipping 0x80 vs 0x08.

Marooned-MB, 2020-12-08:
Not sure about that - as you can see above, it's exactly the same device (id) that sends 0x80 and 0x08 with different telegram structure. Different data?

weetmuts, 2020-12-08:
Oh no! That is weird. When I look at the telegrams, it seems like 0x80 can be properly decypted using your key?
But the 0x08 cannot be decrypted properly using the same key?

Marooned-MB, 2020-12-08:
Unfortunately I have no key (I used `NOKEY`) - I'll ask the company but I'm almost certain they will not provide it.

weetmuts, 2020-12-08:
Ok, so the device sends one un-encrypted message with 0x80 and one encrypted using 0x08....oh well.... 
wmbusmeters will throw away the encrypted telegram, since the decryption fails.......is not a big deal.

However, the warning will be logged....hmmm.....

weetmuts, 2021-01-24:
To deal with the weird problem that the meter sends one un-encrypted and one encrypted telegram,
a fix has been pushed to wmbusmeters, it only logs the warning that it failed to decrypt once.
Without this fix, the log would be full of warnings, since you want to listen to the un-encrypted telegrams.

Closing this for now. Please do not repoen. Open a new instead if the problem persists.

closed 2021-01-24 by weetmuts