wmbusmeters/wmbusmeters #192: Enable Multical603 in docker container

YOYONL89, 2020-12-07:
In addition to issue #165 where i already had some reply's i open this one.
I try to get my docker image of wmbus ready to read out my heat meater (Block heating).

@weetmuts already pushed the update to support the 603 to the image which i pulled today.
I removed the old docker image and made a new config with the following argument:
` --device=/dev/im871a:/dev/im871a`
The docker image has been set to not run in privileged mode cause it was conflicting with my domoticz docker.
the old image was ok where i saw the telegrams comming in but in this version it doesnt see the dongle.
`No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!`

Also the log in the docker environment reports:
`Not a valid meter type "multical603"`

for the moment i haven't been able to open a console/ssh session to the container.

weetmuts, 2020-12-07:
So, the multical603 is in the latest docker container. Or should be. :-)
Is there a way to check the version number of wmbusmeters that is inside the docker?
@BIBOLV ?

YOYONL89, 2020-12-07:
It certainly is different with a version prior to this one.
because the log in the docker environment did not report anything and now it does. :)

YOYONL89, 2020-12-07:
i have managed to get a console window on the docker.
in the /dev folder i see the im871a so its in the container and functioning i guess.

```
/ # cd /dev/
/dev # ls -l
total 0
lrwxrwxrwx    1 root     root            11 Dec  7 14:43 core -> /proc/kcore
lrwxrwxrwx    1 root     root            13 Dec  7 14:43 fd -> /proc/self/fd
crw-rw-rw-    1 root     root        1,   7 Dec  7 14:43 full
crw-rw----    1 root     996       188,   0 Dec  7 14:43 im871a
drwxrwxrwt    2 root     root            40 Dec  7 14:43 mqueue
crw-rw-rw-    1 root     root        1,   3 Dec  7 14:43 null
```
Output of wmbusmeters --listmeters
```

/wmbusmeters # wmbusmeters --listmeters
/bin/sh: wmbusmeters: not found
/wmbusmeters #

```

BIBOLV, 2020-12-07:
Looks like there is no way to tell from container what version is used if `wmbusmeters:latest` image is being used, only way I can think of is to check when image was pulled and compare it with latest version in hub, for example
```shell
~$ docker inspect wmbusmeters | grep Created
        "Created": "2020-12-05T12:32:04.546234315Z",
```
Proper way to find version is to run wmbusmeters inside container and it will print versions, like this
```shell
~$ docker exec -it wmbusmeters /wmbusmeters/wmbusmeters
wmbusmeters version: 1.0.4
........
```
Same way you can check meter list or run any other command
```shell
~$ docker exec -it wmbusmeters /wmbusmeters/wmbusmeters --listmeters | grep multical603
multical603    HeatMeter
``` 

If you would setup docker for first time I would suggest to check if docker user is in dialout group and things like that, but if you`re saying that you already had docker running and receiving messages then everything should be correct.

YOYONL89, 2020-12-07:
Here are the returned values:

```
docker inspect wMbus | grep Created
        "Created": "2020-12-07T13:43:56.83591225Z"
```

seems to be as expected.

Next one:

```
 docker exec -it wMbus /wmbusmeters/wmbusmeters
wmbusmeters version: 1.0.4
```
Looks fine as well.

```
 docker exec -it wMbus /wmbusmeters/wmbusmeters --listmeters | grep multical603
multical603    HeatMeter

```

No problem here either.
Still something is not working anymore ideas?

BIBOLV, 2020-12-07:
Try to set following values in `wmbusmeters.conf`
```
loglevel=debug
device=/dev/im871a
```
restart container and send log output

YOYONL89, 2020-12-08:
@BIBOLV that was the golden goose in getting the usb working.
it seems that the following line does not work any more:

`--device=/dev/im871a:/dev/im871a`

wmbusmeeters only searches for ttyxxxx devices as the log shows here:
```
(wmbusmeters) version: 1.0.4
(config) store meter files in: "/wmbusmeters_data/logs/meter_readings"
(config) number of meters: 0
(main) device /dev/ttyACM0 not currently used, detect contents...
(main) ignoring /dev/ttyACM0, it does not respond as any of the supported wmbus devices.
(serial) EVENT thread interrupted
(main) device /dev/ttyS0 not currently used, detect contents...
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyS0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB1 not currently used, detect contents...
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyUSB1, it does not respond as any of the supported wmbus devices.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 1
(shell) 9 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>6
<<<
(shell) return code 0
(shell) 10 exited
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) EVENT thread interrupted
```
changed the passing of devices to:

`--device=/dev/im871a:/dev/ttyUSB0`

and now it works like before.
Although it doesnt seem to recognise my 603 correctly:

```
(serial) received binary "A5C20333442D2C49822780350C8D20EC45900B21E634B0F4427434899F7D8C2BFEE2E0AA05C3DEFC3F8A79C10844FEC4FD0766545EF6396C4F47"
(im871a) checkIM871AFrame "A5C20333442D2C49822780350C8D20EC45900B21E634B0F4427434899F7D8C2BFEE2E0AA05C3DEFC3F8A79C10844FEC4FD0766545EF6396C4F47"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 108 (-68 dBm)
(im871a) got crc16 474f expected 474f
(im871a) received full frame
(wmbus) parseDLL @0 52
(telegram) DLL L=33 C=44 (from meter SND_NR) M=2c2d (KAM) A=80278249 VER=35 TYPE=0c (Heat volume at inlet meter) (driver unknown!) DEV=im871a[00102618] RSSI=-68
(wmbus) parseELL @10 42
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=ec SN=45900b21 (AES_CTR session=5 time=1095940) CRC=e634
Received telegram from: 80278249
          manufacturer: (KAM) Kamstrup Energi (0x2c2d)
                  type: Heat volume at inlet meter (0x0c)
                   ver: 0x35
                device: im871a[00102618]
                  rssi: -68 dBm
                driver: unknown!
(wmbus) 00: 33 length (51 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 2d2c dll-mfct (KAM)
(wmbus) 04: 49822780 dll-id (80278249)
(wmbus) 08: 35 dll-version
(wmbus) 09: 0c dll-type (Heat volume at inlet meter)
(wmbus) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(wmbus) 0b: 20 ell-cc (slow_resp sync)
(wmbus) 0c: ec ell-acc
(wmbus) 0d: 45900b21 sn (AES_CTR)
(wmbus) 11: e634 payload crc (calculated b84f ERROR)
telegram=||33442D2C49822780350C8D20EC45900B21E634B0F4427434899F7D8C2BFEE2E0AA05C3DEFC3F8A79C10844FEC4FD0766545EF639|+50
```
i hope this will help @weetmuts in pointing me in a good direction. :)

weetmuts, 2020-12-08:
Thanks @BIBOLV for the help with the docker problems!

@YOYONL89 , it seems like this meter sends a different type/version than the multical603 already implemented.
So its not the same kind of meter, at least on the inside. Do you have a configuration number exact model for this meter?

To solve this, please collect a bunch of telegrams, properly decrypted and using the multical603 driver with debug information, then email them to me. This will collect the contents of the telegrams so that we can compare with
the existing multical603 driver, to see if we should create a new driver or just upgrade the existing one.

YOYONL89, 2020-12-12:
@weetmuts I have send you an email with the requested info/logs.

YOYONL89, 2020-12-12:
@BIBOLV 
Perhaps you could help me with a other kind of problem i have with MQTT.
it seems that i can get my data send to Domoticz (Sensor receives an update) but it doesn't contain a value.
When i use Node-Red to check the message i see the following:
`{"idx":248,"svalue":""}`
So it seems no svalue is been added to the mqtt message.

The command that im using:
`shell=/usr/bin/mosquitto_pub -h xxx.xxx.xxx.xxx -t domoticz/in -m "{\"idx\":246,\"svalue\":\"$VOLUME_FLOW_M3H\"}"`

Is it because my meter doesn't have a known driver?

BIBOLV, 2020-12-12:
Yes, values needs to be extracted from message in order to send them anywhere, so if there is no driver then values are not extracted un and you don`t have anything to send.

YOYONL89, 2020-12-12:
That sounds fair.
Just curious cause although the driver reports unknown I get a reading file with a json string in it.
Even in the debug the env variables are filled with the multical603 driver.
Some correctly filled some scrambled.

Where does the mqtt gets its data from then?

BIBOLV, 2020-12-12:
Well if some value is extracted correctly then firstly list the shell env variables available for a meter by running 
```shell
~$ docker exec -it wmbusmeters /wmbusmeters/wmbusmeters --listenvs=multical603
```
I guess variable you are looking for is METER_MAX_FLOW_M3H, change in config, restart container and see if Domoticz now receives value.

YOYONL89, 2020-12-13:
@BIBOLV,

Seems you're right again!
Many thanks although I have to mention that the domotica manual is somewhat off then.
In that manual/wiki its stated to use the command I showed you. Maybe it would be a good idea to update the manual?

BIBOLV, 2020-12-13:
Thanks for letting know, but could you please point out what manual/wiki you used? 
I checked [this](https://weetmuts.github.io/wmbusmeterswiki/DOMOTICZ.html) where it is stated to use METER_TOTAL_M3 that is actual meter reading. Any of supported meter env variables can be used there if user wants to and command to get meter env variables is stated in main wmbusmeter documentation.

YOYONL89, 2020-12-13:
Oh dear... Seems you're right and I have misread the documentation. My apologies! :$

YOYONL89, 2020-12-29:
@weetmuts 

Today i had a mechanic from the energy company at home and we have managed to get the detailed metercode along with the wireless mbus module and telegram.

Metercode:                3-2-407-227-00-24-24-1-0-0-95-10-3-0132 
Translation to config: A-B-CCC-DDD-EE-FF-GG-L-M-N-PP-RR-T-VVVV
English page with configuration is added

Communication:       31-11-111
Translation to config: XX-YY-ZZZ
English page with configuration is added
[Meterconfig.pdf](https://github.com/weetmuts/wmbusmeters/files/5751224/Meterconfig.pdf)
[image]

YOYONL89, 2021-01-19:
Manouvering on my own i start to learn how the telegram is being used and i think i know what the changes need to be.
Although im confined to the use of the docker which does not make it easy to modify the driver.

The telegram looks like this:
`(multical603) 00: 33 length (51 bytes)
(multical603) 01: 44 dll-c (from meter SND_NR)
(multical603) 02: 2d2c dll-mfct (KAM)
(multical603) 04: 49822780 dll-id (xxxxxxxxx)
(multical603) 08: 35 dll-version
(multical603) 09: 0c dll-type (Heat volume at inlet meter)
(multical603) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical603) 0b: 20 ell-cc (slow_resp sync)
(multical603) 0c: 58 ell-acc
(multical603) 0d: axxxxxxx sn (AES_CTR)
(multical603) 11: 033e payload crc (calculated 033e OK)
(multical603) 13: 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(multical603) 14: 80da format signature
(multical603) 16: 8db8 data crc
(multical603) 18: 0FE00000
(multical603) 1c: * C9840300 energy forward kwh (230601)
(multical603) 20: * AD5F0100 energy returned kwh (90029)
(multical603) 24: * B7E00400 total volume (319.671000 m3)
(multical603) 28: * 9B000000 volume flow (0.155000 m3/h)
(multical603) 2c: * 661C T1 flow temperature (72.700000 °C)
(multical603) 2e: * CF07 T2 flow temperature (19.990000 °C)
(multical603) 30: * 00000000 info codes ()`

Where:
`(multical603) 18: 0FE00000`
is the total Heat Energy in Gj (Managed to translate this code and got the correct value out of it)

`(multical603) 1c: * C9840300 energy forward kwh (230601)
(multical603) 20: * AD5F0100 energy returned kwh (90029)`
Will have to be Gj too

`(multical603) 28: * 9B000000 volume flow (0.155000 m3/h)`
Supposed to be L/h but seems to be converted correctly

The next question is how to get my meter properly supported with the last unknown value?
I cant seem to run any tests using docker as i cant reach the files in the docker.

BIBOLV, 2021-01-20:
If you're making some development then docker is definitely not a platform to use, you should make local installation from source. Docker is for day-to-day usage to make it more easily install/upgrade etc. 
When you have implemented changes locally, you can push update here and when @weetmuts will accept, it will be included in docker and you will be able to use it.

weetmuts, 2021-06-20:
Yeah, develop using plain linux. Much much much simpler than mucking around inside a docker container.
Good luck with the development.

closed 2021-06-20 by weetmuts