wmbusmeters/wmbusmeters #193: Support for Hydrocal M3 by BMeters

balczezzz, 2020-12-07:
Hi,
As per the title I would like to add support for the HYDROCAL M3 Heat meter by BMeters. I've copied the meter_vario451.cc file and done most from the _HowToAddaNewMeter.txt_ tutorial. Edited all the files, recompiled using the DEBUG option and I am decoding the telegram in a JSON format as below:

{"media":"heat/cooling load","meter":"hydrocalm3","name":"Heat","id":"02364971","total_kwh":1239.166667,"current_kwh":1239.166667,"previous_kwh":0,"timestamp":"2020-12-07T18:42:11Z"}

It's a brand new meter, so the JSON values should be 0s. Would appreciate help in guiding on how the processContent should look like, so that proper values are displayed. Attaching the initial simulation result as I guess it holds the key to my struggles:

(wmbusmeters) version: 1.0.3-13-gff8fb1b
(config) using device: simulation_mymeter.txt:any 
(config) number of meters: 1
(main) configured "Wasser" "supercom587" "02364971" encrypted
(lookup) with file "simulation_mymeter.txt"
(lookup) driver: simulation file
(main) opening simulation_mymeter.txt:any
Started config simulation on simulation_mymeter.txt listening on any
(simulation) in simulation_mymeter.txt
(main) regular reset of simulation  on simulation_mymeter.txt will happen every 82800 seconds
(main) added simulation_mymeter.txt to files
(wmbus) no alarm (expected activity) for simulation
(main) simulation_mymeter.txt already configured as simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(simulation) from file "8E44B409714936020B0D7AB60080053C8D63A0541967F884BEF5E7763BE83C77BB4D10122253A1E8C2E6776685E30FF439CD832A0ED18237834F62045D33690340D6B74E87BB2AFA6E5AEBB2D6C3C34FFD4EEEE10A025E5D121089F83FDF590D7E0DF31895E5FA82F125993B997F852197D46FC43047F48762B5764C49BAE8D43F33909B0EBB399BACC0637A2F99F1"
(wmbus) parseDLL @0 143
(meter) Wasser: for me? 02364971
(meter) Wasser: meter detection did not match the selected driver supercom587! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Wasser: yes for me
(meter) Wasser supercom587 handling telegram from 02364971
(meter) Wasser 02364971 "8E44B409714936020B0D7AB60080053C8D63A0541967F884BEF5E7763BE83C77BB4D10122253A1E8C2E6776685E30FF439CD832A0ED18237834F62045D33690340D6B74E87BB2AFA6E5AEBB2D6C3C34FFD4EEEE10A025E5D121089F83FDF590D7E0DF31895E5FA82F125993B997F852197D46FC43047F48762B5764C49BAE8D43F33909B0EBB399BACC0637A2F99F1"
(wmbus) parseDLL @0 143
(telegram) DLL L=8e C=44 (from meter SND_NR) M=09b4 (BMT) A=02364971 VER=0b TYPE=0d (Heat/Cooling load meter) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 133
(wmbus) parseAFL @10 133
(wmbus) parseTPL @10 133
(TPL) decrypting "3C8D63A0541967F884BEF5E7763BE83C77BB4D10122253A1E8C2E6776685E30FF439CD832A0ED18237834F62045D33690340D6B74E87BB2AFA6E5AEBB2D6C3C34FFD4EEEE10A025E5D121089F83FDF590D7E0DF31895E5FA82F125993B997F852197D46FC43047F48762B5764C49BAE8D43F33909B0EBB399BACC0637A2F99F1"
(TPL) num encrypted blocks 8 (128 bytes and remaining unencrypted 0 bytes)
(TPL) IV B409714936020B0DB6B6B6B6B6B6B6B6
(TPL) decrypted  "2F2F0C0E00000000046D112D872C0C13000000000C0E000000000C13000000000C13000000000C13000000000A5A94000A5E95000F820000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002F2F"
(dvparser) found new format "0C0E046D0C130C0E0C130C130C130A5A0A5E" with hash 702b, remembering!
(telegram) TPL CI=7a ACC=b6 STS=00 CFG=0580 (AES_CBC_IV nb=8 cntn=0 ra=0 hc=0)
telegram=|8E44B409714936020B0D7AB60080052F2F|0C0E00000000046D112D872C0C13000000000C0E000000000C13000000000C13000000000C13000000000A5A94000A5E95000F820000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002F2F|+0
(supercom587) 00: 8e length (142 bytes)
(supercom587) 01: 44 dll-c (from meter SND_NR)
(supercom587) 02: b409 dll-mfct (BMT)
(supercom587) 04: 71493602 dll-id (02364971)
(supercom587) 08: 0b dll-version
(supercom587) 09: 0d dll-type (Heat/Cooling load meter)
(supercom587) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(supercom587) 0b: b6 tpl-acc-field
(supercom587) 0c: 00 tpl-sts-field
(supercom587) 0d: 8005 tpl-cfg 0580 (AES_CBC_IV nb=8 cntn=0 ra=0 hc=0 )
(supercom587) 0f: 2f2f decrypt check bytes
(supercom587) 11: 0C dif (8 digit BCD Instantaneous value)
(supercom587) 12: 0E vif (Energy MJ)
(supercom587) 13: 00000000
(supercom587) 17: 04 dif (32 Bit Integer/Binary Instantaneous value)
(supercom587) 18: 6D vif (Date and time type)
(supercom587) 19: 112D872C
(supercom587) 1d: 0C dif (8 digit BCD Instantaneous value)
(supercom587) 1e: 13 vif (Volume l)
(supercom587) 1f: * 00000000 total consumption (0.000000 m3)
(supercom587) 23: 0C dif (8 digit BCD Instantaneous value)
(supercom587) 24: 0E vif (Energy MJ)
(supercom587) 25: 00000000
(supercom587) 29: 0C dif (8 digit BCD Instantaneous value)
(supercom587) 2a: 13 vif (Volume l)
(supercom587) 2b: 00000000
(supercom587) 2f: 0C dif (8 digit BCD Instantaneous value)
(supercom587) 30: 13 vif (Volume l)
(supercom587) 31: 00000000
(supercom587) 35: 0C dif (8 digit BCD Instantaneous value)
(supercom587) 36: 13 vif (Volume l)
(supercom587) 37: 00000000
(supercom587) 3b: 0A dif (4 digit BCD Instantaneous value)
(supercom587) 3c: 5A vif (Flow temperature 10⁻¹ °C)
(supercom587) 3d: 9400
(supercom587) 3f: 0A dif (4 digit BCD Instantaneous value)
(supercom587) 40: 5E vif (Return temperature 10⁻¹ °C)
(supercom587) 41: 9500
(supercom587) 43: 0F manufacturer specific data 820000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002F2F
Wasser	02364971	0 m3	2020-12-07 18:00.29
(serial) stopping manager
(serial) waiting for stop
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted simulation

balczezzz, 2020-12-07:
Copying the multical603.cc produces somewhat closer-to-reality results, but still not perfect.

{"media":"heat/cooling load","meter":"hydrocalm3","name":"Heat","id":"02364971","total_energy_consumption_kwh":0,"total_volume_m3":0,"volume_flow_m3h":0,"t1_temperature_c":9.4,"t2_temperature_c":127,"at_date":"","current_status":"","energy_forward_kwh":"0","energy_returned_kwh":"0","timestamp":"2020-12-07T20:41:33Z"}

balczezzz, 2020-12-09:
Created pull requests #195-198 to accommodate changes. Hope this is fine.

weetmuts, 2020-12-10:
Thanks I pulled your code. There are some things that needs to be fixed. 
I will fix the later this week!

balczezzz, 2020-12-17:
We can close the issue. Already created a pulled request for proper parsing for this heat meter. It is available as meter_hydrocalm3.cc

closed 2020-12-17 by balczezzz