wmbusmeters/wmbusmeters #199: Diehl Hydrus watermeter 2. attempt

superfino, 2020-12-10:
Hello,
it's me again. I got a new water meter and would like to use wmbusmeters again. I hope you will allow me one more try. With the new watermeter things are looking differend than earlier this year, don't know if the meter or your new wmbusmeters caused this.
l've a plain terminal log, the standard log and a debug log available. I find my telegrams (ID 00941205) but wmbusmeters seem to have problems with them. Hope you can have a look on this. Thanks in advance ... Jo

[terminal.log](https://github.com/weetmuts/wmbusmeters/files/5671928/terminal.log)
[wmbusmeters.log](https://github.com/weetmuts/wmbusmeters/files/5671930/wmbusmeters.log)
[debug.log](https://github.com/weetmuts/wmbusmeters/files/5671933/debug.log)

superfino, 2020-12-17:
Hi, i have some additional information. I got the encryption key from my watersupplier today. It's a DES key which apparently means OMS3 Mode 5 is implemented. Hope this is helpful.

superfino, 2020-12-18:
last night i tried again to find the problem here. In my cluelessness i reflashed my cul for any reason and voila, one of my problems was solved. The telegrams are recogniced as Hydrometer but the encryption still fails. Regarding the encryption mode i found this OMS doc which i think is my problem because my DES encryption key is only 8 bytes. 
[OMS-Spec_Vol2_Primary_v402.pdf](https://github.com/weetmuts/wmbusmeters/files/5716189/OMS-Spec_Vol2_Primary_v402.pdf)

weetmuts, 2021-01-01:
If the meter uses the deprecated DES mode, then you need to email me a debug line: telegram=|...|
for your meter and your meter key. Then I can implement the missing decryption mode.

superfino, 2021-01-02:
Thanks for your offer. I'll send you the requested info as soon as i'm at home again. Hopefully next week.

zoesn89, 2021-01-21:
Hey, 
I guess, I have a similar problem. My DIEHL Hydrus watermeter is a Mbus meter attached to an wMbus sender named IZAR RADIO EXTERN 686 MBUS. 
When I try `wmbusmeters /dev/ttyUSB0:im871a:t1`, I get:
                          
Started config im871a[00101717] on /dev/ttyUSB0 listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 81100028
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Bus/System component (0x0e)
                   ver: 0x64
                device: im871a[00101717]
                  rssi: -66 dBm
                driver: unknown!

Here is the logged telegram:

telegram=||6644242328001081640E7266567464A51170077A0050050DB829C269937A126D052542911F07CDAD7DB2E1C2FF8BB71CFA54A5BD5CFB84644EEEBEBCAEBC594FCC0D22358A1C4B2AB515483B68A982907CC31DFEBA30B1EF1E4EEF58DF944653A4D8C5A3A8F2F3|+1

weetmuts, 2021-01-24:
Did you have a key?

superfino, 2021-01-24:
Yep, it's underway.

weetmuts, 2021-02-21:
Closing this for now. Please re-open when you have the info.

superfino, 2021-02-21:
Very sad that you closed this issue. I've send you the requested info twice times to your gmail account in January because i would not present the DES key here. I've also send some telegrams i captured. Unfortunately I've not found a way to reopen this issue. I think you only can do this.

weetmuts, 2021-02-21:
Sorry! My email account is full and messy! I will take a look!

weetmuts, 2021-02-21:
Found it!

weetmuts, 2021-02-21:
You are lucky! This was not a deprecated DES encryption, this was the already implemented
proprietary PRIOS encryption. I have just pushed a fix that will accept the PRIOS key for hydrus meters
as well as izar meters. Your telegram decodes nicely to: 96.093 m3.

superfino, 2021-02-21:
This sounds great. Thank you for your efforts. I'll check it out soon.

superfino, 2021-02-22:
Just tested your fix and was able to decode the telegrams. Marvelous! Unfortunately it worked only manualy because the daemon started not anymore. I've checked if everything is wrong with my installation but couldn't find something. The status shows like this:
systemctl status wmbusmeters
● wmbusmeters.service - "wmbusmeters service"
   Loaded: loaded (/lib/systemd/system/wmbusmeters.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-02-22 11:05:35 CET; 18s ago
     Docs: https://github.com/weetmuts/wmbusmeters
           man:wmbusmeters(1)
  Process: 564 ExecStartPre=/bin/mkdir -p /var/log/wmbusmeters/meter_readings (code=exited, st
  Process: 565 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /var/log/wmbusmeters (code=e
  Process: 566 ExecStartPre=/bin/mkdir -p /run/wmbusmeters (code=exited, status=0/SUCCESS)
  Process: 567 ExecStartPre=/bin/chown -R wmbusmeters:wmbusmeters /run/wmbusmeters (code=exite
  Process: 568 ExecStart=/usr/sbin/wmbusmetersd /run/wmbusmeters/wmbusmeters.pid (code=exited,
 Main PID: 569 (code=exited, status=1/FAILURE)

Feb 22 11:05:35 WMBus systemd[1]: wmbusmeters.service: Service RestartSec=1s expired, scheduli
Feb 22 11:05:35 WMBus systemd[1]: wmbusmeters.service: Scheduled restart job, restart counter
Feb 22 11:05:35 WMBus systemd[1]: Stopped "wmbusmeters service".
Feb 22 11:05:35 WMBus systemd[1]: wmbusmeters.service: Start request repeated too quickly.
Feb 22 11:05:35 WMBus systemd[1]: wmbusmeters.service: Failed with result 'exit-code'.
Feb 22 11:05:35 WMBus systemd[1]: Failed to start "wmbusmeters service".

superfino, 2021-02-23:
Everything O.K. now.  Found out that the daemon doesn't accept device=auto:t1 in the wmbusmeters.conf. Changed it and everything run's like a charm. Most commonly, the the problem remains behind the keyboard. Thanks a lot for your support, i think now i will close this case.

closed 2021-02-23 by superfino