wmbusmeters/wmbusmeters #200: Adding support for Techem Smoke detector 2

Kosygor, 2020-12-10:
Hello All

I have this Techem Smoke detector 2 or (Funk-rauchwarnmelder 2) and I am trying to make use of it in wmbusmeters.
I have read issue #165 and [https://github.com/weetmuts/wmbusmeters/blob/master/HowToAddaNewMeter.txt](How to add manual) but I am willing to do testing but need some guidance.

I have manage to receive that kind of message:
```Received telegram from: 91633569
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Unknown (0xf0)
                   ver: 0x76
                device: rtlwmbus[00000001]
                  rssi: 76 dBm
                driver: unknown!
telegram=||294468506935639176F0A0009F2789290060892900000501D7311AF93E1BF93E008DC3009ED4000FE500|+282
```

Ok so I have listened for few messages and copy paste them to tht file like this: 
```
telegram=||294468506935639176F0A0009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500|
telegram=||294468506935639176F0A0009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500|
telegram=||294468506935639176F0A0009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500|
```

from this file I got:
```
wmbusmeters --debug simulation_dymr.txt Wasser lansensm 91633569 NOKEY

(simulation) from file "294468506935639176F0A0009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500"
(wmbus) parseDLL @0 42
(meter) Wasser: for me? 91633569
(meter) Wasser: meter detection did not match the selected driver lansensm! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) Wasser: yes for me
(meter) Wasser lansensm handling telegram from 91633569
(meter) Wasser 91633569 "294468506935639176F0A0009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500"
(wmbus) parseDLL @0 42
(telegram) DLL L=29 C=44 (from meter SND_NR) M=5068 (TCH) A=91633569 VER=76 TYPE=f0 (Unknown) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(telegram) TPL CI=a0
telegram=|294468506935639176F0A0|009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500|+0
(dvparser) warning: cannot extract uint16 from non-existant key "02FD971D"
(wmbus) warning: cannot find offset -1 to add more explanation " info codes (OK)"
(lansensm) 00: 29 length (41 bytes)
(lansensm) 01: 44 dll-c (from meter SND_NR)
(lansensm) 02: 6850 dll-mfct (TCH)
(lansensm) 04: 69356391 dll-id (91633569)
(lansensm) 08: 76 dll-version
(lansensm) 09: f0 dll-type (Unknown)
(lansensm) 0a: a0 tpl-ci-field (Mfct specific)
Wasser	91633569	OK	2020-12-10 22:53.50
(serial) stopping manager
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) deleted simulation
```
I have also tried supercom587 with similar effect. Basically this is moment that I`m stuck with.
So what I understand is my device is not similar enough to others to make it easy and wmbusmeters it is trying to read only first 5 bytes. And for that I have to tell  it which bytes represent what.

So from [fhem](https://forum.fhem.de/index.php?topic=18422.90) forum I have some partially de encrypted data:
(I have added some translations). 
[Techem Funkwarnrauchmelder 2 (1).xlsx](https://github.com/weetmuts/wmbusmeters/files/5675931/Techem.Funkwarnrauchmelder.2.1.xlsx)

Also i have file which is (is it ?) translating messages for fhem system.
[32_TechemSD.txt](https://github.com/weetmuts/wmbusmeters/files/5675935/32_TechemSD.txt).

Probably there are some more information because it has 3x ultra-sound proximity sensors and I hope some thermometer but I will leave that after getting my head around getting fire alarm first.

So i guess I should do this part :
```
Some meters use vendor proprietary fields, typically error codes,
those are found with:
    extractDVuint24(&t->values, "03FD17", &offset, &info_codes_);
    t->addMoreExplanation(offset, " info codes (%s)", status().c_str());
```
But I am just at start with learning programming and i have chosen python and i assume it is C++ ?

PS anyone has any learning materials which will allow me to use [communication documentation](https://www.diehl.com/cms/files/254374-FR-EN-RAY_RADIO_CommunicationDescription_v04_EN.pdf?download=1) communications with bit bytes frames etc ? I have few questions and if there are no straight answers in documentation I feel like this has to be academic level knowledge that I am lacking.

weetmuts, 2020-12-11:
Good start! 

Alas, the log line "0a: a0 tpl-ci-field (Mfct specific)" means, all data that comes after this is secret proprietary shit, not compliant with the wmbus standard. I have not explained this in the HowTo yet.... The howto is focused on meters that
are more or less OMS-compliant. Techem meters are often very naughty....

But if you write in you new meter code:

void XXXXXX::processContent(Telegram *t)
{
    vector<uchar> frame;
    t->extractFrame(&frame);

Now  you have the raw telegram bytes available in the vector frame and you can start reverse-engineering the protocol
and apply the work that fhem as already done. 

Check the meter_rfmtx1.cc for a simple meter driver that can handle proprietary water data for old meters.

weetmuts, 2020-12-28:
Hows it going?

Kosygor, 2020-12-29:
Sorry for taking so long.

Steps taken:
Copied `meter_supercomp586.cc`
Removed everything after `namespace std;’ since (at least I think so) these lines are responsible for using libraries to interpret meter readings.
Pasted Your code with new meter name (MeterTSD2) added indentations and one curly bracket o it looks like this :
```
using namespace std;

void MeterTSD2::processContent(Telegram *t)
{
    vector frame;
    t->extractFrame(&frame);
}
```
Copied it to ../wmbusmeters/src
Then according to HowToAddNewMeter.txt I have added
```
    X(tsd2,       T1_bit, SmokeDetector,        FHKVDATAIII,    FHKVDataIII)     \
```
And did not add anything to Meter_detection block since this is part of searching.
Added to makefile: 
```
$(BUILD)/meter_tsd2.o \
```
Got bunch of errors in while compiling meters.h about redeclaration., so changed it to :
```
    X(tsd2,       T1_bit, SmokeDetector,        TSD2,    TSD2)     \
```
Then while compiling with DEBUG=true:
```
src/meter_tsd2.cc:27:6: error: ‘MeterTSD2’ has not been declared
 void MeterTSD2::processContent(Telegram *t)
      ^~~~~~~~~
src/meter_tsd2.cc: In function ‘void processContent(Telegram*)’:
src/meter_tsd2.cc:29:12: error: missing template arguments before ‘frame’
     vector frame;
            ^~~~~
src/meter_tsd2.cc:30:22: error: ‘frame’ was not declared in this scope
     t->extractFrame(&frame);
                      ^~~~~
src/meter_tsd2.cc:30:22: note: suggested alternative: ‘trace’
     t->extractFrame(&frame);
                      ^~~~~
                      trace
make: *** [Makefile:112: build_debug/meter_tsd2.o] Błąd 1
```
So I think I have removed too much from `meter_tsd2.cc`.
And now I`m stuck because all that I have delated seemed to be responsible for interpreting frame. Only idea I had was to change every “Supercom587” phrase to “TSD2” but from I see/guess part of them directs to libraries so it would be bad idea.

I know that this is ain`t much but I just manage to get acquainted with C++ in very limited matter. 
If I am getting this right Beginning of the frame is interpreted correctly because that what’s giving the meter number manufacturer  etc. , but after that because frame is not completely compliant it refuses to interpret data. Can`t I somehow just allow meter code to run further in itself since data bits (A000/A0001) which are equivalent for making smoke signal Turn up just one value.

Hope I did not stretched Your patience yet.

weetmuts, 2020-12-31:
Great work! The error messages from c++ can be a bit intimidating, even though the error is probably quite simple.

So I redid your work and added a stub for the tsd2 driver and pushed it to master.
Then I took a quick look at the links to the spreadsheet and the code.
It seems like the first byte after 0xa0 is 0 when all is ok and 1 when there is an alarm.
So I added a trivial test for this. Check meter_tsd2.cc

Please try to add all the other fields that seems to be understood already!

Kosygor, 2021-01-01:
Looks like it works like a charm. 

When there is no smoke it gives :
```
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2021-01-01 21:59:12.000;64;179;91633569;0x294468506935639176f0a0009f299e2900889e2900801001e3311af93e1bf93e008dc3009ed4000fe500<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 42
(meter) dymsalon: for me? 91633569
(meter) dymsalon: yes for me
(meter) dymsalon tsd2 handling telegram from 91633569
(meter) dymsalon 91633569 "294468506935639176F0A0009F299E2900889E2900801001E3311AF93E1BF93E008DC3009ED4000FE500"
(wmbus) parseDLL @0 42
(telegram) DLL L=29 C=44 (from meter SND_NR) M=5068 (TCH) A=91633569 VER=76 TYPE=f0 (Smoke detector) (driver tsd2) DEV=rtlwmbus[00000001] RSSI=64
(wmbus) parseELL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(telegram) TPL CI=a0
telegram=|294468506935639176F0A0|009F299E2900889E2900801001E3311AF93E1BF93E008DC3009ED4000FE500|+108
(tsd2) 00: 29 length (41 bytes)
(tsd2) 01: 44 dll-c (from meter SND_NR)
(tsd2) 02: 6850 dll-mfct (TCH)
(tsd2) 04: 69356391 dll-id (91633569)
(tsd2) 08: 76 dll-version
(tsd2) 09: f0 dll-type (Smoke detector)
(tsd2) 0a: a0 tpl-ci-field (Mfct specific)
dymsalon	91633569	OK	2021-01-01 22:59.12
```

When there is smoke: 
```
rtlwmbus) checkRTLWMBusFrame "T1;1;1;2021-01-01 22:00:54.000;70;135;91633569;0x294468506935639176f0a0019f299e2900889e2901801002e3312a9f441bf93e008dc3009ed4000fe500<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 42
(meter) dymsalon: for me? 91633569
(meter) dymsalon: yes for me
(meter) dymsalon tsd2 handling telegram from 91633569
(meter) dymsalon 91633569 "294468506935639176F0A0019F299E2900889E2901801002E3312A9F441BF93E008DC3009ED4000FE500"
(wmbus) parseDLL @0 42
(telegram) DLL L=29 C=44 (from meter SND_NR) M=5068 (TCH) A=91633569 VER=76 TYPE=f0 (Smoke detector) (driver tsd2) DEV=rtlwmbus[00000001] RSSI=70
(wmbus) parseELL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(telegram) TPL CI=a0
telegram=|294468506935639176F0A0|019F299E2900889E2901801002E3312A9F441BF93E008DC3009ED4000FE500|+210
(tsd2) 00: 29 length (41 bytes)
(tsd2) 01: 44 dll-c (from meter SND_NR)
(tsd2) 02: 6850 dll-mfct (TCH)
(tsd2) 04: 69356391 dll-id (91633569)
(tsd2) 08: 76 dll-version
(tsd2) 09: f0 dll-type (Smoke detector)
(tsd2) 0a: a0 tpl-ci-field (Mfct specific)
dymsalon	91633569	SMOKE	2021-01-01 23:00.54

```

Basic function is working I will try to find time to decrypt at least battery information for safety reasons.
Big thanks for Your support.
I wish You to have great year.

Kosygor, 2021-01-06:
I am trying to decode other parts.
From fact that every telegram ends with |+XXX and what have You wrote:
> It seems like the first byte after 0xa0 is 0 when all is ok and 1 when there is an alarm.
I conclude that headers and all frame stuff is pretty big if it is DEC160  and in telegram that is printed in wmbusmeters it is 24`th value (0 or 1).
Your manual is only mentioning of that in three cases:
- using /dev/ttyUSB0:38400 (my device is mounting as /dev/swradio0 (also lirc0 and dvb) a and does not work with or without setting baud rate.
- reading from stdin - that i just cant get my head around how to set it up
- reading from file - which i cannot find. Only telegrams.bin file from wmbusmeters/fuzz_testcases/telegrams is not updated during normal work. I assume that --logtelegrams is saveing only decrypted telegrams.
I have feeling that it is once again lack of basic knowledge, and asking for help.

weetmuts, 2021-01-06:
I am not sure I understand your question. The tsd2 wmbus headers are standard, so wmbusmeters uses the
normal code to understand them, it can decode everything up to the 0xA0 tpl-ci-field.

The best way for you to test again and again on the same telegram is to put lines like these (do not add the +123 at the end that means that wmbusmeter should delay 123 seconds before simulating this telegram):
telegram=|294468506935639176F0A0|009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500|
into a file called for example: simulation_tsd2.txt or simulation_work or simulation_smoke.txt as long as the filename starts with simulation_ then you just run:

wmbusmeters simulation_smoke.txt MySomeDetector tsd2 91633569 NOKEY
and it will decode the telegram, as if it had just arrived from a dongle or otherwise.

Run with --debug 
wmbusmeters --debug simulation_smoke.txt MySomeDetector tsd2 91633569 NOKEY
to see more of the byte analysis of the wmbus-compliant part of the protocol. But it does not really help you that much since the data you are interested in is a proprietary blob.

The extractPayload grabs the data after the 0xA0, that is why the processContent is so simple,
it just does: info_codes_ = data[0];
:-)
You do not need to know the size of the header.
I mentioned the 0xA0 since the fhem documents you referred to, does place the 0xA0 first
in their docs, which makes it easy to find the right offsets to use.

Kosygor, 2021-01-07:
Ok so to make this straight: I should focus on telegram itself and should not mind any other part that would be in raw frame.
 I will give this go although it is as You have wrote "proprietary blob": eg: try to drain battery and change temperatures. 

What should I do if battery and temp data will not be in wmbus-compliant part of the protocol?
Is there any sense trying to decode that or it will be failure before start since wmbusmeters will not be able to handle it?

weetmuts, 2021-01-24:
Run this command:
wmbusmeters --debug simulation_tsd2.txt Smoke tsd2 91633569 NOKEY

Check the output:
...
..
.
telegram=|294468506935639176F0A0|009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500|+0
(tsd2) 00: 29 length (41 bytes)
(tsd2) 01: 44 dll-c (from meter SND_NR)
(tsd2) 02: 6850 dll-mfct (TCH)
(tsd2) 04: 69356391 dll-id (91633569)
(tsd2) 08: 76 dll-version
(tsd2) 09: f0 dll-type (Smoke detector)
(tsd2) 0a: a0 tpl-ci-field (Mfct specific)
Smoke	91633569	OK	2021-01-24 20:25.36

This is the header: |294468506935639176F0A0| and has been fully decoded nothing more to see here.

The remainder is the techem non-wmbus-compliant binary data which other people have decoded already.
009F2782290060822900000401D6311AF93E1BF93E008DC3009ED4000FE500

The first byte, accessed using:   info_codes_ = data[0];
in meter_tsd2.cc load the first byte, which is 00 which means no smoke.

Use this knowledge and your other tables to add/decode the remainder of the telegram.

Kosygor, 2021-01-26:
Finally got some time to interpret data. 
Bad side is that I got annoyed by how slow LED drained battery and I just cut it off then made adjustable power supply to check if there is any kind of warning in telegrams . Sensor seems to be working but it is not sending any radio messages anymore. Will try to find way to activate it, or maybe somebody with knowledge will read this. 

Anyway i have tested sensor in few situations:
- placed in oven 50 deg C. - hoped there will be any information about temperature;
- placed on table up and covered with bowl - to check if US sensors are doing anything;
- placed on table up without cover;
- with dismounted tray/bracket
- without plastic body faced down  - there was only bare PCB left 
- without plastic body faced up
- when draining DC with LED 

Probably i divided this frame in non usual way but:

1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15 | 16
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
00 | 9F29 | 2D2A | 00 | 88 | 2D2A | 00 | 80 | 16 | 02 | E331 | 2A9F44 | 2B9F44 | 008D | C3009ED4 | 000F
  |   |   |   |   |   | 01 | 00 | 17 |   | EF31 |   |   |   |   |  
  |   |   |   |   |   | 02 | 01 | 18 |   | F131 |   |   |   |   |  
  |   |   |   |   |   | 03 |   |   |   | F631 |   |   |   |   |  

Part 7 
changed from 00 to 01 after removing sensor body but not immediately. 
Then changed to 02 after draining battery with led and occasionally switched to 3 
It is somehow connected with battery/load sensor/body sensor.

Part 8 
was 80 until i took sensor out of tray/bracket then became 00 
when started to drain battery became 01

Part 9 
It seems to be a date but it gave 
for 13.01 - 16
for 14.01 - 17
for 16.01 - 18 
so it is not directly connected 

part 11 
was E331
became EF31 when sensor was pulled out of tray
became F131 when i took body off 
became F631 when i started to drain battery

Ok I know that this is ain`t much. With this knowledge I will surely test it better next time because I did not saw any changes in telegrams during test them self. If I will not be able to activate it tomorrow I will buy few of them in next week.

CuBiC3D, 2021-02-09:
I made a PR for last period date. From other findings:
- Part 8 seems to be some sort of state, it's `00` by default, turns `80` overnight and `00` again overnight possibly to signalise a distance check on that day. Not sure, another guy mentioned that.
- Every time part 8 changes state, part 9 will count up - that's probably a positive result counter for checks.
- Part 9 is probably the negative alarm counter.
- Part 7 should also be a sort of state, I've seem someone mentioned it's turning `01` after button press after alarm and back to `00` after 15 minutes.

I'll have to log some messages and see what's there to find. Where did you buy them from? I have a lot of them in radio range, but only 2 accessible.

Kosygor, 2021-02-09:
I am still waiting for RTL-SDR dongle - chinese new year apparently lasts for 2 weeks.
I bought them from olx.pl - polish local trading site guy have like 40 of them for 3 euro each. 
All brand new (not initiated by putting them into tray/bracket) I was stupid enough not to check what they send before they are not initiated. Chassis parts are produced in 2019/2020 so You should have around 7-8 year (Techem states that they last 10+1 year whatever that +1year means). 
They are not pretty, but if I can equip whole house with price of one zigbee smoke sensor ......

weetmuts, 2021-04-10:
What changes do you need help with?

Kosygor, 2021-04-10:
@weetmuts 
Basically in programming changes. 
I will try to add them by myself, but I have to end one more test tomorrow and get aquainted with how github works (as I have seen in other topic it is easier for You to check anything if i make pull request).
Still need some time, all of this takes much more time than I thought.

Kosygor, 2021-04-12:
@weetmuts Ok so below I summarize only things I am 100% sure. So If I Can ask You Frederic could you code that down? If You do not have time I will try to do it myself do not know when  because  just took 30 min of trying to understand how meter_X.cc files work and failed. 

**Part 1 ( byte 1 )**
00 = "OK"
01 = "SMOKE"
04 = "ENVIROMENTAL CONTROLL"
80 = "DEFECT"
ELSE 
"UNKNOWN/ERROR"

**Part 10 ( byte 13)** 
"Alarm Count"
_Expressed in hex max value 255 (see belowe)_

**Part 11 ( byte 14 )**
"Click count"
_in Hex max value 255 (tested it and after 255 clicks nothing happens)_

 I know this is insultingly low in time given.

flyingOwl, 2021-06-10:
Hi,

I would like to join in with some more information. I'm living in an apartment building where every meter is made by Techem and transmits data over the air. With my generic rtl-sdr dongle I receive over 50 different smoke detectors.

Adding to the table @Kosygor posted above, I would split part 11 in two, and  order part 14 and 15 a bit different.

This is the smoke detector in one of my rooms. It never signaled any kind of alarm, nor did I pressed the button:

date | 1 | 2 | 3 | 4 | 5 | 6 | 7 | 8 | 9 | 10 | 11 | 12 | 13 | 14 | 15 | 16 | ...
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
2021-06-09 07:13:44 | 00 | 9f29 | c22a | 00 | b0 | c22a | 00 | 80 | 14 | 00 | 00 | 41 | 006ba1 | 007cb2 | 008dc3 | 009ed4 | 000fe500
2021-06-09 12:17:20 | 00 | 9f29 | c92a | 00 | b0 | c92a | 00 | 80 | 14 | 00 | 00 | 41 | 006ba1 | 007cb2 | 008dc3 | 009ed4 | 000fe500
2021-06-10 | 00 | 9f29 | c92a | 00 | b0 | c92a | 00 | 00 | 15 | 00 | 00 | 41 | 006ba1 | 007cb2 | 008dc3 | 009ed4 | 000fe500

My take on each part:

**Part 1**

Must be the current status, as already said. Though, I've never seen a value other than '00' in my recording.

**Part 2**

This is a date, in some quirky format. Techem uses this format for all of their devices. First you have to swap the low and high byte:

```
0x299f = 0010 1001 1001 1111 # group bits, 7 bits = year, 4 bits = month, 5 bits = day
0010100 1100 11111 = 20 12 31 (2020-12-31)
```
This number/date is used by the heat and water meters as some kind of due date or the payment period reference. I don't know why a smoke detector needs this.

**Part 3 and Part 6** (I have never seen them differ)

Also a date. Should be the date of the last self-check (distance check).

```
0x2ac2 = 0010101 0110 00010 = 2021-06-02
0x2ac9 = 0010101 0110 01001 = 2021-06-09
```

This value changes right after the distance check is done. The date is not in sync with part 8 or 9. The devices are equally distributed over the last few days.

**Part 4**

A counter. Most devices state '00', some '01'. I did not record any transition.

**Part 5**

? - 'b0' for all devices captured

**Part 7**

A counter, maybe? All (except one device) state '00' if part 4 is '00'. One device toggled between '00' and '01', another was at '05', jumped to '06' and back to '05' (at random times).

**Part 8**

Seems to be a bit mask.

`0x80` - Somehow date related. All devices had this bit set on 2020-06-09, but none on 2020-06-10.
`0x01` - Only some devices have this set. Though, I have never seen a transition.

**Part 9**

A day(?) counter. Most of the devices are synchronous, as all apartments were equipped on the same day. Value is not related to part 8. I will continue to capture data to see when it jumps back to '00'.

**Part 10 and Part 11**

Alarm counter / Click counter, as already said.

**Part 12**

Most devices send '41'. Some devices send '31' or '32', but those seem to be from a different time/building/whatever as their ID is very different.

**Part 13, 14, 15, 16**

My guess: this is an event list

If there has never been an alarm, this first byte is '0', followed by some test pattern (6ba1, 7cb2, ...)

Devices with 1 or 2 alarms (Part 10), start with '1' or '2' here. Only part 13 and 14 are populated, Part 15 and 16 always start with '00'.

If the first byte is bigger than '0', part 13 continues with 'a', part 14 with 'b' ('1a5618 1b5618' or '2a7c53 2b7c53'). The following four bytes are the same in part 13 and 14.

Only one device has just part 13 set: '17e93d 007cb2 008dc3 009ed4'

I will continue to record data, and might provide raw samples if you want.

flyingOwl, 2021-06-20:
Adding to my last comment, after I've analysed some more days of data:

Part 8 and part 9 must be evaluated together, with part 9 being the high byte and part 8 being the low byte.

```
# with bytes swapped - values I've seen:

17 00 = 0001  01110  0000000 = 1 14 0 (only seen on June 14)
17 80 = 0001  01111  0000000 = 1 15 0 (only seen on June 15)
18 00 = 0001  10000  0000000 = 1 16 0 ...

27 00 = 0010  01110  0000000 = 2 14 0
27 80 = 0010  01111  0000000 = 2 15 0

17 01 = 0001  01110  0000001 = 1 14 1
17 81 = 0001  01111  0000001 = 1 15 1

# only TSD2 devices from a very different serial number range than the others
07 00 = 0000  01110  0000000 = 0 14 0
07 80 = 0000  01111  0000000 = 0 15 0
```

The middle part is the current day of the month. It jumps to the next number at night, between 0:15 and 0:45 AM, depending on the device. One device is already a day ahead.

I have seen one transition of the first part, going from 0x01 to 0x02. It changed at midnight, simultaneously with the middle part (from "17 00" to "27 80").

The last part is either "0000000" (most devices) or "0000001". I have not recorded a value change for any device.

fhloston, 2022-01-16:
Just got a RTL-SDR dongle these days and have 3 of these Techem Smoke Detector 2 in my flat - as do all neighbors.
However, with 
`$ wmbusmeters --logtelegrams auto:t1`
I do not see the telegrams.

With `rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus` I do see these telegrams:

```T1;0;0;2022-01-16 16:41:39.000;89;163;FF63FFFF;0x2944ff50ffff63ff76f0a0ff9f2bffffffffffffffff5802ff05ffffffffffea00ffff00ffffffffff00
T1;0;0;2022-01-16 16:43:18.000;89;160;FF63FFFF;0x29446850ffff63ffffffff009f2b30ff00ffffffff00ffffffff2a87ffffffff008dc3ffffd4ffffffff
T1;0;0;2022-01-16 16:45:23.000;91;139;FFFFFFFF;0x29ff68ffffffffff76f0a0ffffffffffffffff2cff00ff02ff052affeaffffffff8dc3ff9effff0fffff
T1;0;0;2022-01-16 16:48:05.000;91;140;FFFFFFFF;0x294468ffffffffff76ffa0ff9f2bff2c0008ffff0000ffff0005ffffffff87ffff8dffffffffffffff00
T1;0;0;2022-01-16 16:54:37.000;87;135;FFFFFFFF;0x29ffffffffffffffffffa0ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
T1;0;0;2022-01-16 16:56:13.000;93;128;FFFFFFFF;0x294468fffffffffffff0a0ff9fffffffffffff2cffffffffffff2a87ea2b87eaffffffff9eff000fe500
T1;0;0;2022-01-16 16:57:23.000;97;127;FF637791;0x294468ff917763ffffffff00ff2b302cffffffffffffffff00ff2a87ea2b87ff00ffffffffff00ffe5ff
T1;0;0;2022-01-16 16:59:33.000;94;167;FF637791;0x29446850917763ffffffa000ff2b302c00ffffffffff58020005ff87ffffffffffffffff9ed4000fe5ff
T1;0;0;2022-01-16 17:01:07.000;94;136;FF637791;0x29446850917763ffffffffffffffffffff08302c0000ffffff052a87ffffffff00ffc300ffd4ffffffff
T1;0;0;2022-01-16 17:03:19.000;95;164;51637791;0x29ffff5091776351ffffffffff2bff2c00ffffffffffff02ffff2a87ffffffea008dc3009ed4000fffff
T1;0;0;2022-01-16 17:03:50.000;94;154;FFFFFFFF;0x29ff68ffffffffffffffff009f2bffff00ff30ffffff5802ff052affffffffffff8dff00ffd4ff0fe5ff
T1;0;0;2022-01-16 17:05:31.000;93;167;FFFFFFFF;0x294468ffffffffffffffffffff2b302c0008302cffffffffffffffffea2b87eaffffffff9ed4ff0fe5ff
T1;0;0;2022-01-16 17:10:56.000;98;127;FFFFFFFF;0x29ffff50fffffffffff0a0009fffffffffffff2cff00ff02ff052a87ea2bffff008dc3ff9ed4000fffff
T1;0;0;2022-01-16 17:11:08.000;81;143;FFFFFFFF;0x29ffffffffffffffffffffffffffffffffffffffff00ff00ffffffffffffffffffffffffffd4ffffffff
T1;0;0;2022-01-16 17:13:06.000;89;162;FFFF77FF;0x29446850ff77ffffffffa0ff9fff302c0008ffffffffffffff052a87ffff87ff00ffc3ff9ed4ff0fffff
T1;0;0;2022-01-16 17:14:40.000;99;146;51FFFFFF;0x294468ffffffff51fff0ff00ffffffffffffff2c00ff58ff00ff2affeaff87ff008dc3009ed4ff0fffff
T1;0;0;2022-01-16 17:15:49.000;93;131;51637791;0x29ff68ff9177635176f0ffffffff302c00ffff2cff00ffffffff2a87ff2b87ffffffffffffd4ff0fe5ff
T1;0;0;2022-01-16 17:17:23.000;100;139;FFFFFFFF;0x294468ffffffffffffffff00ff2b302c00ff30ffffff5802ff052a87ffffffffffffffff9ed4000fffff
T1;0;0;2022-01-16 17:17:53.000;95;128;51637791;0x2944685091776351ffffffffff2bff2cffffffffffffffffff052affffffffffffffc3ff9eff00ffe5ff
T1;0;0;2022-01-16 17:20:36.000;91;161;5163FFFF;0x29ff68ffffff6351ffffa000ff2bff2c0008302c00ff5802ff05ffffff2bffffff8dff00ffffffffffff
T1;0;0;2022-01-16 17:23:56.000;95;168;51FF77FF;0x2944ff50ff77ff5176f0a0009f2bffffffff30ffffff58ff00ffffffff2b87ff008dc3009eff000fffff
T1;0;0;2022-01-16 17:26:01.000;91;148;516377FF;0x29446850ff776351ffffa0ffffffff2cff08ff2c00ff58ff00ff2a87ff2bffeaff8dc300ffffffffffff
T1;0;0;2022-01-16 17:27:42.000;92;129;FFFFFFFF;0x294468ffffffffff76f0a000ffffffffffff30ff00ff58ff0005ff87ea2bffea008dc3009ed4000fffff
T1;0;0;2022-01-16 17:30:55.000;90;132;51FFFFFF;0x29ff6850ffffff5176f0ffffffff302c00ffffffffffffffffff2a87ea2b87ea008dffff9effff0fe500
T1;0;0;2022-01-16 17:34:08.000;89;164;FF637791;0x29446850917763ffffffff009fff302cffffffffffff58ff0005ffffffffffffffffffff9ed4ff0fffff
T1;0;0;2022-01-16 17:36:20.000;85;126;FFFFFFFF;0x29fffffffffffffffff0a0009f2bffff0008ffffffffff02ff052affffffffffffffc3ffffd400ffe500
T1;0;0;2022-01-16 17:39:03.000;97;152;51FF77FF;0x2944ffffff77ff51fff0a000ff2bffffffff302c00005802ffffffffeaffffff00ffc3ffffffffffe500
T1;0;0;2022-01-16 17:39:33.000;92;163;51FFFFFF;0x29ff6850ffffff51fff0ffffff2bffff00ff30ffff00580200052affffffffff008dc3009ed400ffffff
T1;0;0;2022-01-16 17:41:45.000;95;132;FF637791;0x294468ff917763ffffffffffff2b302c0008ffffffff580200052affffffffffffffff009ed4ff0fffff
T1;0;0;2022-01-16 17:53:44.000;93;171;FF637791;0x29ff68ff917763ffffffffffffff302c00ffffffffffff02ffff2a87ffffffffff8dffff9ed4ff0fffff
T1;0;0;2022-01-16 17:55:18.000;94;144;FFFFFFFF;0x29446850ffffffff76f0a0009fffffffffffff2cff00ffffff052a87ea2bffea00ffffffffd4ff0fffff
T1;0;0;2022-01-16 17:56:26.000;98;127;51FFFFFF;0x29ffff50ffffff5176ffa000ff2bffffffff302cff00ffff00ffff87ea2b87ffffffc3ffffffffffffff
T1;0;0;2022-01-16 17:58:00.000;90;142;FF63FFFF;0x2944ffffffff63ff76f0ffffff2b302c000830ffff00580200052a87ffffffff008dc3ffffffffffffff
T1;0;0;2022-01-16 18:01:51.000;81;127;51FF7791;0x29ff68ff9177ff51ffffff00ffff302cffff30ff000058ffffffffffeaff87ea008dc300ffffffffffff
T1;0;0;2022-01-16 18:03:26.000;96;128;FFFFFFFF;0x294468fffffffffffff0a000ffffffffffffff2cff005802ff05ffffeaff87ff00ffc3009ed4000fffff
T1;0;0;2022-01-16 18:06:08.000;90;156;FFFFFFFF;0x29ff6850fffffffffff0a0009fffffffffffffff00ffff0200ffffffffffffffffffff00ffd400ffe5ff
T1;0;0;2022-01-16 18:09:21.000;97;169;51FF7791;0x2944ff509177ff51ffffa0ffff2b30ffffffffffffffffffff05ff87ea2b87ea008dc3009effffffff00
T1;0;0;2022-01-16 18:12:17.000;31;16;FFFFFFFF;0x09ff6850ffffffffffff
T1;0;0;2022-01-16 18:15:56.000;91;133;51FF7791;0x29ffffff9177ff51ffffff00ff2b30ff00ffffffff00ffff00ff2a87ffffffffff8dc3009eff000fffff```

Can I help getting these devices supported?

weetmuts, 2022-01-16:
Yes, T1;0 means that it received a telegram but the CRC checks failed. I.e. there ware errors when receiving the radio signal. You visually see that with all the fffffff:s above.
wmbusmeters just ignores those telegrams, but you can see the discard messages if your run with --debug or --verbose.

See if you can get a better antenna or be closer to your detector.

fhloston, 2022-01-17:
Oh. I am almost sitting directly under one - but I do not know how often they send a telegram. Antenna used is from a lorawan EU gateway for 868MHz.

There are 18 flats in 3 housese, I am sitting in the middle on top of them all. Roughly 18x3 devices. So most of them are probably too far away. 

When exactly do these send their messages? 

And assuming the S/N on the device is `44836118` I should expect to see frames with:
`0x29446850 18618344...`

Correct?

Kosygor, 2022-01-18:
Try adjusting frequency to 868.9M or 869.0M it helped me in 2 of my 3 RTL-SDR`s.
Also provide ground if You are using monopole antenna.
If I can recall right they send frames in 4 or 5 minutes.

BIBOLV, 2023-01-01:
Closing due to inactivity. Please reopen issue if future improvements of tsd2 driver is necessary.

closed 2023-01-01 by BIBOLV