wmbusmeters/wmbusmeters #203: Support for Qundis WHE465 Heat Cost Allocator.

ib0ware, 2020-12-22:
Dear Programmers,

I have the [Qundis WHE465](https://qundis.com/wp-content/uploads/2015/05/N2877WBen_20100120_WHE465_WHE466.pdf) which sends S1 datagrams in my appartement and I am trying to implement it into wmbusmeters, but I am stuck. 

Command: **sudo wmbusmeters --logtelegrams auto:s1**

`Received telegram from: 60367815
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x18
                device: im871a[00102632]
                  rssi: -92 dBm
                driver: unknown!`
                
I added the following telegrams into simulate.txt
`telegram=||344465321578366018087A90040000046D1311962C01FD0C03326CFFFF01FD7300025AC2000DFF5F0C0008003030810613080BFFFC|
telegram=||344465325002366018087A8A040000046D1A11962C01FD0C03326CFFFF01FD7300025AFF000DFF5F0C0008003030810613080BFFFC|
telegram=||344465320069366018087ABF040000046D1D11962C01FD0C03326CFFFF01FD7300025ACB000DFF5F0C0008003030810613080BFFFC|
telegram=||344465326265366018087A95040000046D1D11962C01FD0C03326CFFFF01FD7300025AC0000DFF5F0C0008003060810613080BFFFC|`

I tried to decode it with all 4 of the heat allocator supported types but no success.

`ib0ware@Zer0day-Ubuntu:~$ wmbusmeters --debug simulation.txt
(wmbusmeters) version: 1.0.4-16-g9c7d7dd
(config) using device: simulation.txt:any 
(config) number of meters: 0
(lookup) with file "simulation.txt"
(lookup) driver: simulation file
(main) opening simulation.txt:any
Started config simulation on simulation.txt listening on any
(simulation) in simulation.txt
(main) regular reset of simulation  on simulation.txt will happen every 82800 seconds
(main) added simulation.txt to files
(wmbus) no alarm (expected activity) for simulation
(main) simulation.txt already configured as simulation
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(simulation) from file "344465325002366018087A8A040000046D1A11962C01FD0C03326CFFFF01FD7300025AFF000DFF5F0C0008003030810613080BFFFC"
(wmbus) parseDLL @0 53
(telegram) DLL L=34 C=44 (from meter SND_NR) M=3265 (LSE) A=60360250 VER=18 TYPE=08 (Heat Cost Allocator) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 43
(wmbus) parseAFL @10 43
(wmbus) parseTPL @10 43
(dvparser) found new format "046D01FD0C326C01FD73025A0DFF5F" with hash c9af, remembering!
(telegram) TPL CI=7a ACC=8a STS=04 CFG=0000 ()
Received telegram from: 60360250
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x18
                driver: unknown!
(wmbus) 00: 34 length (52 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 6532 dll-mfct (LSE)
(wmbus) 04: 50023660 dll-id (60360250)
(wmbus) 08: 18 dll-version
(wmbus) 09: 08 dll-type (Heat Cost Allocator)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: 8a tpl-acc-field
(wmbus) 0c: 04 tpl-sts-field
(wmbus) 0d: 0000 tpl-cfg 0000 ( )
(wmbus) 0f: 04 dif (32 Bit Integer/Binary Instantaneous value)
(wmbus) 10: 6D vif (Date and time type)
(wmbus) 11: 1A11962C
(wmbus) 15: 01 dif (8 Bit Integer/Binary Instantaneous value)
(wmbus) 16: FD vif (Second extension of VIF-codes)
(wmbus) 17: 0C vife (Model/Version)
(wmbus) 18: 03
(wmbus) 19: 32 dif (16 Bit Integer/Binary Value during error state)
(wmbus) 1a: 6C vif (Date type G)
(wmbus) 1b: FFFF
(wmbus) 1d: 01 dif (8 Bit Integer/Binary Instantaneous value)
(wmbus) 1e: FD vif (Second extension of VIF-codes)
(wmbus) 1f: 73 vife (Reserved)
(wmbus) 20: 00
(wmbus) 21: 02 dif (16 Bit Integer/Binary Instantaneous value)
(wmbus) 22: 5A vif (Flow temperature 10⁻¹ °C)
(wmbus) 23: FF00
(wmbus) 25: 0D dif (variable length Instantaneous value)
(wmbus) 26: FF vif (Vendor extension)
(wmbus) 27: 5F vife (duration of limit exceed last upper  is 3)
(wmbus) 28: 0C varlen=12
(wmbus) 29: 0008003030810613080BFFFC
telegram=||344465325002366018087A8A040000046D1A11962C01FD0C03326CFFFF01FD7300025AFF000DFF5F0C0008003030810613080BFFFC|+0
(serial) stopping manager
(serial) waiting for stop
(serial) event loop stopped!
(wmbus) deleted simulation`

Output of the other commands:
[fkhvdataiii.txt](https://github.com/weetmuts/wmbusmeters/files/5731255/fkhvdataiii.txt)
[qcaloric.txt](https://github.com/weetmuts/wmbusmeters/files/5731256/qcaloric.txt)
[somtex868.txt](https://github.com/weetmuts/wmbusmeters/files/5731257/somtex868.txt)

I hope you can help me supporting this type of sensor.

weetmuts, 2020-12-28:
Thanks, I examined the telegram. Unfortunately it stores most of its data inside a proprietary vendor extension string,
I have added a stub for the whe46x driver to wmbusmeters. If you rebuild master you can do:

./build/wmbusmeters --format=json  simulation_whe456.txt HCA whe46x  60367815 NOKEY

{"media":"heat cost allocation","meter":"whe46x","name":"HCA","id":"60367815","current_consumption_hca":0,"set_date":"","consumption_at_set_date_hca":0,"set_date_1":"","consumption_at_set_date_1_hca":0,"flow_temperature_c":19.4,"error_date":"2127-15-31","device_date_time":"2020-12-22 17:19","unknown":"0008003030810613080BFFFC","timestamp":"2020-12-28T18:22:26Z"}

The only proper values found are the flow_temperature, the device_date_time, the remaining information
is probably inside the unknown. See if you can run reverse engineer the unknown data, perhaps by comparing
with the data visible on the device display?

Since this is a S1 meter and it contains an odd looking vif, listening_window_management_data_type_L_
perhaps we need to transmit a wakeup command to the meter to get even more information?

The QCaloric HCA sends two different types of telegrams. Perhaps if you wait long enough you can see
two different types of telegrams here as well?

weetmuts, 2021-01-26:
Ok, I got a telegram from a WHE542-D291S Heat Cost Allocator.
It contained the proper HCA value. So a working driver whe5x is added.

@ib0ware Have you collected more telegrams from your whe4x meter?

ib0ware, 2021-01-26:
Sorry for the late reply, long vacation :)

I have added output of the command: sudo wmbusmeters --logtelegrams auto:s1 , so you can find al sorts of telegram. If you need more information let me know.

[output.txt](https://github.com/weetmuts/wmbusmeters/files/5876666/output.txt)

makuser, 2021-01-28:
> Thanks, I examined the telegram. Unfortunately it stores most of its data inside a proprietary vendor extension string,
> I have added a stub for the whe46x driver to wmbusmeters. If you rebuild master you can do:

Why are we creating a new device driver for a device family that already exists? Does it make sense to maintain it that way?

Both the Qundis Caloric 5.5 and the Siemens WHE54.. and WHE63.. are the same device and can be set to S- or C-Mode via software configuration. The different Siemens model numbers just show the default configuration: WHE542 is S-Mode, WHE632 is C-Mode.
And since WHE54.. and WHE63.. are compatible to WHE55.. and WHE56.. and also WHE3.. and WHE4..., we can expect them to be using the same data structure.

I would suggest using only the qcaloric driver moving forward, because theoretically also the Qundis Caloric can be set to S-Mode and send the same data as seen here. Feature-wise they would be the same in the end.

weetmuts, 2021-01-30:
It seems like you know much more about these meters than I do! Which is not surprising :-) I only know the telegrams users send me and the brand name they call the meter. Can you assist in creating a pull-request?

gswdn, 2021-11-28:
I'd like to push this topic up again. Seems, like I also have Qundis/Siemens devices installed. 
They are printed with "Kalo K1.5X" and "Qundis GmbH, Erfurt". Kalo seems to be some kind of service provider. Maybe, Qundis sells their devices white-labelled? Or is it a cooperation with Siemens? Or Siemens white-label devices?

Anyway, they transmit as follows:

`(telegram) DLL L=26 C=44 (from meter SND_NR) M=3265 (LSE) A=89186680 VER=35 TYPE=08 (Heat Cost Allocator) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=99
(telegram) TPL CI=7a ACC=8c STS=00 CFG=0000 ()
Received telegram from: 89186680
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x35
                device: rtlwmbus[00000001]
                  rssi: 99 dBm
                driver: unknown!
telegram=||264465328066188935087A8C0000000B6E2600004B6E000000426CFFFF326CFFFF046D0905BC2B|+0
`

This means a value of 26 or 27.

Same manufacturer ID as discussed here earlier. I have 7 in my flat and receive roughly 120 devices in total from all my neighbors here - it's a bigger real estate. So, have plenty of test-data, if required.
For mines, I can also provide more telegram data and readings from their displays.

I also see some more Qundis requests in the "issues". Perhaps, they all use the same or a similar data format? Can someone just try the other existing qundis drivers, please?

I'm not that c++/linux dev and just have wmbusmeter running in a docker container.

LukasACRIOS, 2021-12-06:
Hello! 

If it would help, I have collected some data from Qundis HCA. From what I can see it is sending 2 telegrams - 1 shorter and 1 longer. I did parse it with a tool here and it seems like it is parsed correctly (judging by the correct date), but the heat values are "0". The heat cost allocator is from the apartment building, so probably should not be a zero.

Data collector:
"timestamp": "2021-12-05T14:15:23.993480",
    "unit": "901288002328172",
    "wmbusId": "28177404",
    "found": true,
    "data": "49449344047417283508780dff5f350082e50000f10007b06effff86000000ffff00000000be2b7300000000800080008000800080008000000000000004001a002b002f046d110fa52c"
   
Parser: 
{
  "media": "heat cost allocation",
  "meter": "qcaloric",
  "name": "WMBUS",
  "id": "28177404",
  "current_consumption_hca": 0,
  "set_date": "",
  "consumption_at_set_date_hca": 0,
  "set_date_1": "",
  "consumption_at_set_date_1_hca": 0,
  "set_date_17": "",
  "consumption_at_set_date_17_hca": 0,
  "error_date": "",
  "device_date_time": "2021-12-05 15:17",
  "timestamp": "2021-12-06T14:01:32Z"
}

___________________________________________________________________
Data collector:
 "timestamp": "2021-12-05T17:39:14.148112",
    "unit": "901288002328172",
    "wmbusId": "28177404",
    "found": true,
    "data": "244493440474172835087a0000082004ed390a09a42801fd0c10046d2912a52c02fd3cc201",

Parser:
{
  "media": "heat cost allocation",
  "meter": "qcaloric",
  "name": "WMBUS",
  "id": "28177404",
  "current_consumption_hca": 0,
  "set_date": "",
  "consumption_at_set_date_hca": 0,
  "set_date_1": "",
  "consumption_at_set_date_1_hca": 0,
  "set_date_17": "",
  "consumption_at_set_date_17_hca": 0,
  "error_date": "",
  "device_date_time": "2021-12-05 18:41",
  "timestamp": "2021-12-06T14:01:47Z"
}
___________________________________________________________________
Data collector: 
 "timestamp": "2021-12-05T21:21:53.581349",
    "unit": "901288002328172",
    "wmbusId": "28177404",
    "found": true,
 "data":"314493440474172835087a1d0000200b6e8600004b6e000000426cffffcb086e730000c2086cbe2b326cffff046d1716a52c",

Parser:
{
  "media": "heat cost allocation",
  "meter": "qcaloric",
  "name": "WMBUS",
  "id": "28177404",
  "current_consumption_hca": 86,
  "set_date": "2127-15-31",
  "consumption_at_set_date_hca": 0,
  "set_date_1": "2127-15-31",
  "consumption_at_set_date_1_hca": 0,
  "set_date_17": "2021-11-30",
  "consumption_at_set_date_17_hca": 73,
  "error_date": "2127-15-31",
  "device_date_time": "2021-12-05 22:23",
  "timestamp": "2021-12-06T14:02:06Z"
}

gswdn, 2021-12-06:
The 0-values are also already mentioned here: https://github.com/weetmuts/wmbusmeters/issues/403
Perhaps the same thing?

In general, I got my HCA's working. See https://github.com/weetmuts/wmbusmeters/pull/404

makuser, 2021-12-09:
Yeah, that 0 total had been discussed previously, I do not remember which issue it was though.
Basically, if your HKV/HCAs are set to send in C-mode you'll be fine, because you can just set wmbusmeters to listen to c-mode alone and not listen to the telegrams that yield that value, which to my knowledge are only being sent in S-mode.

LukasACRIOS, 2021-12-09:
I did continue the conversation within issue #403. You are correct that you can set the HCA to transmit in the C or S-mode, however, even within the C-mode it sends 2 types of telegrams - at least that is my case. One telegram is for the walk-by readings and it is being sent every 112 seconds. The other is OMS format and it is sent every 450 seconds.

I attach the log of readings. The readings are from an apartment building done every hour. The HCA's are set to transmit in a C-mode. You can see that only some telegrams have data - the ones that start with a "31" (length). It can be also distinguished by filtering the Control Field which should be "7A".

[Qundis HCA parse.csv](https://github.com/weetmuts/wmbusmeters/files/7684612/Qundis.HCA.parse.csv)

weetmuts, 2021-12-09:
I have added some new functionality to wmbusmeters to assist in figuring out which driver to use, or build upon, it is also a help to add missing functionality to an existing driver. 

If you try:

wmbusmeters --analyze 314493440474172835087a1d0000200b6e8600004b6e000000426cffffcb086e730000c2086cbe2b326cffff046d1716a52c

You will get this output. (Looks better in the terminal where understood fields are green and not understood
content fields are red.)

The automatic driver selection picked "qcaloric" based on mfct/type/version!
Which is also the best matching driver with 19/19 content bytes understood.
000   : 31 length (49 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 9344 dll-mfct (QDS)
004   : 04741728 dll-id (28177404)
008   : 35 dll-version
009   : 08 dll-type (Heat Cost Allocator)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 1d tpl-acc-field
012   : 00 tpl-sts-field (OK)
013   : 0020 tpl-cfg 2000 (synchronous )
015   : 0B dif (6 digit BCD Instantaneous value)
016   : 6E vif (Units for H.C.A.)
017 C!: 860000 current consumption (86.000000 hca)
020   : 4B dif (6 digit BCD Instantaneous value storagenr=1)
021   : 6E vif (Units for H.C.A.)
022 C!: 000000 consumption at set date (0.000000 hca)
025   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
026   : 6C vif (Date type G)
027 C!: FFFF set date (2127-15-31)
029   : CB dif (6 digit BCD Instantaneous value storagenr=1)
030   : 08 dife (subunit=0 tariff=0 storagenr=17)
031   : 6E vif (Units for H.C.A.)
032 C!: 730000 consumption at set date 17 (73.000000 hca)
035   : C2 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
036   : 08 dife (subunit=0 tariff=0 storagenr=17)
037   : 6C vif (Date type G)
038 C!: BE2B set date 17 (2021-11-30)
040   : 32 dif (16 Bit Integer/Binary Value during error state)
041   : 6C vif (Date type G)
042 C!: FFFF error date (2127-15-31)
044   : 04 dif (32 Bit Integer/Binary Instantaneous value)
045   : 6D vif (Date and time type)
046 C!: 1716A52C device datetime (2021-12-05 22:23)

BIBOLV, 2023-01-01:
Closing due to inactivity.

closed 2023-01-01 by BIBOLV