wmbusmeters/wmbusmeters #207: IZAR RC868 i W Driver Unknown

theWagNerd, 2021-01-06:
Hi.

Apologies from the start - I've not done much programming since the 1980's (and even then, it was BASIC). During the last couple of months, I've stumbled upon home energy monitoring and programming in Raspian. I've persevered the last couple of days and have finally got wmbusmeters to run and have detected a number of  water meters including my own.

My meter is a IZAR RC868 i W with both SAPPEL and DIEHL also prominent on the front of the meter.

When I run wmbusmeters, I get device unknown, but can read both the realtime total consumption, and the consumption as of end of December. I'm still trying to get to grips with wmbusmeters syntax.

My water meter file (watermeter1) contains:
name=watermeter1
type=izar
id=a0c7625f
key=

If I run wmbusmeters rtlwmbus:868.9M:t1:
Started config rtlwmbus[00000001] listening on t1 using fq 868.9M
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: a0c7625a
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 104 dBm
                driver: unknown!
Received telegram from: a0c769c2
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 72 dBm
                driver: unknown!
Received telegram from: a0c75ab3
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 50 dBm
                driver: unknown!
Received telegram from: a0c7625f
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 117 dBm
                driver: unknown!
Received telegram from: a0c76986
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 52 dBm
                driver: unknown!
Received telegram from: a0c7625a
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 109 dBm
                driver: unknown!
Received telegram from: a0c769c2
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 70 dBm
                driver: unknown!

Please can you take a look at the driver issue.

Thanks

BIBOLV, 2021-01-06:
Log some telegrams and post them back here so we could take a look.
```bash
# wmbusmeters --logtelegrams rtlwmbus:868.9M watermeter1 izar:t1 a0c7625f NOKEY
```

theWagNerd, 2021-01-06:
Thanks BIBOLV

Here you go...

root@raspberrypi:/home/pi# wmbusmeters --logtelegrams rtlwmbus:868.9M watermeter1 izar:t1 a0c81054 NOKEY
Started config rtlwmbus[00000001] listening on none using fq 868.9M
(meter) watermeter1: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|1944304C5410C8A00604A2|41100013C2C839D2F611FDF7D77BC7|+6
watermeter1     a0c81054        1630.271 m3     1625.869 m3     2020-12-31      8 y     no_alarm        no_alarm        2021-01-06 20:49.46
(meter) watermeter1: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|1944304C5410C8A00604A2|51100013C2DF169B5D302E2E0C62C5|+14
watermeter1     a0c81054        1630.271 m3     1625.869 m3     2020-12-31      8 y     no_alarm        no_alarm        2021-01-06 20:49.54
(meter) watermeter1: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|1944304C5410C8A00604A2|61100013C2E66741A0525A446149C2|+23
watermeter1     a0c81054        1630.271 m3     1625.869 m3     2020-12-31      8 y     no_alarm        no_alarm        2021-01-06 20:50.03
(meter) watermeter1: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|1944304C5410C8A00604A2|71100013C2F148080B73899DBA50C0|+31
watermeter1     a0c81054        1630.271 m3     1625.869 m3     2020-12-31      8 y     no_alarm        no_alarm        2021-01-06 20:50.11

weetmuts, 2021-01-06:
Good work @theWagNerd  to get everything up and running!

As you write, your telegram seems to decode properly using the izar driver,
so it seems to be just another weird media type put in use by Sappel.....sigh.
I added it to the lookup table and pushed to master. 

It should be fine now! Please test!

theWagNerd, 2021-01-07:
Thanks all - that worked!!

root@raspberrypi:/home/pi# wmbusmeters rtlwmbus:868.9M:t1:

Started config rtlwmbus[00000001] listening on t1 using fq 868.9M
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: a0c81054
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Heat meter (0x04)
                   ver: 0x06
                device: rtlwmbus[00000001]
                  rssi: 67 dBm
                driver: izar

closed 2021-01-24 by weetmuts