wmbusmeters/wmbusmeters #208: Multical 803 E1 missing

Nikodemski2002, 2021-01-07:
Hi @weetmuts. Thanks for your help. After connecting the meter I saw that the total energy consumption is still displaying as 0, but it should be sth around 30. The meter wMBUS module is set for 30-10-102 configuration (Image 1). So the total energy consumption stands for E1 in the table. The meter is sending the E1 in GJ (Image 2). So do you have any idea, what I should change in the meter_multical803.cc file or somewhere else to see the total energy consumption. I also want to see the actual power in kWh, but I have to wait for the optical eye to change the wMBUS module configuration, to make it send this particular data.

This is what I've got from the ` wmbusmeters --format=json --debug auto:c1 79806477 multical803:c1 79806477 [KEY]`
```
(multical803) 00: 8c length (140 bytes)
(multical803) 01: 44 dll-c (from meter SND_NR)
(multical803) 02: 2d2c dll-mfct (KAM)
(multical803) 04: 77648079 dll-id (79806477)
(multical803) 08: 39 dll-version
(multical803) 09: 04 dll-type (Heat meter)
(multical803) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical803) 0b: 20 ell-cc (slow_resp sync)
(multical803) 0c: 59 ell-acc
(multical803) 0d: 93b84520 sn (AES_CTR)
(multical803) 11: 4770 payload crc (calculated 4770 OK)
(multical803) 13: 78 tpl-ci-field (EN 13757-3 Application Layer (no tplh))
(multical803) 14: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 15: FB vif (First extension of VIF-codes)
(multical803) 16: 08 vife (10^-1 GJ)
(multical803) 17: A5010000
(multical803) 1b: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 1c: FF vif (Vendor extension)
(multical803) 1d: 07 vife (?)
(multical803) 1e: * 24120000 something A (0.000000)
(multical803) 22: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 23: FF vif (Vendor extension)
(multical803) 24: 08 vife (?)
(multical803) 25: * 270E0000 something B (0.000000)
(multical803) 29: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 2a: 15 vif (Volume 10⁻¹ m³)
(multical803) 2b: * 831B0000 total volume (704.300000 m3)
(multical803) 2f: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 30: 40 dife (subunit=1 tariff=0 storagenr=0)
(multical803) 31: 14 vif (Volume 10⁻² m³)
(multical803) 32: 00000000
(multical803) 36: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 37: 80 dife (subunit=0 tariff=0 storagenr=0)
(multical803) 38: 40 dife (subunit=2 tariff=0 storagenr=0)
(multical803) 39: 14 vif (Volume 10⁻² m³)
(multical803) 3a: 00000000
(multical803) 3e: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 3f: 3C vif (Volume flow 10⁻² m³/h)
(multical803) 40: * 47060000 volume flow (16.070000 m3/h)
(multical803) 44: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical803) 45: 59 vif (Flow temperature 10⁻² °C)
(multical803) 46: * AD1A T1 flow temperature (68.290000 °C)
(multical803) 48: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical803) 49: 5D vif (Return temperature 10⁻² °C)
(multical803) 4a: * B614 T2 flow temperature (53.020000 °C)
(multical803) 4c: 14 dif (32 Bit Integer/Binary Maximum value)
(multical803) 4d: 2E vif (Power kW)
(multical803) 4e: 04010000
(multical803) 52: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 53: 10 dife (subunit=0 tariff=1 storagenr=0)
(multical803) 54: FB vif (First extension of VIF-codes)
(multical803) 55: 08 vife (10^-1 GJ)
(multical803) 56: 00000000
(multical803) 5a: 84 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 5b: 20 dife (subunit=0 tariff=2 storagenr=0)
(multical803) 5c: FB vif (First extension of VIF-codes)
(multical803) 5d: 08 vife (10^-1 GJ)
(multical803) 5e: 00000000
(multical803) 62: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical803) 63: FF vif (Vendor extension)
(multical803) 64: 22 vife (per hour)
(multical803) 65: * 00000100 info codes ()
(multical803) 69: 02 dif (16 Bit Integer/Binary Instantaneous value)
(multical803) 6a: 6C vif (Date type G)
(multical803) 6b: * A321 target date (2021-01-03 00:00)
(multical803) 6d: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 6e: FB vif (First extension of VIF-codes)
(multical803) 6f: 08 vife (10^-1 GJ)
(multical803) 70: 00000000
(multical803) 74: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 75: 15 vif (Volume 10⁻¹ m³)
(multical803) 76: 00000000
(multical803) 7a: C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 7b: 40 dife (subunit=1 tariff=0 storagenr=1)
(multical803) 7c: 14 vif (Volume 10⁻² m³)
(multical803) 7d: 00000000
(multical803) 81: C4 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 82: 80 dife (subunit=0 tariff=0 storagenr=1)
(multical803) 83: 40 dife (subunit=2 tariff=0 storagenr=1)
(multical803) 84: 14 vif (Volume 10⁻² m³)
(multical803) 85: 00000000
(multical803) 89: 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(multical803) 8a: 6C vif (Date type G)
(multical803) 8b: 0000
{"media":"heat","meter":"multical803","name":"79806477","id":"79806477","total_energy_consumption_kwh":0,"total_volume_m3":704.3,"volume_flow_m3h":16.07,"t1_temperature_c":68.29,"t2_temperature_c":53.02,"at_date":"2021-01-03 00:00","current_status":"","energy_forward_gj":"4644","energy_returned_gj":"3623","timestamp":"2021-01-03T15:31:37Z","device":"im871a[00101983]","rssi_dbm":-58}
```
**img1**
[image]

**img2**
[image]

weetmuts, 2021-01-24:
Ok, I think i fixed it. Your code originally cloned the multical603 driver and it expected kwh in the telegram.
But your multical803 transmits Joule. Perhaps all kamstrup heat meters can send either kwh or Joule,
we will deal with that in the future.

For the moment the multical803 driver is fixed and picks up the joul values and converts them to kwh in the json.
If you want joule, the add the option --addconversions=MJ to get MegaJoule in the json as well.

Please test!

weetmuts, 2021-03-06:
Any results from testing?

weetmuts, 2021-06-20:
Closing this. Assuming the fix works.

closed 2021-06-20 by weetmuts