wmbusmeters/wmbusmeters #211: qcaloric values fluctuating

makuser, 2021-01-13:
[image]
Unfortunately the values for all of our Caloric 5 are all over the place, as you can see in the graphs, and I do not think that the values on the actual units are
[image]

Example historic file content from one of the meters:
```
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":42,"set_date":"2020-12-31","consumption_at_set_date_hca":69,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":69,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":69,"error_date":"2127-15-31","device_date_time":"2021-01-13 05:38","timestamp":"2021-01-13T04:37:46Z","device":"rtlwmbus[00000001]","rssi_dbm":116}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":0,"set_date":"2020-12-31","consumption_at_set_date_hca":4,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":4,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":4,"error_date":"2127-15-31","device_date_time":"2021-01-13 05:45","timestamp":"2021-01-13T04:45:03Z","device":"rtlwmbus[00000001]","rssi_dbm":126}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":42,"set_date":"2020-12-31","consumption_at_set_date_hca":69,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":69,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":69,"error_date":"2127-15-31","device_date_time":"2021-01-13 05:52","timestamp":"2021-01-13T04:52:19Z","device":"rtlwmbus[00000001]","rssi_dbm":123}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":42,"set_date":"2020-12-31","consumption_at_set_date_hca":69,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":69,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":69,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:00","timestamp":"2021-01-13T04:59:35Z","device":"rtlwmbus[00000001]","rssi_dbm":116}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":42,"set_date":"2020-12-31","consumption_at_set_date_hca":91,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":91,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":91,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:03","timestamp":"2021-01-13T05:02:53Z","device":"rtlwmbus[00000001]","rssi_dbm":124}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":42,"set_date":"2020-12-31","consumption_at_set_date_hca":69,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":69,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":69,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:07","timestamp":"2021-01-13T05:06:51Z","device":"rtlwmbus[00000001]","rssi_dbm":119}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":4,"set_date":"2020-12-31","consumption_at_set_date_hca":21,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":21,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":21,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:08","timestamp":"2021-01-13T05:08:17Z","device":"rtlwmbus[00000001]","rssi_dbm":123}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":42,"set_date":"2020-12-31","consumption_at_set_date_hca":91,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":91,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":91,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:10","timestamp":"2021-01-13T05:10:07Z","device":"rtlwmbus[00000001]","rssi_dbm":119}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":0,"set_date":"2020-12-31","consumption_at_set_date_hca":0,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":0,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":0,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:14","timestamp":"2021-01-13T05:13:51Z","device":"rtlwmbus[00000001]","rssi_dbm":122}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":4,"set_date":"2020-12-31","consumption_at_set_date_hca":21,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":21,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":21,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:16","timestamp":"2021-01-13T05:15:31Z","device":"rtlwmbus[00000001]","rssi_dbm":122}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"123123123","current_consumption_hca":42,"set_date":"2020-12-31","consumption_at_set_date_hca":69,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":69,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":69,"error_date":"2127-15-31","device_date_time":"2021-01-13 06:22","timestamp":"2021-01-13T05:21:25Z","device":"rtlwmbus[00000001]","rssi_dbm":118}
```

Our main issue right now, is that `consumption_at_set_date_17_hca` seems to be changing as well.

weetmuts, 2021-01-13:
You need to run with --logtelegrams and then capture a good+bad+good triple of telegrams.
Then we can see what is actually sent in the telegram!

weetmuts, 2021-01-24:
So, just collect a long log of telegrams and email me. :-)

makuser, 2021-01-25:
> You need to run with --logtelegrams and then capture a good+bad+good triple of telegrams.
> Then we can see what is actually sent in the telegram!

So that is an interesting.
Todays log of one meter until now:
```json
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:05"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:12"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:19"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:26"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:34"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:41"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:48"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 00:56"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 01:03"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 01:10"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 01:18"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 01:25"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 01:32"}
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":38,"set_date":"2020-12-31","consumption_at_set_date_hca":17,"set_date_1":"2020-12-31","consumption_at_set_date_1_hca":17,"set_date_17":"2020-12-31","consumption_at_set_date_17_hca":17,"error_date":"2127-15-31","device_date_time":"2021-01-25 01:40"}
[... cut ...]
There seem to be two types of telegrams though, long and short ones and the short one is usually decoded fine, the long one on the other hand tends to yield unusable values. You will probably know right away what type of telegram this is, right?
short:
`telegram=|314493448936842635087AD6000020|0B6E3900004B6E170000426C9F2CCB086E170000C2086C9F2C326CFFFF046D150AB921|+0`
long:
`telegram=|4944934489368426350878|0DFF5F350082D60000610107B06EFFFF390000009F2C170000009F2C170000000080008000800080008000800080000000000000000011002F046D170AB921|+0`

weetmuts, 2021-01-25:
There are two type of telegrams sent, one with the actual values, the short one.
And another with meta-data about the meter, the long one.

When the driver receives the short one, it updates the values in memory for the meter
and invokes the shell script to send the telegram with the latest values.

When the driver receives the long one, it re-uses the values in-memory, 
sends the other values in the telegram, to the shell script.

Since the log looks just fine when I run it in a single run, I think
your wmbusmeters deamon/process is restarting every now and then!

Each time it has restarted, the last value is of course set to zero.
So if a long telegram arrives just after, then it will create a telegram
with zero value. This will get fixed the next time a short telegram ar, rives.

Sooo, this is kind of a bug/problem, possible solutions:
1) Do not send the consumption values when they are zero, ie the json
will look different since there will be no consumption_at_set_date_17_hca 
key etc. Its now your monitoring systems problem... :-)
 2) Do not send any telegram at all until at least a short telegram has arrived.
Simplest fix. Sensible for qcaloric, you are really not that interested in the meta data.
3) Create more than shell invocation, one for the short and one for the long,
so that you can easily direct the telegram to different mqtt receivers.

I am leaning towards 2 right now. Its a simple fix that you could do and post a pr for.

makuser, 2021-01-25:
> There are two type of telegrams sent, one with the actual values, the short one.
> And another with meta-data about the meter, the long one.

Thank you for that information

> When the driver receives the short one, it updates the values in memory for the meter
> and invokes the shell script to send the telegram with the latest values.
> 
> When the driver receives the long one, it re-uses the values in-memory,
> sends the other values in the telegram, to the shell script.

But does it use the in-memory values for that specific meter or for the last meter in memory, which might be a totally different one?

> Since the log looks just fine when I run it in a single run, I think
> your wmbusmeters deamon/process is restarting every now and then!

According to systemd it did not restart, which would also have been noted in the wmbusmeters.log file, no?

> Each time it has restarted, the last value is of course set to zero.
> So if a long telegram arrives just after, then it will create a telegram
> with zero value. This will get fixed the next time a short telegram ar, rives.

The **json** value is not zero though. It is all over the place. 4, 91, 17, 7, 6, etc. Would it be 0, it would have made sense to me.

When the simulation is run just with a long telegram alone, json values are indeed actually zero:
```json
telegram=|4944934489368426350878|0DFF5F350082D500005E0107B06EFFFF390000009F2C170000009F2C170000000080008000800080008000800080000000000000000011002F046D0F0AB921|+0
{"media":"heat cost allocation","meter":"qcaloric","name":"Heat Cost Allocator","id":"26843689","current_consumption_hca":0,"set_date":"","consumption_at_set_date_hca":0,"set_date_1":"","consumption_at_set_date_1_hca":0,"set_date_17":"","consumption_at_set_date_17_hca":0,"error_date":"","device_date_time":"2021-01-25 10:15","timestamp":"2021-01-25T10:23:39Z"}
```

Could it be that in such cases the value is actually taken from the last received short telegram in the memory buffer, instead of the last received telegram **for that specific serial**? That would make sense, because the values I got are not arbitrary at all, and definitely not 0=zero, they might very well be from the last meter that was received just before.

weetmuts, 2021-01-25:
>Could it be that in such cases the value is actually taken from the last received short telegram in the memory buffer, instead of the last received telegram for that specific serial? 

Yes, you are absolutely right. You are using a wildcard for the meters, right?

makuser, 2021-01-25:
> > Could it be that in such cases the value is actually taken from the last received short telegram in the memory buffer, instead of the last received telegram for that specific serial?
> 
> Yes, you are absolutely right. You are using a wildcard for the meters, right?

Yes I am, I am monitoring lots of meters. Would it make more sense to generate a file per actual meter? The only downside to me was that I can not assign an individual name per meter, but I am doing that in HASS anyway, so..

weetmuts, 2021-01-25:
Well, that would work around the bug/problem, but clearly it should be working with wildcards instead.

So, we have a meter that sends multiple types of telegrams. Should wmbusmeters wait for several
before sending the combined state. Should it relay each telegram.

How easy/difficult is it to deal with different types of json received into the same mqtt queue in HASS?

weetmuts, 2021-01-25:
Or we could have wmbusmeters create/new an exact meter driver C++ object, when a wildcard has been matched.
After that the exact meter driver will handle the telegram. This is perhaps useful for more complex meters.
Then have a configurable cap on the number of dynamic meters....

makuser, 2021-01-25:
> Well, that would work around the bug/problem, but clearly it should be working with wildcards instead.

I generally fully agree to rather fix bugs in the fundamental structure of a program, but that requires a lot of programming work and if it is just me that is affected by this, I would in this case take the immediate workaround and come back later to have it run properly.

> So, we have a meter that sends multiple types of telegrams. Should wmbusmeters wait for several
> before sending the combined state. Should it relay each telegram.

That is a really good question. Do we know what type of information is in the meta data? IMO the approach taken right now (merging new data with present in-memory data, overwriting old data but keeping unchanged or not present fields) is the right way, and we should pipe this combined information out via mqtt.

> How easy/difficult is it to deal with different types of json received into the same mqtt queue in HASS?

Really unpleasant. Conditional templating would be required, which does not really exist. At the moment HASS acts upon each topic change and if the json field that is expected by the template suddenly disappears, the sensor value will change to unavailable, which is not desired in this case.

> Or we could have wmbusmeters create/new an exact meter driver C++ object, when a wildcard has been matched.
> After that the exact meter driver will handle the telegram. This is perhaps useful for more complex meters.
> Then have a configurable cap on the number of dynamic meters....

I second that. Yes. That way we could keep merging the new data, with the present in-memory data.

makuser, 2021-01-30:
Alright, to verify the origin of the problem, I generated all the meter config files individually, which also allowed to give them individual names for the json content, but that is just a side effect.

The fluctuating values did indeed vanish, just as expected, so the solution/fix would be to spawn individual meter objects for every received unique and new serial number to only merge the alternating update object types (short&long) of one single meter and always be able to send out the merged state of a meter via mqtt.

I also did a rudimentary database cleanup afterwards, so my dataset looks good now, and new data looks much cleaner even.

weetmuts, 2021-02-20:
I have pushed a fix. Please test! Now each C++ object is spawned individually per distinct received meter.
A bit tricky to get right. But definitely worthwile, I am sure there will be other meter drivers with state.

weetmuts, 2021-03-06:
Have you had time to test the new version of wmbusmeters?

weetmuts, 2021-04-08:
Closing this for now. The problem should be fixed.

closed 2021-04-08 by weetmuts