wmbusmeters/wmbusmeters #212: A very basic question from a new beginner

vitto62, 2021-01-18:
First of all, a heartfelt thanks for your work. I have been using wmbusmeters for a few hours and am trying to follow the advice given in HowToAddaNewMeter.txt.
I understand that the command acts differently (in the sense that it outputs an "exended debug info") if invoked on input = file compared to input = device. I am wrong?
  So I provide a file that contains ONLY the following line
` telegram = || 32446850487480126980A0119E28420020230C00AF0818090000000000000000050403000000000000000000141500010001 |` 

 but the syntax appears to be incorrect singe I get: `no sensible telegram found, clearing buffer.`

weetmuts, 2021-01-18:
It does not like the whitespace. Try:
telegram=|......|
on a single line. Also such a simulation file must be named:
simulation_foobar.txt
I.e. it must start with simulation_

Then you can run:
wmbusmeters simulation_foobar.txt
and it will decode the telegrams.

vitto62, 2021-01-18:
Perfect!    Solved!   Thank you so much!!!!

vitto62, 2021-01-18:
Can you suggest other sources of information about your application or about newly added meters?  Maybe someone has already sorted out Perry's (SaveEnergy) HCA syntax, so I am wondering if my work may be useless.
THNX!!!

weetmuts, 2021-01-18:
Is your meter 12807448 ? If so, then its already supported with the fhkvdataiii driver. :-)
If not, then you have to post some telegram=|...| lines here so that we can take a look.

vitto62, 2021-01-18:
That was understood! :)  I wast exercising myself with wmbusmeters.   As I mentioned before I am currently "playing" with **Techem** HCAs.    And 12807448 is one of those!

weetmuts, 2021-01-18:
Nope, never heard of it. But if it is wmbus compliant, then it should be easy to add it.

vitto62, 2021-01-18:
Wonderful, I believe it is!    I'm eager to try.  :)

GRAZIE !!!

closed 2021-02-04 by weetmuts