wmbusmeters/wmbusmeters #214: mkradio3 logs smoke detectors with total_m3 / target_m3

deveth0, 2021-01-21:
Hey, 

first of all, thank you very much for this tool, I'm having a lot of fun with it ;)

While testing the usage with Techem Radio3 devices yesterday, I found a slightly weird behavior. My configuration is a wildcard device for testing:
```
name=Testing
type=mkradio3:t1
id=*
key=NOKEY
```

This catches a lot of devices around but also includes some weird ones:
```
{"media":"smoke detector","meter":"mkradio3","name":"Testing","id":"xXxXxXxXxX","total_m3":2160,"target_m3":1080,"timestamp":"2021-01-21T07:58:26Z","device":"cul","rssi_dbm":0}
```

I'm not sure, if those devices actually return interesting values or if they should simply be filtered out by the mkradio3 driver. 

As far as I can see, those are the devices:
https://www.gewobau-ni.de/wp-content/uploads/2014/08/RWM-2_Bedienungsanleitung_Auflage_4_25-04-2014-EN.pdf

If I can help you here in any way, please let me know!

weetmuts, 2021-01-24:
Thanks! It is fun! Normally, when a new meter comes out, which is backwards compatible with the old driver
it is necessary to be able to specify a meter id with the "wrong" driver since it might parse nicely.

But as you say, when the telegram is matched on a wildcard, that does not quite make sense.
So I have added an additional test, so that if the telegrams was matched against a wildcard rule, then
it will simply be ignored if the driver is wrong.

Pushed. Please test!

deveth0, 2021-01-24:
Cool, I just checked it out, my pi zero needs a bit until it's build.

deveth0, 2021-01-24:
Works like a charm!

closed 2021-01-24 by deveth0