wmbusmeters/wmbusmeters #216: Support for IZAR RADIO EXTERN 868 MBUS + Hydrus watermeter

zoesn89, 2021-01-23:
Hey guys,

I have a DIEHL Hydrus watermeter. It is a Mbus meter attached to an wMbus sender named IZAR RADIO EXTERN 868 MBUS. Unfortunately, this combination isn't supported yet.

Here is the output from the cli:

```
wmbusmeters /dev/ttyUSB0:im871a:t1               
Started config im871a[00101717] on /dev/ttyUSB0 listening on t1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 81100028
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Bus/System component (0x0e)
                   ver: 0x64
                device: im871a[00101717]
                  rssi: -66 dBm
                driver: unknown!
```
When I try the izar and hydrus driver, I get this:
```
wmbusmeters /dev/ttyUSB0:im871a:t1 Hydrus izar 81100028 NOKEY
Started config im871a[00101717] on /dev/ttyUSB0 listening on t1
(meter) Hydrus: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) Not printing this warning agin for id: 81100028 mfct: (HYD) Hydrometer (0x2324) type: Bus/System component (0x0e) ver: 0x64
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
```
Here are some telegrams:
```
telegram=||6644242328001081640E7266567464A51170071F0050052C411A08674048DD6BA82A0DF79FFD401309179A893A1BE3CE8EDC50C2A45CD7AFEC3B4CE765820BE8056C124A17416C3722985FFFF7FCEB7094901AB3A16294B511B9A740C9F9911352B42A72FB3B0C|+5

telegram=||6644242328001081640E7266567464A51170072000500511BB62025DDC52526495E29082A2A00D03386C80921C1F0594465FA7CE186BFEB2B0F3F91B00FA47A4017C909765720C3860CE887AF417F4B651B7E6D2BA7C96C4D6222876EADA756B9CAEAEE0D63D17|+23

telegram=||6644242328001081640E7266567464A511700721005005C8B60A0870C7328812325CA0A9D6C9D357F03C1C2D57A1D3FAB87C79BEFAA171098018FD5EE905A3DD24845C5200A42F92A4EC0E2AD3248086C98806D5A35F6131D1D6582B06CC841B672E9E46CEB26D|+38

telegram=||6644242328001081640E7266567464A51170072200500522FDC6C5B14CA3F6E50542033C972236CECF6722232A814199D60A7B38D899AB68428C67AC47D157DE85B7EB45D8D364D11E38A7A05DE62720D993643BDAAED0DC4D8F555C0484785F6547E870AFD0AC|+54

telegram=||6644242328001081640E7266567464A511700723005005E2DA60776484647D5B026B1DC7C0D233A6EF9F3F9556673073E4C32F4E1EA6FAC675F0C2C8C091BC426063151ABBA8D473598E52B3B69EC649E693F8C333B5545864326248E76E2A334904E4DAD88A52|+69

telegram=||6644242328001081640E7266567464A51170072400500595619C6DD4DCC01C3BA15D60393553C5FC91D6266D37C34B8EDB4DA1D476785DC5D07C1926D7E02F151186029DC1B64EBC7EA48AB4136691DA6B0963AC523582785F804D7D225552F955CD755D736F06|+85

```

I would really appreciate, when you can take a look at this 😉 

Thanks in advance!

Greets
Felix

weetmuts, 2021-01-24:
Thanks, your telegrams are using the standard AES encryption. Do you have a decryption key?

zoesn89, 2021-01-24:
Hey,

yes, I got two keys, but I don't know, if they are correct:

79 04 80 10 E6 6D 83 F8 51 72 89 10 E6 6D 83 F8

51 72 89 10 E6 6D 83 F8 51 72 89 10 E6 6D 83 F8

Thank you 😉

weetmuts, 2021-01-24:
Thanks, but unfortunately it seems like none of the keys work.

zoesn89, 2021-01-25:
> Thanks, but unfortunately it seems like none of the keys work.

ok, I requested the right key at my utility. When I get it, I'll send it to you by mail ;)

weetmuts, 2021-01-31:
Thanks, I have now added support for telegrams with a dll id from the radio converter, and a tpl id from the actual meter.
The decryption now uses the tpl id, which it should, so your telegrams now properly decrypts.

Also when listeningen for any telegram, such telegrams are printed thus:
Received telegram from: 81100028
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Bus/System component (0x0e)
                   ver: 0x64
      Concerning meter: 64745666
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07)
                   ver: 0x70
                driver: hydrus

weetmuts, 2021-01-31:
So now its time to decode yet another crazy telegram. Closing this issue and opening https://github.com/weetmuts/wmbusmeters/issues/223

closed 2021-01-31 by weetmuts

don-vip, 2021-02-07:
> telegram=||6644242328001081640E7266567464A51170071F0050052C411A08674048DD6BA82A0DF79FFD401309179A893A1BE3CE8EDC50C2A45CD7AFEC3B4CE765820BE8056C124A17416C3722985FFFF7FCEB7094901AB3A16294B511B9A740C9F9911352B42A72FB3B0C|+5

The meter uses the default Diehl AES key (`51728910E66D83F851728910E66D83F8`). Using the tool I created in https://github.com/weetmuts/wmbusmeters/issues/226#issuecomment-774565756, here is the decoded frame:

```java
  @Test
  void testFrameHydrus216() {
    assertEquals("ReadingData={"
        + "  rawMessage={"
        + "    frame=[66,44,24,23,28,00,10,81,64,0E,72,66,56,74,64,A5,11,70,07,1F,00,50,05,2C,41,1A,08,67,40,48,DD,6B,A8,2A,0D,F7,9F,FD,40,13,09,17,9A,89,3A,1B,E3,CE,8E,DC,50,C2,A4,5C,D7,AF,EC,3B,4C,E7,65,82,0B,E8,05,6C,12,4A,17,41,6C,37,22,98,5F,FF,F7,FC,EB,70,94,90,1A,B3,A1,62,94,B5,11,B9,A7,40,C9,F9,91,13,52,B4,2A,72,FB,3B,0C], timestamp=1612657129739.000, moduleId=0"
        + "  },"
        + "  frameDescription={ type=FrameDescMBusRadio,"
        + "    meterId=UDME700764745666000(parent=UHYD640E81100028000),"
        + "    meterName=HYDRUS_R4_DME,"
        + "    telegramType=OPEN_METERING_LONG,"
        + "    withAlarm=false,"
        + "    withError=false,"
        + "    meterDeviceNumber=64745666,"
        + "    mediumByte=0x07,"
        + "    version=112,"
        + "    manufacturer=HYD,"
        + "    errorFlagBytes=0x00,"
        + "    telegramContentType=STANDARD,"
        + "    installFrame=false,"
        + "    synchronizeFrame=false,"
        + "    lField=0x66,"
        + "    cField=0x44,"
        + "    mField=[A5,11],"
        + "    aFieldSecondaryAddress=[66,56,74,64,70,07],"
        + "    ciField=0x72,"
        + "    encMode=AES_CBC_IV },"
        + "  telegramFormatError=false,"
        + "  parsingLevel=FULL_CIPHERED,"
        + "  measurements={"
        + "    UDME700764745666000 -> {"
        + "       0 -> {"
        + "Date/Time= 2021.1.23 08:27:00,"
        + "        values=["
        + "          { semantic=actuality.duration,  type=QUANTITY,  isValid=true,  value=6673s },"
        + "          { semantic=volume,  type=QUANTITY,  isValid=true,  value=137.291㎥ },"
        + "          { semantic=volume.tariff[1],  type=QUANTITY,  isValid=true,  value=0.000㎥ },"
        + "          { semantic=volume.tariff[2],  type=QUANTITY,  isValid=true,  value=137.291㎥ },"
        + "          { semantic=volume.flow,  type=QUANTITY,  isValid=true,  value=0.000m³/h },"
        + "          { semantic=operating.time,  type=QUANTITY,  isValid=true,  value=14678h },"
        + "          { semantic=flow.temperature,  type=QUANTITY,  isValid=true,  value=24.5°C },"
        + "          { semantic=external.temperature,  type=QUANTITY,  isValid=true,  value=23.9°C },"
        + "          { semantic=date.time,  type=DATE,  isValid=true,  value=2021.1.23 8:27:0 }"
        + "        ]"
        + "      },"
        + "      1 -> {"
        + "Date= 2020.12.31,"
        + "        values=["
        + "          { semantic=volume,  type=QUANTITY,  isValid=true,  value=128.638㎥ },"
        + "          { semantic=volume.tariff[1],  type=QUANTITY,  isValid=true,  value=0.000㎥ },"
        + "          { semantic=volume.tariff[2],  type=QUANTITY,  isValid=true,  value=128.638㎥ },"
        + "          { semantic=date,  type=DATE,  isValid=true,  value=2020.12.31 },"
        + "          { semantic=date.FUTURE_VALUE,  type=DATE,  isValid=true,  value=2021.12.31 }"
        + "        ]"
        + "      }"
        + "    }"
        + "  }"
        + "}",
    new InterpretMBusRadioImpl().interpretFull(new RawMessage(
        HexString.getByteArray("6644242328001081640E7266567464A51170071F0050052C411A08674048DD6BA82A0DF79FFD401309179A893A1BE3CE8EDC50C2A45CD7AFEC3B4CE765820BE8056C124A17416C3722985FFFF7FCEB7094901AB3A16294B511B9A740C9F9911352B42A72FB3B0C"),
        1612657129739L)).toStringHumanReadable().replace("\r", "").replace("\n", ""));
  }
```