wmbusmeters/wmbusmeters #217: Techem Compact Ve support

deveth0, 2021-01-23:
We have a Compact Ve Heat meter in our flat which seems to use a similar protocol as the Compact5, except that the scaling is off by 1k (wmbusmeters: 1.72kwh, device: 1720kwh). 

I created #215 to add the detection but I guess we'll need to verify, if the scaling for the compact5 meter is also wrong, sadly there is no telegram to check.

weetmuts, 2021-01-24:
Hi @deveth0 and @IzeQube !

We now have two telegrams:

One for the Techem Ve, scales incorrectly
telegram=|36446850260833432243A1|009F2962060088560000809B410F30C0E043061858E0C001000020C00101142081041258D081062980E0C106| 
"total_kwh":1.72,"current_kwh":0.086,"previous_kwh":1.634
but should be multiplied with 1000, ie 1720 kwh

and one for the Techem V (inside simulation_t1.txt) scales correctly 
telegram=|36446850626262624543A1|009F2777010060780000000A00000000000000000000000000000000.0000000000000000A0400000B4010000|
"total_kwh":0.495,"current_kwh":0.12,"previous_kwh":0.375

Can you see if there is any obvious scale factor somewhere in the telegram?

deveth0, 2021-01-24:
Are you sure, that the scaling for the Techem V is correct? 0.375 kWh for heating seems way too low ?

IzeQube, 2021-01-24:
You, we don't heat a lot, so the value is quite low.

deveth0, 2021-01-24:
Yeah, but 0.375 kWh it really low. Even a the best low energy buildings use around ~10-15 kWh / m2 per year. Can you probably verify the value on the Compact5's display? 
Looking at some pictures of the Compact V, it does not even show values < 1 kWh:
https://www.as-gmbh.de/abrechnung/selbstablesung/selbstablesung-waermezaehler/selbstablesung-techem-compact-v

IzeQube, 2021-01-24:
Thinking about it, the value also has to be multiplied with 1000... That is also what i can read on my display.

deveth0, 2021-01-24:
Awesome, thanks for the confirmation! 
@weetmuts i added a PR #218 that removes the scaling.

weetmuts, 2021-01-24:
Great work!

weetmuts, 2021-01-24:
Fix pushed and test case is updated. Closing this for now. :-)

closed 2021-01-24 by weetmuts