wmbusmeters/wmbusmeters #220: Support Maddalena water meter

merbanan, 2021-01-26:
https://github.com/merbanan/rtl_433/pull/1616

A few telegrams there.

weetmuts, 2021-01-26:
Thanks! I added the driver evo868 for that meter, the example telegram in the rtl433 issue decodes into
the rather long json, some things are of course missing. error_flags == 0 decodes into OK, otherwise ERROR and the bits.
Is the max_flow measured since the max_flow_datetime, seems reasonable....but I am guessing at the purposes of that wmbus fields.

Set date and set date 2 are the same date. Probably because it is a new meter. It ends with FFFF, but I would have
expected 2F2F. Perhaps intended, perhaps a bug in their meter. Padding to fill up a block should be 2F.

{"media":"water","meter":"evo868","name":"Vatten","id":"20012906","total_m3":1.798,"device_date_time":"2021-01-17 17:30","current_status":"OK","fabrication_no":"000218400887","consumption_at_set_date_m3":1.225,"set_date":"2020-12-31","consumption_at_set_date_2_m3":1.225,"set_date_2":"2020-12-31","max_flow_since_datetime_m3h":0.666,"max_flow_datetime":"2021-01-07 20:05","consumption_at_history_1_m3":1.225,"history_1_date":"2020-12-31","consumption_at_history_2_m3":0.027,"history_2_date":"2020-11-30","consumption_at_history_3_m3":0,"history_3_date":"2020-10-31","consumption_at_history_4_m3":0,"history_4_date":"2020-09-30","consumption_at_history_5_m3":0,"history_5_date":"2020-08-31","consumption_at_history_6_m3":0,"history_6_date":"2020-07-31","consumption_at_history_7_m3":0,"history_7_date":"2020-06-30","consumption_at_history_8_m3":0,"history_8_date":"2020-05-31","consumption_at_history_9_m3":0,"history_9_date":"2020-04-30","consumption_at_history_10_m3":0,"history_10_date":"2020-03-31","consumption_at_history_11_m3":0,"history_11_date":"2020-02-29","consumption_at_history_12_m3":0,"history_12_date":"2020-01-31","timestamp":"2021-01-26T21:29:41Z"}

closed 2021-02-04 by weetmuts

MightyPork, 2021-12-22:
Here are a couple samples for this meter, in case it's useful for verification/debugging. The decoded JSONs look sensible enough, I mean, all I really want is the total consumption. 

set_date and set_date2 are different here

```
AC4424340054011950077AEC0000002F2F04135F000000046D0821B62C04FD17404000000E7843935620190044135D000000426C9F2C8401135F00000082016CBE2BD3013BC30000C4016D342CBC258104FD280182046CBE2B8404135F000000C404135F0000008405135F000000C405135F0000008406135F000000C406135F0000008407135F000000C407135D0000008408135D000000C408135D0000008409135D000000C409135D000000

AC4424342156011950077AE80000002F2F04133AF80000046D0921B62C04FD17000000000E781001552019004413D9600000426C9F2C840113FBEE000082016CBE2BD3013BAF0400C4016D1B2FA6268104FD280182046CBE2B840413FBEE0000C40413BAE1000084051307D50000C4051325C900008406136BBC0000C4061391B00000840713DFA50000C40713D6970000840813A58A0000C40813F27C00008409132A700000C40913D9600000

AC4424349753011950077AEA0000002F2F0413DE350000046D0821B62C04FD17004000000E784493562019004413BD000000426C9F2C840113F32E000082016CBE2BD3013BF80200C4016D1828B2268104FD280182046CBE2B840413F32E0000C40413F6260000840513651E0000C40513AA150000840613750C0000C40613A0080000840713B6010000C4071307010000840813F1000000C40813BD000000840913BD000000C40913BD000000

AC4424341549011950077AE70000002F2F0413586B0000046D0821B62C04FD17014100000E787499542019004413FE010000426C9F2C8401130664000082016CBE2BD3013B530400C4016D2D35AF2C8104FD280182046CBE2B84041306640000C40413DD590000840513C14F0000C40513AE4500008406134A3E0000C40613AA350000840713CD2C0000C407131223000084081317180000C40813D20C00008409137C030000C40913FE010000

AC4424348355011950077A880000002F2F041305060000046D0721B62C04FD17404000000E781486542019004413FE050000426C9F2C8401130506000082016CBE2BD3013BCF0200C4016D042CBA248104FD280182046CBE2B84041305060000C404130506000084051305060000C405130406000084061303060000C406130206000084071302060000C407130106000084081301060000C40813FE050000840913FE050000C40913FE050000

AC4424346355011950077A770000002F2F04139B220100046D0721B62C04FD17000000000E78238754201900441302690000426C9F2C8401139017010082016CBE2BD3013B040400C4016D3333B8218104FD280182046CBE2B84041390170100C40413280901008405131FFA0000C40513AAF20000840613A2E10000C4061342D700008407135EC60000C4071301B10000840813FDA10000C40813328F0000840913667D0000C4091302690000
```