wmbusmeters/wmbusmeters #222: IM871a-USB unknown device

alalons, 2021-01-29:
Hello, My name is Alvaro.
I got wmbusmeters working with rtl-sdr without any problem.
but can not get it with none of my two im871a dongles. They do work as I use them with other software jmbus. 

dmesg 
[ 2265.359985] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2265.359988] usb 5-1: Product: WiMOD iM871A-usb
[ 2265.359990] usb 5-1: Manufacturer: Silicon Labs
[ 2265.359993] usb 5-1: SerialNumber: 00B93F7D
[ 2265.365077] cp210x 5-1:1.0: cp210x converter detected
[ 2265.371098] usb 5-1: cp210x converter now attached to ttyUSB0

alvaro@alvaro-Compaq-Presario-CQ61-Notebook-PC:~$ wmbusmeters --debug  /dev/ttyUSB0:im871a:c1
(wmbusmeters) version: _
(config) using device: /dev/ttyUSB0:im871a:c1 
(config) number of meters: 0
(lookup) with file "/dev/ttyUSB0"
(im871a) detectando. <-- trace inserted myself
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
Expected im871a on /dev/ttyUSB0 but found unknown instead, ignoring it!
(verbose) ignoring device c1
No wmbus device detected, waiting for a device to be plugged in.

Any clue?

BIBOLV, 2021-01-29:
You need to add your user to group that grants permission to read USB device. Like dialout, or check group ownership of /dev/ttyUSB0 to be sure what is correct group.

alalons, 2021-02-01:
Thank you, works now

closed 2021-02-02 by alalons