wmbusmeters/wmbusmeters #223: Another crazy Hydrus telegram

weetmuts, 2021-01-31:


weetmuts, 2021-01-31:
Can you first run with just wmbusmeters auto:t1
to capture a bunch of:

Received telegram from: 81100028
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Bus/System component (0x0e)
                   ver: 0x64
      Concerning meter: 64745666
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07)
                   ver: 0x70
                driver: hydrus

To see if the type/ver change...

Then collect a bunch of --debug telegrams from your 64745666 meter
and grab the debug decoding lines using grep "(hydrus) ..:"
see if you can figure out any pattern and email me/post here the grepped lines as well.

zoesn89, 2021-02-08:
Hey @weetmuts,

did you get my logfiles? If yes, did they help? ;)

Felix

don-vip, 2021-02-13:
@weetmuts @zoesn89 I'm on it. I'm playing with this telegram:
`telegram=||6644242328001081640E7266567464A51170071F0050052C411A08674048DD6BA82A0DF79FFD401309179A893A1BE3CE8EDC50C2A45CD7AFEC3B4CE765820BE8056C124A17416C3722985FFFF7FCEB7094901AB3A16294B511B9A740C9F9911352B42A72FB3B0C|+5`
I'm observing odd behaviour right now, I'll submit a PR once I get things sorted out.

zoesn89, 2021-02-13:
> @weetmuts @zoesn89 I'm on it. I'm playing with this telegram: [...]

Nice, thank you! 😉

don-vip, 2021-02-14:
https://github.com/weetmuts/wmbusmeters/pull/248 should allow you to monitor your Hydrus meter.

weetmuts, 2021-02-21:
@zoesn89 Can you confirm that it works for you now?

zoesn89, 2021-02-21:
Unfortunately not -.-
I did the following: 

- make clean (thanks for that hint)
- ./configure && make && sudo make install
--> at make install, it says "nothing changed"?! normal?

After that, I tried: 
`chip@chip:~$ wmbusmeters --debug /dev/ttyUSB0:im871a:t1 Hydrus hydrus 81100028 NOKEY`
and I got:
`(wmbusmeters) version: 1.0.5-62-g05d52e0
(config) using device: /dev/ttyUSB0:im871a:t1 
(config) number of meters: 1
(main) configured "Hydrus" "hydrus" "81100028" not-encrypted
(lookup) with file "/dev/ttyUSB0"
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32517171000010001FF070232000100000000D01B"
(im871a) checkIM871AFrame "A5810617FF000300B32517171000010001FF070232000100000000D01B"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 1bd0 expected 1bd0
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(device config bytes) "FF000300B32517171000010001FF070232000100000000"
(im871a) config: other link_mode=t1 id=00101717 media=00 version=01 c_field=00 auto_rssi=01
(im871a) are you there? yes 00101717
(main) opening t1
Started config im871a[00101717] on /dev/ttyUSB0 listening on t1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32517171000010001FF070232000100000000D01B"
(im871a) checkIM871AFrame "A5810617FF000300B32517171000010001FF070232000100000000D01B"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 1bd0 expected 1bd0
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00101717
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) set config to set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
[... 105 lines cut ...]
(serial /dev/ttyUSB0) sent "A5010306000203300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(wmbus) disconnected /dev/ttyUSB0 im871a
(serialtty) closed /dev/ttyUSB0 (im871a)
(serial) TIMER thread interrupted
(serial) EVENT thread interrupted
(serial) event loop stopped!
(wmbus) deleted im871a
(serial) closing 1 devices`

The same is with the DES key instead of "NOKEY".

don-vip, 2021-02-21:
That's interesting, it appears the meter is in error. If I decode `1044242328001081640E70F002FD175000` using official IZAR library this is what I got:
```
ReadingData={
	rawMessage={
		frame=[10,44,24,23,28,00,10,81,64,0E,70,F0,02,FD,17,50,00], timestamp=1612657129739.000, moduleId=0
	},
	frameDescription={ type=FrameDescMBusRadio,
		meterId=UHYD640E81100028000,
		meterName=IZAR_RTE_MBUS_OM,
		telegramType=OPEN_METERING_ERROR,
		withAlarm=true,
		withError=true,
		meterDeviceNumber=81100028,
		mediumByte=0x0E,
		version=100,
		manufacturer=HYD,
		errorFlagBytes=0x00,
		telegramContentType=STANDARD,
		installFrame=false,
		synchronizeFrame=false,
		lField=0x10,
		cField=0x44,
		mField=[24,23],
		aFieldSecondaryAddress=[28,00,10,81,64,0E],
		ciField=0x70,
		encMode=UNKNOWN },
	telegramFormatError=false,
	parsingLevel=FULL_CLEARTEXT,
	measurements={
		UHYD640E81100028000 -> {
 			0 -> {
				values=[
					{ semantic=errorflags,	type=NUMBER,	isValid=true,	value=80 }
				]
			}
		}
	}
}
```

weetmuts, 2021-02-22:
Can you do a git pull. Clean rebuild. Then check
wmbusmeters --version
it should be something like
wmbusmeters: 1.1.0.....

zoesn89, 2021-02-23:
@weetmuts here is my output:
wmbusmeters: 1.1.0-7-g7b1c8df
7b1c8dfe7ef454bb78e57be529f355863375e086

I still have the same problems -.-

weetmuts, 2021-08-08:
@zoesn89 I looked at the telegram that could be found inside your log but alas, I can only agree with @don-vip that the telegram seems to be broken.

I hope you have solved your problem in some way, since this ticket is really old. :-)