wmbusmeters/wmbusmeters #224: Not receiving data for Techem Vario 4

duennwalder, 2021-02-01:
Hi,

I found this git while looking for the possibility to receive the readings from measuring unit which was installed in my flat the other day. 

I've installed and got this git running, but I don't see any data coming in. I'm using an usb cul made by busware. I'm quite sure that the receiver can receive the messages sent by the measuring unit, because I see them with another application, but it can't parse the incoming message. When I tail the wmbusmeters log the only content are the following lines:

```
(wmbusmeters) logging started 2021-02-01 19:42:56 using 1.0.5-6-g5c5050f
Started auto cul on /dev/ttyACM0 listening on t1
```

The exact name of the measuring unit is "Techem vario 4 Typ 4.1.2".

Is there any possibility to get things running?

Regards
Mark

weetmuts, 2021-02-02:
Please run: "wmbusmeters --debug auto:t1"
for a while, then post the log here. It should contain some (serial) data sent to and from the CUL dongle.
If the CUL receives any wmbus telegram , then it should send that data over the serial as well,
even though it might not get properly decoded.
 
Which other application are you using?

weetmuts, 2021-02-02:
@chester4444 have you tested the latest wmbusmeters with your nano-CUL? Does it work as it should?

chester4444, 2021-02-02:
i moved from nanoCUL to a dedicated esp8266-cc1101 solution
tailored to my multical21 water meter.

if i find my nanoCUL and some minutes of time, i hope i can test it.
(but i can't promise)

csr-1-0, 2021-02-02:
Hello,
as I already had contact with Fredrik for that reason, here are my results. Before that, a short description of my hardware.

- Raspberry Pi 4 (4Gb) the whole thing should later run on a Raspberry Pi 3+)
- CUL USB stick https://www.smarthome-agentur.de/produkt/nanocul-868-mhz-premium-wie-ftdi-cc1101-sma-transceiver-cul-stick-ccu2-fhem/, this works with one acquaintances with wmbusmeters
- Kamstrup Multical 21 Wirless m_bus mode C1 water meter (type 02140 ...)

**Nothing goes on with the command, stay silent:
(the following commands were tried with c1 and t1)**

`pi @ raspi: ~ $ wmbusmeters --listento = c1 / dev / ttyACM0`
` meters supplied. You must supply which link modes to listen to. 11 Eg. auto: c1`

**the same with:**

`pi @ raspi: ~ $ wmbusmeters auto: c1`
`Started auto cul on / dev / ttyACM0 listening on c1`

**Debug with c1 (keeps going, doesn't stop):**
Sorry, but I don't know why the line breaks in the code are not made correctly when I insert everything correctly? Fredrick, code extra by mail? Ask for help, that's totally confusing :-(, is there a spoiler for this or something?

> `pi@raspi:~ $ wmbusmeters --debug auto:c1
(wmbusmeters) version: 1.0.5
(config) number of meters: 0
(main) device /dev/ttyACM0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B9600) input(ICRNL|IXON) output(ONLCR|OPOST) control(CLOCAL|CREAD|CSIZE|HUPCL) local(ECHO|ECHOCTL|ECHOE|ECHOK|ECHOKE|ICANON|IEXTEN|ISIG) special_chars(3,28,127,21,4,1,17,19,26,18,15,23,22) modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyACM0) sent "FF0A02008077"
(serial /dev/ttyACM0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "3F2028FF20697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046204720682069204B204C206C204D206D20522054207420552075205620762057205820782059205A0D0A3F20280220697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046204720682069204B204C206C204D206D20522054207420552075205620762057205820782059205A0D0A"
(serialtty) closed /dev/ttyACM0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyACM0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a) are you there? no.
(serialtty) closed /dev/ttyACM0 (detect im871a)
(serial) before config: /dev/ttyACM0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect rc1180)
(serial /dev/ttyACM0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyACM0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyACM0 (detect rc1180)
(serial) before config: /dev/ttyACM0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (detect cul)
(cul) get version
(serial /dev/ttyACM0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "3F20280220697320756E6B6E6F776E2920557365206F6E65206F66204120422062204320652046204720682069204B204C206C204D206D20522054207420552075205620762057205820782059205A0D0A"
(cul) probe response "? (<02> is unknown) Use one of A B b C e F G h i K L l M m R T t U u V v W X x Y Z<0D><0A>"
(cul) get version
(serial /dev/ttyACM0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "5620312E36372043554C3836380D0A"
(cul) probe response "V 1.67 CUL868<0D><0A>"
(serialtty) closed /dev/ttyACM0 (detect cul)
[... 159 lines cut ...]
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) after config:  /dev/ttyACM0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyACM0 fd 3 (cul)
(main) regular reset of cul  on /dev/ttyACM0 will happen every 82800 seconds
(cul) set link mode t
(serial /dev/ttyACM0) sent "6272740A0D"
(serial) EVENT thread interrupted
(serial) received binary "544D4F44450D0A"
(cul) checkCULFrame "TMODE<0D><0A>"
(cul) no leading 'b' so it is text and no frame
(cul) received "TMODE"(serial /dev/ttyACM0) sent "5830310A0D"
(wmbus) no alarm (expected activity) for cul
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
[... cut ...] S)
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 4 (detect amb8465)
(serial) received binary ""`

---------------------------------------------------------------------------------------
I hope to do everything right here, if not, please let me know, I am actually "only" a user and can do what I am told, but I am still too unsure about further things.

Thank you
Christian

duennwalder, 2021-02-02:
Hello,

> Please run: "wmbusmeters --debug auto:t1"
> for a while, then post the log here. It should contain some (serial) data sent to and from the CUL dongle.
> If the CUL receives any wmbus telegram , then it should send that data over the serial as well,
> even though it might not get properly decoded.
> 

I've attached two files. One was running with "wmbusmeters --debug auto:t1" [wmbusmeters_debug_with_auto.txt](https://github.com/weetmuts/wmbusmeters/files/5912662/wmbusmeters_debug_with_auto.txt) and the other with "wmbusmeters --debug /dev/ttyACM0:t1" [wmbusmeters_debug_with_ttyACM0.txt](https://github.com/weetmuts/wmbusmeters/files/5912661/wmbusmeters_debug_with_ttyACM0.txt).

> Which other application are you using?
The application I used the cul previously with was FHEM ([https://fhem.de/](https://fhem.de/)), where I received some data but could not get it parsed. Everything is running on an rasberry pi 3.

BIBOLV, 2021-02-02:
@chester4444 can you elaborate on "dedicated esp8266-cc1101 solution"? I was actually thinking of building esp32+wmbus as I haven`t seen anything similar in market, but maybe I missed something?

chester4444, 2021-02-02:
[https://github.com/chester4444/esp-multical21](https://github.com/chester4444/esp-multical21)

of course its inspired from weetmuts great job done here - it only supports multical21, WMBUS C1.

SW: I did it with platformIO, it was really fun to get the first frames being decoded
HW: nodemcu (but any esp will do) with a SPI connected cc1101 - 2 minutes of soldering

weetmuts, 2021-02-04:
@chester4444 Neat solution! Do you think you could add a send USB packet feature? Because it is really easy to make wmbusmeters listen to a udp port if you have for example netcat installed.

wmbusmeters 'rtlwmbus:CMD(nc -lku 4444)'

then you can do for example:
`rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus | nc -u localhost 4444`

or you could transmit from th esp to the 4444 port on the computer where wmbusmeters is running.

weetmuts, 2021-02-04:
@duennwalder Interesting! Your debug file run with auto:t1 show that it detects configures and receives telegrams.
You can see that it prints "(cul) received full T1 frame" however then silence.
But when you run with /dev/ttyACM0:t1 it proceeds to (wmbus) parseDLL @0 51 etc.

Silly bug somewhere....

@csr-1-0 Can you test with "wmbusmeters /dev/ttyACM0:c1" to see if it properly proceeds to parsing printing?

csr-1-0, 2021-02-04:
> @ csr-1-0 Können Sie mit "wmbusmeters / dev / ttyACM0: c1" testen, ob der Parsing-Druck ordnungsgemäß ausgeführt wird?

` $ wmbusmeters / dev / ttyACM0: c1`
`You must supply at least one device (eg auto:c1) to receive wmbus telegrams.`

-------------

`$ wmbusmeters /dev/ttyACM0:c1`
` wmbus device detected, waiting for a device to be plugged in.`
` meters configured. Printing id:s of all telegrams heard!`

-------------

` $ ls -l /dev/serial/by-id` 
` insgesamt 0` 
` lrwxrwxrwx 1 root root 13 Feb  4 12:32 usb-SHA_CUL868-if00 -> ../../ttyACM0`

csr-1-0, 2021-02-09:
Hello,

short feedback on my part, i have now bought an iM871A-USB, and it works great with the Multical 21, as described by Fredrik.

The CUL will have to look for a new owner ;-) .

duennwalder, 2021-02-17:
@weetmuts I don't know what changed, but as for today and yesterday my cul recevied data which was parsed succesfully. I see files in the meter_readings-directory which contains data from various devices.

But i can't find my device in the data. I've received data from about 40 differnet devices, but the whether the id nor the readings are matching. What can I do now?

weetmuts, 2021-02-21:
Thanks @duennwalder  and @csr-1-0 ! I have examined the logs you sent, but I cannot really see a reason why the problems occured. I have a nano-cul and I have tested it a couple of times with the latest wmbusmeters and it seems to work.

The cul is a bit tricky since you have to load it with the right software to run wmbus, and you need the right
version of the software to listen to c1 mode, etc. Comparatively the im871a is a very nice reliable receiver.

@duennwalder , Yes the Techem Vario is not OMS compliant, so the it seems like the id extracted from the 
telegram is not the one printed on the meter. We had the same problem with izar meters, but @don-vip fixed this!

Perhaps you can examine the read-out values and compare with what is displayed on the meter?

I will close this issue, for now, please reopen an issue if you need more help finding your Vario meter!

closed 2021-02-21 by weetmuts

duennwalder, 2021-02-23:
@weetmuts Thank you very much.

But one last question. Till now I only had run
` wmbusmeters --debug /dev/ttyACM0:t1`

but now, after some research,I'm most certain, that it should be
` wmbusmeters --debug /dev/ttyACM0:c1`

but when I try to run this mode it will exit after a few second with
`(cul) received ""(cul) setting link mode(s) c1 is not supported for this cul device!`

What can I do to get things running?

chester4444, 2021-02-24:
https://github.com/weetmuts/wmbusmeters/issues/45#issuecomment-716424361

duennwalder, 2021-02-24:
> [#45 (comment)](https://github.com/weetmuts/wmbusmeters/issues/45#issuecomment-716424361)

Thank you very much, @chester4444  - that worked. I can now see raw data for my meter.

BenRoe, 2023-10-24:
If you have a CUL V3 from Busware with an Atmega32U4 chip, check this comment by michipi85 https://github.com/wmbusmeters/wmbusmeters/issues/45#issuecomment-733601146