wmbusmeters/wmbusmeters #225: Add support for ppm correction values

vistalba, 2021-02-02:
My rtl-sdr dongle has an offcet of 28ppm and I want to specify this as in other rtl_sdr commands. 
Is it possible to add this flag/option to wmbusmeters too?

weetmuts, 2021-02-02:
Sure you can specify the device in the conf file like this:

device=rtlwmbus:c1,t1:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus)

or from the command line:

wmbusmeters 'rtlwmbus:c1,t1:CMD(rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus)'

or using a pipe:

rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | rtl_wmbus | wmbusmeters stdin:rtlwmbus

weetmuts, 2021-02-02:
If you want to just tune the frequency, then there is a short cut:
device=rtlwmbus:868.9M

What is the ppm option you want to add?

vistalba, 2021-02-02:
I will test you suggested solution.

ppm is option `-p` in rtl_sdr.
ppm stands for parts per million and is the frequency offset error. (This value can be calculated with this tool: https://github.com/steve-m/kalibrate-rtl)

This is a problem with cheap dvb-t dongles.

weetmuts, 2021-02-21:
Closing this for now.

closed 2021-02-21 by vistalba

weetmuts, 2021-03-05:
@vistalba Short question!

The latest rtl_wmbus now supports listening to s1,t1 and c1 at the same time, alas the frequency then must be set to 868.625M therefore the -p [ppm] options seems like a better choice, than setting the absolute frequency.

What are the -p values that you have been using?

vistalba, 2021-03-08:
@weetmuts PPM `-p` option is to use a offset which is caused by cheap USB dongles. The problem is that they have an offset which differs based on the frequency. So "parts per million" does respect this offset which can be calibrated to tune to the correct frequency.

weetmuts, 2021-03-08:
Yes, I was curious about the exact value you have used for your dongle. Is it 7, 17, 177, 17000 etc...

The latest wmbusmeters on github now has support for setting the ppm. Try this:
wmbusmeters 'rtlwmbus(ppm=177)'
 
or 
device=rtlwmbus(ppm=177)

in the conf file.

weetmuts, 2021-03-08:
You need to install the latest rtl_wmubs as well, since wmbusmeters by default now uses the -s setting which enables it to receive S1,C1 and T1 telegrams at the same time.

vistalba, 2021-03-08:
This can vary a lot... one of my usb dongles have below 1 so no correction is needed. Another one has a offset of 28ppm.
Okay. Perfect.