wmbusmeters/wmbusmeters #226: Add support for unknown Hydrometer type 0x68

don-vip, 2021-02-02:
I just setup wmbusmeters and listened to my neighborhood (recent constructions, so recent hardware).
In auto:t1 mode I see 16 of these:
```
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Unknown (0x68)
                   ver: 0x93
                device: rtlwmbus[00000001]
                  rssi: 125 dBm
                driver: unknown!
```

The 16 addresses are:
```
Received telegram from: 19760420
Received telegram from: 19780420
Received telegram from: 19790420
Received telegram from: 19840420
Received telegram from: 20040420
Received telegram from: 20280420
Received telegram from: 20290420
Received telegram from: 20390420
Received telegram from: 20410420
Received telegram from: 59730420
Received telegram from: 59780420
Received telegram from: 60250420
Received telegram from: 60290420
Received telegram from: 60400420
Received telegram from: 60730420
Received telegram from: 61020420
```

The 10 first telegrams:
```
telegram=||1944304CEC2732212404A2531D00135FC7E159731CC9CF340195|+9
telegram=||534424232004781993687A1E0045F671EDE7D8701937F0EF63B6A6E65F75F46F2B589DA6666B7C38A99F116E47CDEA71350709657E97B97D08422C226EF3CCB64F8E5D11A02BDF9139FF9C5EA3447FFB46794AB0|+9
telegram=||24442D2C37097767300D8D2050C2A11F21162D55747AC7759057FFEB7C7B6132709537782F|+14
telegram=||534424232004292093687AAA0045773BBB4C9B187CAA33FDC8A7B32E8530AD7ABDA7C52641579DE8B71145EAB2E2C361420CDFA1D8964C4CA2B88BAA9E9228A4CF84612B63391A7313376D2E676F7B3E0611A3BF|+14
telegram=||534424232004406092687AE40045D0549EB06EA115633E12609DE37BD6D88EBD8D16EF4F99640123327FF5C451CC24AA76144FFFBCFFCB8007E755CE78D844AFBE6EE3159DFC730ADA0319A3AD4AEB2320F2F95C|+14
telegram=||534424232004256092687A8000459E22B391158F98E02D9F2FFEBBF0A2969B8D21F5BB4E1093B493C2E2B07877D6EAE1068CAC89BA74ED3CC304BB2971070CAAB55A8E973F72BEE3446508F29C4E71D982FE826C|+17
telegram=||1944304C88CF0321CC01A1711F001368B8C644118AE898EE5866|+23
telegram=||534424232004412093687A8100459042E9639B6078FB1DDA2A61ED2C62DEA27A8E4E8061246389B30E9265923E5372627599FA20D5E0CE683869DAE0CEF2653DEACD1925D5E59F63B041DB1E677F402310A93D81|+23
telegram=||534424232004042093687A5E0045161A12489FD378EFB69BDD4749B3C0C700200FAADDD382F87CBCFB0CA0913A7128AE5099F3B929FAC5CCDAC262FA5A492D9E64E05B91D71E4DB5EB60C0FBCD923E1CA32ECBFA|+24
telegram=||534424232004735992687A610045536EF6B9CB0E72384BC57EE6D89B4191AC3C21F3E0F69F77B67D08647226ED4C7515C9574BF2A6B62EE6099F57E72AF9181544107C4FE279751434212B03E33F99FA23CF9970|+28
```

```
(serial) received ascii "T1;1;1;2021-02-02 21:43:24.000;126;89;21315FDB;0x1944304cdb5f31212404a2031d00133f9e1f034e46d09adaa5ef<0A>"
(serial) received ascii "T1;0;0;2021-02-02 21:43:24.000;80;41;60290420;0x534424232004296092687a730045da2c30843670387833aabba33334074cb079e6af1268179c11b38dd2fcea2c528a1666528de8aa9cce63c7a108bba2227f9a26ede0cbc0ca2effc101b3f1ff23ffe9ff64ffff<0A>"
(serial) received ascii "T1;1;1;2021-02-02 21:43:25.000;62;65;61020420;0x534424232004026192687afc0045140adb79caa6635861be6e15650c8057697f36e32357df1c9b691e61ea75d797df31d5496d4c994b1d0ced6e4e411f9e67c9615f746e8db183983c394a9d875682f5ed5c3a1e<0A>"
(serial) received ascii "T1;1;1;2021-02-02 21:43:28.000;52;39;213227EC;0x1944304cec2732212404a2531d00135fc7e159731cc9cf340195<0A>"
(serial) received ascii "T1;1;1;2021-02-02 21:43:28.000;44;50;19780420;0x534424232004781993687a1e0045f671ede7d8701937f0ef63b6a6e65f75f46f2b589da6666b7c38a99f116e47cdea71350709657e97b97d08422c226ef3ccb64f8e5d11a02bdf9139ff9c5ea3447ffb46794ab0<0A>"
(serial) received ascii "T1;1;1;2021-02-02 21:43:28.000;53;66;20410420;0x534424232004412093687a8000459f42e8116ffacae620479bf1c618d9b3c8562a66a531f9822fbaf330ebe1b786ad6ea3d9722552d2ff3303464ecc18677d5744a5c9e88b363ecbe97fbc6019db4e38332e2c88<0A>"
(serial) received ascii "T1;0;0;2021-02-02 21:43:29.000;26;24;FF01FF20;0x5344242320ff01ffffff7ae40045ff09509898fff524f25a4bffd22f5a549effff4fffbacce7d1ffcecfea4b7a05c2e0ffffa3ffffff6ffff657ff917dff9da64de5a9326bf16cb42485ff0e74ffa5ff40263bff<0A>"
(serial) received ascii "T1;0;0;2021-02-02 21:43:29.000;44;46;20040420;0x534424232004042093687a5d00457f1a11df827caec8f03d0ff734ee0d70be55e3d2b222e4da96a6fdeb3205a00f49bb2aff6bb7a0ac962197b2de8f21f6052f96592ac68b6baf4d002269794e7e2f31c7a7f9e1<0A>"
(serial) received ascii "T1;1;1;2021-02-02 21:43:29.000;99;98;20390420;0x534424232004392093687a5900451f2c4bd7bd9e92f16d9b9fbc076a28b1cf153a17a446d7f1cf094e8c638c4d123b192c329c96b22469949dae20bfe0930efc34fa85172832ecac7a82c8c861b71ba47957e29f<0A>"
(serial) received ascii "T1;1;1;2021-02-02 21:43:33.000;95;61;213227E7;0x1944304ce72732212404a2731d0013dfb5236ef0739007cb6a81<0A>"
```

I don't think I will be able to know for sure what model it is, as the addresses I get do not match anything I could get my hand to in my residential building. Is the log file with telegrams enough to add support for those? If I can do anything to help you, just let me know.

[stderr.zip](https://github.com/weetmuts/wmbusmeters/files/5914783/stderr.zip)

don-vip, 2021-02-04:
I think they are the heat meters I was looking for, the [SHARKY 775](https://www.diehl.com/metering/en/portfolio/thermal-energy-metering-/thermal-energy-metering-product/sharky-775/63755/) from Diehl Metering. I have physical access to them.

I believe the telegram format for the software version I have is [this one](https://www.diehl.com/cms/files/254302-FR-EN-SHARKY_775_CommunicationDescription_v1.4_EN.pdf?download=1).

[Older version](https://www.wms.nl/application/files/4615/0607/4805/Sharky775_M-Bus_telegram_E.pdf) of this document shows the meter was originally built by Hydrometer, before the Diehl acquisition.

https://github.com/rscada/libmbus/issues/145 might help to add support for this meter.

don-vip, 2021-02-04:
Is https://github.com/weetmuts/wmbusmeters/blob/master/HowToAddaNewMeter.txt up to date?
I created a simulation file [sharky.txt](https://github.com/weetmuts/wmbusmeters/files/5929001/sharky.txt) but cannot replay it:

```
$ wmbusmeters sharky.txt
Started config rawtty on sharky.txt listening on any
No meters configured. Printing id:s of all telegrams heard!
```

don-vip, 2021-02-07:
I managed to download, decompile and run the IZAR@NET decryption library on my frames. Although the device is capable (and supposed) to emit in AES, the heat meters of my building emit in "real data" mode instead of OMS mode. It means the encryption is based on [LSFR](https://en.wikipedia.org/wiki/Linear-feedback_shift_register) (same algorithm as IZAR, with same PRIOS keys: 
```java
PRIOS_CIPHER_DEFAULT.add(HexString.getByteArray("39BC8A10E66D83F8"));
PRIOS_CIPHER_DEFAULT.add(HexString.getByteArray("51728910E66D83F8"));
```
My meter uses the second key. The frame can be decoded as follows:
```java
package com.diehl.metering.izar.module.readout.impl.java.mbus;

import static org.junit.jupiter.api.Assertions.assertEquals;
import org.junit.jupiter.api.Test;
import com.diehl.metering.izar.module.common.api.v1r0.bean.RawMessage;
import com.diehl.metering.izar.module.common.api.v1r0.hexstring.HexString;

class InterpretMBusRadioImplTest {

	@Test
	void testFrame() {
		assertEquals(("ReadingData={\n"
				+ "	rawMessage={\n"
				+ "		frame=[53,44,24,23,20,04,25,60,92,68,7A,37,00,45,75,22,35,85,4D,EE,EA,59,39,FA,D8,1C,25,FE,EF,5A,23,C3,8F,B9,16,84,93,C5,63,F0,8D,B1,0B,AF,87,F6,60,FB,A9,12,96,BA,23,97,E8,F4,22,0B,86,D3,A1,92,FB,51,E0,BF,CF,24,DC,E7,21,18,E0,C7,5A,9E,89,F4,3B,DF,E3,70,82,4B], timestamp=1612657129739.000, moduleId=0\n"
				+ "	},\n"
				+ "	frameDescription={ type=FrameDescMBusRadio,\n"
				+ "		meterId=UHYD200468926025000,\n"
				+ "		meterName=SHARKY_3_R4,\n"
				+ "		telegramType=REAL_DATA,\n"
				+ "		withAlarm=false,\n"
				+ "		withError=false,\n"
				+ "		meterDeviceNumber=68926025,\n"
				+ "		mediumByte=0x04,\n"
				+ "		version=32,\n"
				+ "		manufacturer=HYD,\n"
				+ "		errorFlagBytes=0x00,\n"
				+ "		telegramContentType=STANDARD,\n"
				+ "		installFrame=false,\n"
				+ "		synchronizeFrame=false,\n"
				+ "		lField=0x53,\n"
				+ "		cField=0x44,\n"
				+ "		mField=[24,23],\n"
				+ "		aFieldSecondaryAddress=[20,04,25,60,92,68],\n"
				+ "		ciField=0x7a,\n"
				+ "		encMode=LFSR },\n"
				+ "	telegramFormatError=false,\n"
				+ "	parsingLevel=FULL_CIPHERED,\n"
				+ "	measurements={\n"
				+ "		UHYD200468926025000 -> {\n"
				+ " 			0 -> {\n"
				+ "				values=[\n"
				+ "					{ semantic=energy,	type=QUANTITY,	isValid=true,	value=2651000W·h },\n"
				+ "					{ semantic=energy.tariff[1],	type=QUANTITY,	isValid=true,	value=0W·h },\n"
				+ "					{ semantic=volume,	type=QUANTITY,	isValid=true,	value=150.347㎥ },\n"
				+ "					{ semantic=volume.tariff[2],	type=QUANTITY,	isValid=true,	value=0.018㎥ },\n"
				+ "					{ semantic=errorflags,	type=NUMBER,	isValid=true,	value=0 },\n"
				+ "					{ semantic=volume.flow,	type=QUANTITY,	isValid=true,	value=0.000m³/h },\n"
				+ "					{ semantic=power,	type=QUANTITY,	isValid=true,	value=0W },\n"
				+ "					{ semantic=flow.temperature,	type=QUANTITY,	isValid=true,	value=42.3°C },\n"
				+ "					{ semantic=return.temperature,	type=QUANTITY,	isValid=true,	value=28.1°C },\n"
				+ "					{ semantic=temperature.difference,	type=QUANTITY,	isValid=true,	value=14.1°C }\n"
				+ "				]\n"
				+ "			}\n"
				+ "		}\n"
				+ ",\n"
				+ "		UHYD200468926025001 -> {\n"
				+ " 			0 -> {\n"
				+ "				values=[\n"
				+ "					{ semantic=volume,	type=QUANTITY,	isValid=true,	value=0.000㎥ }\n"
				+ "				]\n"
				+ "			}\n"
				+ "		}\n"
				+ ",\n"
				+ "		UHYD200468926025002 -> {\n"
				+ " 			0 -> {\n"
				+ "				values=[\n"
				+ "					{ semantic=volume,	type=QUANTITY,	isValid=true,	value=0.000㎥ }\n"
				+ "				]\n"
				+ "			}\n"
				+ "		}\n"
				+ "	}\n"
				+ "}").replace("\r", "").replace("\n", ""),
				new InterpretMBusRadioImpl().interpretFull(new RawMessage(
						HexString.getByteArray("534424232004256092687A370045752235854DEEEA5939FAD81C25FEEF5A23C38FB9168493C563F08DB10BAF87F660FBA91296BA2397E8F4220B86D3A192FB51E0BFCF24DCE72118E0C75A9E89F43BDFE370824B"),
						1612657129739L)).toStringHumanReadable().replace("\r", "").replace("\n", ""));
	}
}
```

I will create a PR to add decoding support for SHARKY. Need to look how much IZAR code I can reuse. Probably the distinction between "real data" and standard OMS mode will require a modification to core.

weetmuts, 2021-02-07:
Please do not add decryption keys to the issues. I have removed this post. Please re-open issue without posting keys.