wmbusmeters/wmbusmeters #228: new install not working

rattinca, 2021-02-04:
I have an old version that works.
tried to install new version (on a new machine), "./configure && make && make test" says everything is fine. But when i start wmbusmeters i just get a lot of dots and some random letters and it gets stuck.
Im at a loss, what can i do?

weetmuts, 2021-02-04:
Which GNU/Linux version are you using?

rattinca, 2021-02-04:
armv7l GNU/Linux

Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              1
On-line CPU(s) list: 0
Thread(s) per core:  1
Core(s) per socket:  1
Socket(s):           1
Vendor ID:           ARM
Model:               2
Model name:          Cortex-A8
Stepping:            r3p2
BogoMIPS:            996.14
Flags:               half thumb fastmult vfp edsp thumbee neon vfpv3 tls vfpd32

conected via coolterm. i just realized, the random stuff might be a construct of coolterm, but something definitely goes wrong.

wmbusmeters --debug --verbose auto:t1
(wmbusmeters) version: 1.0.5-22-gfa87570
(config) number of meters: 0
(main) device /de........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................–.xÄÜ.f....òò.òòÜû.f~`.ÜÜÜ.òûf.~.f~`.ÜÜÜ.ò‡Êû¯û`Ü¯Êû‡ûÄÜ..ò˛ò.f.ò‡ÊÄÜòf`..`‡ÊÄÊòfòòf.fÜ..û˛û.f.ûÄÜÜÜ.ò¯.fx.fû..û˛f‡.û.Ü¯Ê¯.ÜÜÜ.òûÊû`fxÊûÄÜ`.f.fÜò.f.fÊÄ.f.f.`.fòfÜòÄÄÜ.fx.fû..û`. etc. and then its stuck and i have to reboot.

might auto be probing something that does not like it?
How do i find out which tty to use if i cant use auto?

weetmuts, 2021-02-04:
Which wmbus device are you using?

Can you build a debug version of the binary, like this: make DEBUG=true
Then try ./build_debug/wmbusmeters --debug /dev/ttyUSBxx:yourdevice

rattinca, 2021-02-10:
It seems it was tty0, not sure whats there, but it does not want to be touched. its working now.

weetmuts, 2021-02-12:
Does not want to be touched for sure! :-)
Maybe /dev/tty0 is the serial port you are accessing the device through?

Anyway, you can avoid probing it:
--donotprobe=/dev/tty0
or just
donotprove=/dev/tty0 
in the conf file.

Closing this for now!

closed 2021-02-12 by weetmuts