wmbusmeters/wmbusmeters #229: rtl_wmbus S1 supported now

alalons, 2021-02-04:
xaelsouth just adedd S1 mode to rtl_wmbus. Same command but 868,3M as frequency.
Tried to modify the source but still not familiar with the structure.
Could you integrate S1 mode?

Thank you

weetmuts, 2021-02-04:
Cool, will help you. Do you have an S1 meter to test with?

don-vip, 2021-02-04:
I have [this one](https://www.diehl.com/cms/files/212612-FR-EN-IZAR-RC-i-R4.pdf?download=1) and I think it emits in S1 as I cannot hear it in T1. I will check directly with rtl_wmbus.

alalons, 2021-02-04:
Yes I have.

Álvaro Alonso
Gerente
Arvacalor S.L.

alalons, 2021-02-05:
Tried my best to integrate S1 mode. 
Always listens to all modes, whether I select S1 or C1 does not matter.
So long without coding a line!!
Pull requested.
rtl_wmbus must be up to date with latest changes

weetmuts, 2021-03-06:
I have now pushed support for the latest rtl_wmbus to master. 

Run: wmbusmeters rtlwmbus
and it will default to s1,t1,c1 mode.

Run: wmbusmeters 'rtlwmbus(ppm=17)'
Listen to s1,t1,c1 and tune the dongle correctly, the above will add "-p 17" to the rtl_sdr command line.

Run: wmbusmeters rtlwmbus:433M
and it will listen only on 433MHz (ie no -s switch)

Can you test and verify that it works?
Can you provide an example S1 telegram here?

Thanks!

alalons, 2021-03-06:
There is some problem with received s mode telegrams. C mode is parsed ok and printed onscreen with logtelegrams.
S mode is received but does not log them with logtelegrams.

alvaro@alvaro-Compaq-Presario-CQ61-Notebook-PC:~/wmbusmeters-master$ wmbusmeters --logtelegrams --verbose --debug rtlwmbus
(wmbusmeters) version: _
(config) using device: rtlwmbus 
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 8526 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 8528 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(serial) EVENT thread interrupted
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on none
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 8534 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "S1;1;1;2021-03-06 21:22:29.000;126;127;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 21:22:29.000;65;128;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 21:22:29.000;126;127;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 21:22:29.000;65;128;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) only text
(serial) received ascii "S1;1;1;2021-03-06 21:22:50.000;125;125;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 21:22:50.000;57;131;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 21:22:50.000;125;125;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 21:22:50.000;57;131;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) only text
^C(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) EVENT thread interrupted
(serial) event loop stopped!
(shell) sending SIGTERM to process group 8534
(bgshell) sent sigterm, now waiting for child 8534 to exit.
(serial) TIMER thread interrupted
(bgshell) 8534 terminated due to signal 15
(wmbus) disconnected 00000001 rtlwmbus
(serialcmd) closed /bin/sh pid=8534 fd=5 (rtlwmbus)
(wmbus) deleted rtlwmbus
(serial) closing 1 devices

weetmuts, 2021-03-06:
Ah, fix pushed! Try again!

weetmuts, 2021-03-06:
What is the product/brand name of that heat cost allocator?

alalons, 2021-03-06:
It’s a Qundis demo transmiter. It is used to test Rf signal level from radiator ubication to the concentrator when planning an installation.

[image]
[image]

alalons, 2021-03-06:
fixed 

alvaro@alvaro-Compaq-Presario-CQ61-Notebook-PC:~/wmbusmeters-master$ wmbusmeters --logtelegrams --verbose --debug rtlwmbus
(wmbusmeters) version: _
(config) using device: rtlwmbus 
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 5038 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(serial) EVENT thread interrupted
(shell) output: >>><<<
(shell) return code 0
(shell) 5040 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(serial) EVENT thread interrupted
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on none
(rtlwmbus) using command: /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "/usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>/dev/null | /usr/bin/rtl_wmbus -s"
(serialcmd) opened /bin/sh pid 5046 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "S1;1;1;2021-03-06 22:49:33.000;124;126;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 22:49:33.000;58;126;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 22:49:33.000;124;126;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 22:49:33.000;58;126;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 42
(telegram) DLL L=29 C=44 (from meter SND_NR) M=3265 (LSE) A=04998541 VER=01 TYPE=08 (Heat Cost Allocator) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=124
(wmbus) parseELL @10 32
(wmbus) parseAFL @10 32
(wmbus) parseTPL @10 32
(dvparser) found new format "82046C8B046E046D02FDAC7E01FD0C" with hash 511e, remembering!
(telegram) TPL CI=7a ACC=00 STS=80 CFG=0000 ()
Received telegram from: 04998541
          manufacturer: (LSE) Landis Staefa electronic (0x3265)
                  type: Heat Cost Allocator (0x08)
                   ver: 0x01
                device: rtlwmbus[00000001]
                  rssi: 124 dBm
                driver: unknown!
(wmbus) 00: 29 length (41 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 6532 dll-mfct (LSE)
[... 20 lines cut ...]
(wmbus) 22: AC vife (Duration since last readout [second(s)])
(wmbus) 23: 7E vife (Reserved)
(wmbus) 24: 0123
(wmbus) 26: 01 dif (8 Bit Integer/Binary Instantaneous value)
(wmbus) 27: FD vif (Second extension FD of VIF-codes)
(wmbus) 28: 0C vife (Model/Version)
(wmbus) 29: 01
telegram=||294465324185990401087A0080000082046C7F018B046E210300046D1A0E6F0202FDAC7E012301FD0C01|+31
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 22:49:33.000;58;126;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
(serial) received ascii "S1;1;1;2021-03-06 22:49:54.000;125;128;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 22:49:54.000;63;124;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 22:49:54.000;125;128;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 22:49:54.000;63;124;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 22:49:54.000;63;124;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
(serial) received ascii "S1;1;1;2021-03-06 22:50:15.000;122;124;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 22:50:15.000;67;129;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 22:50:15.000;122;124;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>S1;1;1;2021-03-06 22:50:15.000;67;129;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
(rtlwmbus) checkRTLWMBusFrame "S1;1;1;2021-03-06 22:50:15.000;67;129;04998541;0x294465324185990401087a0080000082046c7f018b046e210300046d1a0e6f0202fdac7e012301fd0c01<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.

weetmuts, 2021-05-22:
I believe S mode telegrams can be heard now. Please open a new issue if you need help with the particular HCA. Thanks!

closed 2021-05-22 by weetmuts