wmbusmeters/wmbusmeters #231: Diehl Altair watermeter IZAR : Unknown tpl-ci-field a3

alafanechere, 2021-02-05:
Hello there,
I'm trying to receive mbus data from my cold water meter (which is a Diehl Altair on which an Izar dongle seems to be plugged, but I cannot figure out the exact model). I collect payload from my area (with my USB dongle RTL2838UHIDIR) but I'm not sure that they are coming from my own meter, and they seems to not properly parsed by `wmbusmeters`.

Here's a picture of my meter. Do you think the Tx S/N label is my meter ID ? 
[image]

Moreover I encounter a `Unknown tpl-ci-field a3` error when running the following command : 
```bash
$ ./build/wmbusmeters --format=json --debug rtlwmbus --listento=t1 izar '*' NOKEY
``` 

<details><summary> Command output :  </summary><p>

```bash
wmbusmeters (ad12261*) » ./build/wmbusmeters --format=json --debug rtlwmbus --listento=t1 izar '*' NOKEY                                              ~/homespace/wmbusmeters 1 ↵
(wmbusmeters) version: HEAD_1.0.4
(config) using device: rtlwmbus
(config) number of meters: 1
(main) configured "--listento=t1" "izar" "*" not-encrypted
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr --help < /dev/null"
(shell) output: >>>rtl_sdr: illegal option -- -
rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
(shell) return code 1
(shell) 35667 exited
(serial) EVENT thread interrupted
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 2
(shell) 35669 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on none
(rtlwmbus) using command: rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "rtl_sdr -d 0 -f 868.95M -s 1.6e6 - 2>/dev/null | rtl_wmbus"
(serialcmd) opened /bin/sh pid 35671 fd 5 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) received ascii "T1;0;0;2021-02-05 14:07:20.000;49;171;FFFFFF9E;0x19ffff4c9effffff0cffa373ffff135f064e48ffff87c89142ff<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2021-02-05 14:07:20.000;49;171;FFFFFF9E;0x19ffff4c9effffff0cffa373ffff135f064e48ffff87c89142ff<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;1;1;2021-02-05 14:07:30.000;33;81;21245899;0x1944304c995824210c04a3131400133a0434ae882375e9077f18<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2021-02-05 14:07:30.000;33;81;21245899;0x1944304c995824210c04a3131400133a0434ae882375e9077f18<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 26
(meter) --listento=t1: for me? 21245899
(meter) --listento=t1: meter detection did not match the selected driver izar! correct driver is: unknown!
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) --listento=t1: yes for me
(meter) --listento=t1 izar handling telegram from 21245899
(meter) --listento=t1 21245899 "1944304C995824210C04A3131400133A0434AE882375E9077F18"
(wmbus) parseDLL @0 26
(telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=21245899 VER=0c TYPE=04 (Heat meter) (driver unknown!) DEV=rtlwmbus[00000001] RSSI=33
(wmbus) parseELL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(wmbus) Unknown tpl-ci-field a3
(wmbus) telegram from 21245899 ignored by all configured meters!
```
</p></details>

I'm using @weetmuts for of `rtl-wmbus` on branch `FixLengthByte` as suggested by @Domochip who had a [similar issue in the past](https://github.com/weetmuts/wmbusmeters/issues/93).

So here are my question : 
* Does someone is aware of my meter model / id according to the picture I attached (I know its a bit out of the scope of this repo...)
* Is there something I can do to fix the `Unknown tpl-ci-field a3` error ? 

Thanks for your help, I hope I'm not off topic, and I'd like to congrats the contributor for this tool which seems really powerful !

alafanechere, 2021-02-05:
For easier debugging here are some received telegram by `rtl_wmbus`: 
```
T1;1;1;2021-02-05 16:20:26.000;55;99;2124589C;0x1944304c9c5824210c04a363140013716577ec59e8663ab0d31c
T1;1;1;2021-02-05 16:20:35.000;53;128;213234A3;0x1944304ca33432210c04a303150013149902d7493ade09ef2757
T1;1;1;2021-02-05 16:20:54.000;42;104;21245899;0x1944304c995824210c04a3331400133a2a6a3dde60d25ab14d1d
T1;1;1;2021-02-05 16:21:00.000;66;98;2124589C;0x1944304c9c5824210c04a373140013717258a5f2c9b5e36bca1e
T1;1;1;2021-02-05 16:21:12.000;59;145;213234A3;0x1944304ca33432210c04a313150013148e2d9ee21b0dd0343e55
T1;1;1;2021-02-05 16:21:26.000;39;52;21245899;0x1944304c995824210c04a3431400133a4fa6c18f85e957b00210
T1;1;1;2021-02-05 16:21:32.000;59;146;2124589C;0x1944304c9c5824210c04a30314001371179459a32c8eee6a8513
T1;1;1;2021-02-05 16:21:44.000;41;109;213234A3;0x1944304ca33432210c04a32315001314b75c441f7979ba591552
T1;1;1;2021-02-05 16:22:02.000;43;64;21245899;0x1944304c995824210c04a3531400133a58898824a43a8e6b1b12
T1;1;1;2021-02-05 16:22:10.000;67;166;2124589C;0x1944304c9c5824210c04a3131400137100bb10080d5d37b19c11
T1;1;1;2021-02-05 16:22:17.000;61;128;213234A3;0x1944304ca33432210c04a33315001314a0730db458aa63820c50
T1;1;1;2021-02-05 16:22:36.000;40;50;21245899;0x1944304c995824210c04a3631400133a61f852d9c64ee4063015
T1;1;1;2021-02-05 16:22:45.000;67;132;2124589C;0x1944304c9c5824210c04a3231400137139cacaf56f295ddcb716
T1;1;1;2021-02-05 16:22:50.000;21;29;213234A3;0x1944304ca33432210c04a34315001314c5bff1e5bd916e83435d
T1;1;1;2021-02-05 16:23:08.000;37;58;21245899;0x1944304c995824210c04a3731400133a76d71b72e79d3ddd2917
```

don-vip, 2021-02-07:
> For easier debugging here are some telegrams received by `rtl_wmbus`:

Can you please share some telegrams using wmbusmeters `--logtelegrams` option? I think I should be able to add yours to the test data set I'll write when implementing #226.

weetmuts, 2021-02-07:
It is easy to rewrite the rtlwmbus output into simulation_foo.txt output, just copy the hex bytes after 0x and place the hex between pipes in the line: telegram=|1944304c...|
You can also read the rtlwmbus file directly just run wmbusmeters, put your T1;1... lines into foo.txt and run:
`wmbusmeters foo.txt:rtlwmbus`

So your telegrams show yet another weird diehl/sappel/izar meter type/version, this one reports itself
as a heat meter, oh well. now we have seen izar water meters reporting as oil, circuit breaker and heat.
:-) Anything but water.

However what blocked the decoding was that they use yet another tpl ci field, which I can only assume means
manufacturer specific. Oh well, so I added A3 and then it decodes nicely with the izar driver.

`wmbusmeters --format=json simulation_izar_again.txt Vatten izar 2124589c NOKEY`

{"media":"water","meter":"izar","name":"Vatten","id":"2124589c","total_m3":38.944,"last_month_total_m3":38.691,"last_month_measure_date":"2021-02-01","remaining_battery_life_y":10,"current_alarms":"no_alarm","previous_alarms":"no_alarm","timestamp":"2021-02-07T07:31:22Z"}

Or you can decode all your telegrams like this:
`wmbusmeters foo.txt:rtlwmbus Water izar '*' NOKEY`

You will have to monitor the water consumption to figure out which meter id is yours, since I do not know of any way of correlating the numbers printed on the device itself, with the numbers is transmits. Yet another izar quirk. 

Fix pushed, please test!
on the

alafanechere, 2021-02-07:
You're awesome, your fix allows me to read the telegrams now. I indeed found my meter ID according to my water consumption, the ID is not printed on the meter itself. 
Thanks for your tips and availability @weetmuts and @don-vip !

closed 2021-02-07 by alafanechere