wmbusmeters/wmbusmeters #238: MQTT Meter id Topic?

theWagNerd, 2021-02-06:
Hi,

I've had wmbusmeters running successfully for a month - currently decoding telegrams from a few meters and outputting the data to discrete water meter CSV logs.

I'd like to utilise the MQTT functionality to also publish some of the data to econcms (local). Would it be possible to incorporate meter ids as sub topics in the MQTT shell command in the config?

Hope I've explained myself okay, but to illustrate.... If a telegram from meter1 is decoded, it would publish to say emon/meter1; if the telegram is from meter2, it would publish to say emon/meter2 etc....

Or if there are other solutions available that could do this, that would be great.

Thanks

BIBOLV, 2021-02-06:
Hi,

Isn't that what's being done by this that is shown as example in documentation?

`shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "$METER_JSON"`

theWagNerd, 2021-02-07:
Thanks @BIBOLV , it looks like it from the construct - I'm unfamiliar with the JSON format so will go and learn. My wmbusmeters currently outputs to semi colon format files which I then data import to Excel so that'll no doubt have to change (less of an issue if I'm publishing to econcms).

Sounds like this non-issue can be closed

Cheers

closed 2021-02-07 by theWagNerd