wmbusmeters/wmbusmeters #239: Add support for Diehl/Hydrometer SHARKY 775

weetmuts, 2021-02-07:


weetmuts, 2021-02-07:
@don-vip Thank you very much for contributing! Please contact me at oehrstroem@gmail.com if you have any questions!

don-vip, 2021-02-07:
Hi @weetmuts ! Sorry for having shared something that I shouldn't have.

One explanation concerning the IZAR device types "oddities": In fact in several (and numerous) conditions, the telegrams deviate from the OMS spec concerning the A field. Instead of having Serial number on first 4 bytes, then version, then device type, many IZAR meters will transmit first version, then device type, then serial number. That's why you find so many "exotic" device types. In fact they use the proper device type but a an unexpected position. Example for my heat meter

`telegram=|534424232004256092687A37004575|2235854DEEEA5939FAD81C25FEEF5A23C38FB9168493C563F08DB10BAF87F660FBA91296BA2397E8F4220B86D3A192FB51E0BFCF24DCE72118E0C75A9E89F43BDFE370824B|`

"A field" is transmitted as `200425609268` which implies a device type of 0x68 which does not exist, a serial number that doesn't match the one printed on the device, and a version which differs between several identical meters.

I should been read as `256092682004` which implies an heat meter (0x04: correct), a version of 0x20 identical across identical meters, and a serial number that matches the ones printed on the meter.

I will try to create a first PR to handle this mode I'll call "swapped mode". Conditions to interpret how to read the A field are based on C field, M field, CI field and a flag in CFG field. Pseudo algorithm for IZAR devices:

```
if C field == 0x68 or 0x70: // SND-NR or ?
  switch CI field:
    case 113: // 0x71
      // swapped mode
    case 122: // 0x7A
      if flag set in CFG field:
        // swapped mode
      else:
        // normal mode
    case 160, 161, 162, 163, 164, 165, 166, 167: // 0xA0-0xA7
      if prios (M field in "HYD", "EWT", "DME", "SPL", "SAP"):
        // swapped mode for "HYD", "EWT", "DME", "SPL". Not sure yet for "SAP" code, custom SAP logic involved
      else:
        // normal mode
    case 176: // 0xB0
      if M field == "SAP":
        // don't know yet, custom SAP logic involved
      else:
        // normal mode
    case 177, 178, 179: // 0xB1-0xB3
        // swapped mode
    case 168, 169, 170, 171, 172, 173, 174, 175, 180, 181, 182, 183: // 0xA8-0xAF + 0XB4-0xB7
    default:
        // normal mode
```

weetmuts, 2021-02-07:
Thank you! This explains alot! Now, the questions is how to integrate it with wmbusmeters.....
Hmm.  I would prefer that the generic wmbus standard header decoding code is not modified 
to deal with quirks from non-compliant meters. After all wmbusmeters is about implementing
the wmbus standard. :-)

So the cleanest way would be to run a preProcesser that fixes such telegrams to be compatible with
the standard...in this case it seems pretty easy, shuffling the A field bytes around.

Perhaps something like this:
1) Add the function preProcess to the Telegram class.
2) Invoke preProcess() just before parseDLL in Telegram::parse and Telegram::parserHeader 
3) preProcess will invoke some set of hard coded fixup functions that will decide based on some bytes
if and how it should change the telegram. The original bytes are stored in a vector Telegram::original
that is usually empty. Only a limited number of bytes need to be stored.
4) Decoding proceeds normally, the detailed --debug info works and logs on the modified telegram data, 
unfortunately, but it is not so much we can do about that.
5) The final logTelegram must be adjusted to log the original telegram, so if the Telegram::original is 10 bytes,
then it will print those first, then print the normal parsed data, skipping the original.

don-vip, 2021-02-07:
Thank you for the guidance! I totally agree with you and was looking to write something this way.

rattinca, 2021-02-11:
Is this the heat meter?

don-vip, 2021-02-13:
@rattinca I'm not sure to understand your question. I just added support for SHARKY 775 heat meter in #247

rattinca, 2021-02-13:
It sais hydrometer, so i was wondering if it is the same as the heat meter...
774 should be pretty similar?

weetmuts, 2021-02-13:
@rattinca Well, we will not know for sure if it is similar until you collect some telegrams. :-)

From their web-page it says that sharky 774 is oms compliant, so it might not even need
the new excellent code that @don-vip has written. But it might very well send a very similar
data structure as 775, so perhaps the sharky driver can be re-used.

weetmuts, 2021-02-13:
@rattinca The meter companies have been bought and changed names over time.
So the same type of meter, that is mechanically identical, can have different 
brand/manufacturer names and indeed different manufacturer codes in the telegram. 

In this case, meters from Diehl can have these manufacturer codes.
DME = Diehl
HYD = Hydrometer
SAP = Sappel 
and some more.

It sais hydrometer,

don-vip, 2021-02-13:
Sharky 775 is said to be OMS compliant too :) It's just compliant once you understand the few parts where it isn't :)
@rattinca if you can capture and share some telegrams it should be easy to add support for 774 too.

weetmuts, 2021-02-13:
> t's just compliant once you understand the few parts where it isn't 

Haha!  :grin: True, true!

weetmuts, 2021-03-06:
@rattinca Please open a new issue for 774 when you have collected some telegrams!
Closing this for now.

closed 2021-03-06 by weetmuts