wmbusmeters/wmbusmeters #242: rtl_wmbus detection not working anymore

don-vip, 2021-02-08:
I updated rtl_wmbus to latest git version and the binary detection doesn't work anymore:

```
$ ./build/wmbusmeters --ignoreduplicates --logtelegrams rtlwmbus:t1 foo izar '*' NOKEY
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
[ALARM SpecifiedDeviceNotFound] the device rtlwmbus:t1 is not working
No wmbus device detected, waiting for a device to be plugged in.
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
```

If I disable the check in util.cc, everything works fine.

I see wmbusmeters expects a return code of 2, but the command returns 0 on my system:

```
pi@raspberrypi:~/wmbusmeters2 $ /bin/sh -c rtl_wmbus < /dev/null
pi@raspberrypi:~/wmbusmeters2 $ echo $?
0
```

theWagNerd, 2021-02-08:
Watching..... I had to reinstall yesterday (I managed to snap a microsd card extracting a pi zero from its case) and I get the same error messages but rtl_sdr works perfectly fine with rtl_wmbus, e.g. rtl_sdr` -f 868.9M -s 1.6e6 - | rtl_wmbus outputs the telegrams as you'd expect

weetmuts, 2021-02-09:
Yes! Thanks for reporting. Simple fix pushed. https://github.com/weetmuts/wmbusmeters/commit/aa3bded7997f3d576017f693ed81ce1dc5f58570

I have tested and it works now. Closing this fo now. Please re-open if there is a problem.

closed 2021-02-09 by weetmuts