wmbusmeters/wmbusmeters #244: Techem Radio4 watter meter - Radio converter (meter side)

Unlink, 2021-02-10:
Hi, the wmbusmeters wokrs well with Techem Heat Cost Allocator and some water meters using mkradio3 or mkradio4. Problem is that there are many water meters and I can't find which one is mine. 
I ran scan and found my number (printed on the meter) on a weird telegram, which cannot be decoded
The real number is in "Concerning meter" filed.

```
"3644685042810592953772616121632423FE0762002025D8E5F272CF34CD01C0CEFDDFEFACC677F09E06341F5D24C685A92263B6191FEC"
(wmbus) parseDLL @0 55
(telegram) DLL L=36 C=44 (from meter SND_NR) M=5068 (TCH) A=92058142 VER=95 TYPE=37 (Radio converter (meter side)) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 45
(wmbus) parseAFL @10 45
(wmbus) parseTPL @10 45
(telegram) TPL CI=72 ACC=62 STS=00 CFG=2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0) ID=61612163 MFT=2423 VER=fe TYPE=07 (Water meter)
Received telegram from: 92058142
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
      Concerning meter: 63216161
          manufacturer: (HYD) Hydrometer (0x2324)
                  type: Water meter (0x07)
                   ver: 0xfe
                driver: unknown!
(wmbus) 00: 36 length (54 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 6850 dll-mfct (TCH)
(wmbus) 04: 42810592 dll-id (92058142)
(wmbus) 08: 95 dll-version
(wmbus) 09: 37 dll-type (Radio converter (meter side))
(wmbus) 0a: 72 tpl-ci-field (EN 13757-3 Application Layer (long tplh))
(wmbus) 0b: 61612163 tpl-id (63216161)
(wmbus) 0f: 2423 tpl-mfct (HYD)
(wmbus) 11: fe tpl-version
(wmbus) 12: 07 tpl-type (Water meter)
(wmbus) 13: 62 tpl-acc-field
(wmbus) 14: 00 tpl-sts-field
(wmbus) 15: 2025 tpl-cfg 2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
telegram=||3644685042810592953772616121632423FE0762002025D8E5F272CF34CD01C0CEFDDFEFACC677F09E06341F5D24C685A92263B6191FEC|+0

```
Another (using mkradio4) works fine.

```
"2F446850515487739562A2069F292B01A0241F000000090B090E0E0A0B0D0D0B040905070A0908000001020B0A0B0B0A"
(wmbus) parseDLL @0 48
(telegram) DLL L=2f C=44 (from meter SND_NR) M=5068 (TCH) A=73875451 VER=95 TYPE=62 (Warm water) (driver mkradio4) DEV= RSSI=0
(wmbus) parseELL @10 38
(wmbus) parseAFL @10 38
(wmbus) parseTPL @10 38
(telegram) TPL CI=a2
Received telegram from: 73875451
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Warm water (0x62)
                   ver: 0x95
                driver: mkradio4
(wmbus) 00: 2f length (47 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 6850 dll-mfct (TCH)
(wmbus) 04: 51548773 dll-id (73875451)
(wmbus) 08: 95 dll-version
(wmbus) 09: 62 dll-type (Warm water)
(wmbus) 0a: a2 tpl-ci-field (Mfct specific)
telegram=||2F446850515487739562A2069F292B01A0241F000000090B090E0E0A0B0D0D0B040905070A0908000001020B0A0B0B0A|+0
```

The debug prints that it is encrypted. I don't have a key so the question is, it is really encrypted, or it is not decoded properly? There are bunch of unencrypted water meters in my area, so it is bad luck that my is encrypted?

Thanks

Unlink, 2021-02-10:
It is weird, now I am getting correct values 
```(meter) Water mkradio4 handling telegram from 92058142
(meter) Water 92058142 "2F446850428105929572A2069F29A603A02423000000081007090A0C080317220B100A060F0C01000001112B282F261B"
(wmbus) parseDLL @0 48
(telegram) DLL L=2f C=44 (from meter SND_NR) M=5068 (TCH) A=92058142 VER=95 TYPE=72 (Cold water) (driver mkradio4) DEV= RSSI=0
(wmbus) parseELL @10 38
(wmbus) parseAFL @10 38
(wmbus) parseTPL @10 38
(telegram) TPL CI=a2
telegram=|2F446850428105929572A2|069F29A603A02423000000081007090A0C080317220B100A060F0C01000001112B282F261B|+0
(mkradio4) 00: 2f length (47 bytes)
(mkradio4) 01: 44 dll-c (from meter SND_NR)
(mkradio4) 02: 6850 dll-mfct (TCH)
(mkradio4) 04: 42810592 dll-id (92058142)
(mkradio4) 08: 95 dll-version
(mkradio4) 09: 72 dll-type (Cold water)
(mkradio4) 0a: a2 tpl-ci-field (Mfct specific)
(mkradio4) 0e: * a603 prev consumption (93.400000 m3)
(mkradio4) 12: * 2300 curr consumption (3.500000 m3)
Water   92058142        96.9 m3 93.4 m3 2021-02-10 15:09.32
```

It is possible that it sometimes use encrypted telegrams and sometime not?

weetmuts, 2021-02-21:
Can you test with the latest version of wmbusmeters from github? There is now support for properly
decoding the water meter (thanks @don-vip ) and if you check the json output there is a field "serial_number":"123456"
that should print the number printed on the meter.

The mkradio4 meters are probably sending two different telegrams. One OMS-compliant and encrypted and
one using a proprietary format which has been very rudimentary decoded.

Unlink, 2021-02-28:
Thanks, I can see the serial number in json.
You are right, they are probably sending two different packets. The first one, encrypted could be used to determine meter id and then I can read meter data observing the second one.

closed 2021-02-28 by Unlink