wmbusmeters/wmbusmeters #245: Apator Elf1,5 heat meter and Apator e-rm30 water meter radio module

alalons, 2021-02-11:
I have been capturing those telegrams at a site with Apator Elf1,5 heat meter and Apator e-rm30 radio module for water meter.
The e-rm30 is produced both for  js water meters and gsd8-rf water meters suspiciously named similar to bmeters gsd8-rfm.
I got there one telegram with Bmeters manufacturer code hydrodigit driver suggested which I think could be the e-rm30 water meter, and multiple Apator telegrams which I am convinced  are the Apator ELF heat meters.
Apator telegrams seem to be standard T1 with long header.

Could you implement Elf meter?
Has any one have ever captured E-ITN30 heat cost allocators?

Regards, 
Alvaro

[E-RM30_manual.pdf](https://github.com/weetmuts/wmbusmeters/files/5962881/E-RM30_manual.pdf)
[i-en-009-2017-elf-13-01.pdf](https://github.com/weetmuts/wmbusmeters/files/5962904/i-en-009-2017-elf-13-01.pdf)
[telegramasApator.txt](https://github.com/weetmuts/wmbusmeters/files/5962909/telegramasApator.txt)

weetmuts, 2021-02-14:
Thanks for the telegrams. I pushed support for the new heat meter elf, which decodes your heat meter into this json:

{"media":"heat","meter":"elf","name":"Hetta","id":"01885619","meter_date":"2021-02-09","total_energy_consumption_kwh":3112.49977,"current_power_consumption_kw":0,"total_volume_m3":201.364,"total_energy_consumption_at_date_kwh":3047.8,"flow_temperature_c":69,"return_temperature_c":58,"external_temperature_c":37.64,"status":"200000","operating_time_h":44760,"version":"1","battery_v":0,"timestamp":"2021-02-14T21:57:45Z"}

The voltage is not yet properly decoded, I need to fix the dvparser. I am assuming the voltage is for the battery.

Please test!

weetmuts, 2021-02-14:
Try another driver (like supercom587) on your water meter and post the debug output here. 

I have never heard about the E-ITN30 heat cost allocator before. Please post a telegram here.

weetmuts, 2021-02-21:
Closing this for now. Open a new issue for each of the new meters that you want to add!

closed 2021-02-21 by weetmuts