wmbusmeters/wmbusmeters #246: Adding smoke detector

rattinca, 2021-02-11:
what i did so far:
wmbusmeters --logtelegrams <tty>:c1

put telegrams in simulation_smoke.txt

wmbusmeters simulation_smoke.txt 
works fine.

wmbusmeters --debug simulation_smoke.txt smoke tsd2 id key
gives me a lot of this:
`(simulation) from file "5E462515112801000C1A7A37005025771FDE5731527CBEE4C97311D5A36EF20DFFF4D8997A2A0B5183B22CEE3A71B6D0CDFBEF3E09E13AF8B085E1FC43EA704A5CA5414CDD46C9869957B537CF911906053C914CA17E1545B5D307394ACA81"
(wmbus) parseDLL @0 95
(wmbus) parseELL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(meter) smoke: for me? 00012811
(meter) smoke: meter detection did not match the selected driver tsd2! correct driver is: unknown!
(meter) Not printing this warning agin for id: 00012811 mfct: (EIE) Ei Electronics, Ireland (0x1525) type: Smoke detector (0x1a) ver: 0x0c
(meter) please consider opening an issue at https://github.com/weetmuts/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) smoke: yes for me
(meter) smoke tsd2 handling telegram from 00012811
(meter) smoke 00012811 "5E462515112801000C1A7A37005025771FDE5731527CBEE4C97311D5A36EF20DFFF4D8997A2A0B5183B22CEE3A71B6D0CDFBEF3E09E13AF8B085E1FC43EA704A5CA5414CDD46C9869957B537CF911906053C914CA17E1545B5D307394ACA81"
(wmbus) parseDLL @0 95
(telegram) DLL L=5e C=46 (from meter SND_IR) M=1525 (EIE) A=00012811 VER=0c TYPE=1a (Smoke detector) (driver unknown!) DEV= RSSI=0
(wmbus) parseELL @10 85
(wmbus) parseAFL @10 85
(wmbus) parseTPL @10 85
(TPL) decrypting "771FDE5731527CBEE4C97311D5A36EF20DFFF4D8997A2A0B5183B22CEE3A71B6D0CDFBEF3E09E13AF8B085E1FC43EA704A5CA5414CDD46C9869957B537CF911906053C914CA17E1545B5D307394ACA81"
(TPL) num encrypted blocks 5 (80 bytes and remaining unencrypted 0 bytes)
(TPL) IV 2515112801000C1A3737373737373737
(TPL) decrypted  "2F2F0BFD0F060101046D300CAB2202FD17000082206CAB22426C01018440FF2C000F11008250FD61000082506C01018260FD6100008360FD3100000082606C01018270FD61010082706CAB222F2F2F2F"
(telegram) TPL CI=7a ACC=37 STS=00 CFG=2550 (synchronous AES_CBC_IV nb=5 cntn=0 ra=0 hc=0)
telegram=|5E462515112801000C1A7A370050252F2F|0BFD0F060101046D300CAB2202FD17000082206CAB22426C01018440FF2C000F11008250FD61000082506C01018260FD6100008360FD3100000082606C01018270FD61010082706CAB222F2F2F2F|+31
(tsd2) 00: 5e length (94 bytes)
(tsd2) 01: 46 dll-c (from meter SND_IR)
(tsd2) 02: 2515 dll-mfct (EIE)
(tsd2) 04: 11280100 dll-id (00012811)
(tsd2) 08: 0c dll-version
(tsd2) 09: 1a dll-type (Smoke detector)
(tsd2) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(tsd2) 0b: 37 tpl-acc-field
(tsd2) 0c: 00 tpl-sts-field
(tsd2) 0d: 5025 tpl-cfg 2550 (synchronous AES_CBC_IV nb=5 cntn=0 ra=0 hc=0 )
(tsd2) 0f: 2f2f decrypt check bytes
(tsd2) 11: 0B dif (6 digit BCD Instantaneous value)
(tsd2) 12: FD vif (Second extension of VIF-codes)
(tsd2) 13: 0F vife (Software version #)
(tsd2) 14: 060101
(tsd2) 17: 04 dif (32 Bit Integer/Binary Instantaneous value)
(tsd2) 18: 6D vif (Date and time type)
(tsd2) 19: 300CAB22
(tsd2) 1d: 02 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 1e: FD vif (Second extension of VIF-codes)
(tsd2) 1f: 17 vife (Error flags (binary))
(tsd2) 20: 0000
(tsd2) 22: 82 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 23: 20 dife (subunit=0 tariff=2 storagenr=0)
(tsd2) 24: 6C vif (Date type G)
(tsd2) 25: AB22
(tsd2) 27: 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
(tsd2) 28: 6C vif (Date type G)
(tsd2) 29: 0101
(tsd2) 2b: 84 dif (32 Bit Integer/Binary Instantaneous value)
(tsd2) 2c: 40 dife (subunit=1 tariff=0 storagenr=0)
(tsd2) 2d: FF vif (Vendor extension)
(tsd2) 2e: 2C vife (per litre)
(tsd2) 2f: 000F1100
(tsd2) 33: 82 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 34: 50 dife (subunit=1 tariff=1 storagenr=0)
(tsd2) 35: FD vif (Second extension of VIF-codes)
(tsd2) 36: 61 vife (Cumulation counter)
(tsd2) 37: 0000
(tsd2) 39: 82 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 3a: 50 dife (subunit=1 tariff=1 storagenr=0)
(tsd2) 3b: 6C vif (Date type G)
(tsd2) 3c: 0101
(tsd2) 3e: 82 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 3f: 60 dife (subunit=1 tariff=2 storagenr=0)
(tsd2) 40: FD vif (Second extension of VIF-codes)
(tsd2) 41: 61 vife (Cumulation counter)
(tsd2) 42: 0000
(tsd2) 44: 83 dif (24 Bit Integer/Binary Instantaneous value)
(tsd2) 45: 60 dife (subunit=1 tariff=2 storagenr=0)
(tsd2) 46: FD vif (Second extension of VIF-codes)
(tsd2) 47: 31 vife (Duration of tariff [minute(s)])
(tsd2) 48: 000000
(tsd2) 4b: 82 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 4c: 60 dife (subunit=1 tariff=2 storagenr=0)
(tsd2) 4d: 6C vif (Date type G)
(tsd2) 4e: 0101
(tsd2) 50: 82 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 51: 70 dife (subunit=1 tariff=3 storagenr=0)
(tsd2) 52: FD vif (Second extension of VIF-codes)
(tsd2) 53: 61 vife (Cumulation counter)
(tsd2) 54: 0100
(tsd2) 56: 82 dif (16 Bit Integer/Binary Instantaneous value)
(tsd2) 57: 70 dife (subunit=1 tariff=3 storagenr=0)
(tsd2) 58: 6C vif (Date type G)
(tsd2) 59: AB22
(tsd2) 5b: 2F skip
(tsd2) 5c: 2F skip
(tsd2) 5d: 2F skip
(tsd2) 5e: 2F skip
smoke.00012811.SMOKE.2021-02-11 13:38.55

There seems to be a lot of information already in there. Whats my next step? I can send you logs if you want.

weetmuts, 2021-02-12:
A lot of stuff in there indeed! Have you read https://github.com/weetmuts/wmbusmeters/blob/master/HowToAddaNewMeter.txt

Kosygor, 2021-02-14:
May I ask what smoke detector do You have? Please provide manufacturer and model.
tsd2 is for Techem smoke detector 2 only .
Have You tried Smoke detector from documentation > Lansen Smoke Detector (lansensm).

rattinca, 2021-02-16:
Yes, i tried lansensm too with the same result.
I have now added a working copy of tsd2 for ie6500. Would be good for newbies to add to the HowTo that the manufacturers are listed in manufacturers.h. Took me a while to find the 3 letter code x-)
Now trying to adapt for my smoke detector.

rattinca, 2021-02-17:
I extracted a few dates. Whats the correct function to extract an int? Its a counter, no quantity like volume but i cant use text with extractDVuint16...

rattinca, 2021-02-17:
also: how to i extract a time thats not a date? just counted time?

rattinca, 2021-02-17:
and, whats the recomended way to extract just the binary to implement error codes?

rattinca, 2021-02-19:
ok, i got at the binaries with
    vector<uchar> content;
    t->extractPayload(&content);
but it starts after the Encription Verification. How do i get at the stuff before? I need the tpl-sts-field

weetmuts, 2021-02-19:
The Telegram class has all the already decoded data, so just use t->tpl_sts

rattinca, 2021-02-19:
thx!

ok, i added what i could understand so far. im new at this but i pushed to a fork (i think). if you want to take a look. 
i will test and see if i can decode more.
there does not seem to be an indicator for smoke other than a date...

weetmuts, 2021-02-21:
Can you make a pull request? It makes it easier for me to see the code.
You can write that it is a DRAFT, if you like.

rattinca, 2021-02-23:
ok, i think i did that now

ElKaPa, 2021-03-31:
Any progress on this topic?
Frame byte 84 40 FF 2C contains the smoke detector head status.
bit 8 to bit 11 are for the battery voltage.
If bit 8 to 11 are 1 the value is 15.
15* 0.05V + 2.25V would be the battery voltage level.

I haven't found a way yet how to get bit 8 to 11 to calculate the value and show when decrypting the data.

ElKaPa, 2021-04-01:
@Kosygor 
they are using 4 bits in total so 0 to 15.
This 4bit value is multiplied with 0.05V and 2.25V is the bottom line value at which battery is declared as fully empty.

I never worked with Techem yet but as far as I remember they are not giving out the AES key for decrypting the data package

Kosygor, 2021-04-01:
@ElKaPa Sorry to bother You and for OT but I think that I am missing something.

> I never worked with Techem yet but as far as I remember they are not giving out the AES key for decrypting the data package.

This is probably true. I am only trying to interpret what wmbusmeters is currently reading. 

Ok I get fact that in 4 bits You mentioned You can pass Value up to "15" decimal value. 
But You also have mentioned:  
>Frame byte 84 40 FF 2C

And if I am which is written in hex (dec/bin)
- 84 = 132 = 10000100
- 40 = 64 = 1000000
- FF = 255 = 11111111
- 2C = 44 = 101100

Then You have mentioned that
>bit 8 to bit 11 are for the battery voltage.

Which in that exact case is 0100  - last bit of first byte and first 3 from 2nd byte. Am I right?
And thing that I should keep in mind decrypting my frame is that: 
I should not only pay attention to frame in hex but also analyze it on binary info level (that can be trans byte/hex)?

ElKaPa, 2021-04-01:
@Kosygor 

The frame bytes are referring to the decrypted package.

(TPL) decrypted "2F2F0BFD0F060101046D300CAB2202FD17000082206CAB22426C01018440FF2C000F11008250FD61000082506C01018260FD6100008360FD3100000082606C01018270FD61010082706CAB222F2F2F2F"

8440FF2C are the frame bytes. 000F1100 is the data.
00110F00 = 100010000111100000000

bit 8-11 = 1111 = 15

15*0.05V + 2.25V = 3V
So battery is almost new.

You need to work on decrypted data.
Ei6500 is OMS compliant but the OMS group has no native support for smoke detectors yet.
That's why the data is vendor specific

Kosygor, 2021-04-07:
@ElKaPa 
Thank You for explanation. 
According to post written by [Fredrik`s post](https://github.com/weetmuts/wmbusmeters/issues/200#issuecomment-755737903) header is already decoded.
How do You analyse data ? Is it Excel or something else ?

weetmuts, 2022-05-21:
Hi @rattinca !
I updated the ei6500 driver to use correct units for duration and counter. I.e. no longer strings but instead
"duration_remove_h":1.3 (for 1.3 hours of removed)
and
"smoke_alarm_counter":0

Can you check the analysis:
https://wmbusmeters.org/analyze/5E462515112801000C1A7A370050252F2F0BFD0F060101046D300CAB2202FD17000082206CAB22426C01018440FF2C000F11008250FD61000082506C01018260FD6100008360FD3100000082606C01018270FD61010082706CAB222F2F2F2F

And see if it looks ok. Also there are two dates that we do not know what they do, and a vendor extension value.
Do you have any guess to what they mean?

The new driver format is much easier to edit. Check out https://github.com/weetmuts/wmbusmeters/blob/master/src/driver_ei6500.cc

weetmuts, 2022-05-21:
You can read https://github.com/weetmuts/wmbusmeters/issues/367
for how the  tpl status byte is included into the "status" field. (PrintProperty::JOIN_TPL_STATUS)

ElKaPa, 2022-05-21:
@weetmuts 

I mentioned all details here:

https://github.com/weetmuts/wmbusmeters/pull/256#issuecomment-849980383

weetmuts, 2022-05-22:
Thanks @ElKaPa ! Yikes, that was a lot of bits. :-)

I think I have added them all, but two questions from your table.
The dust is 0-15 but you specify 5 bits 0-4.... is bit 0x10 not used/reserved/ignored? Or is the dust level 0-31?
Is the bit 23 not used/reserved/ignored?

What does OBSTACLE_DETECTING_INSTALLATION_OK 0x0010000 mean? 

I have updated wmbusmeters.org so you can test the link above again.
Do you have some telegrams from installed meters?

ElKaPa, 2022-05-22:
@weetmuts 
You are right, the dust level is from 0-31.
Bit 23 is declared for future use.

`What does OBSTACLE_DETECTING_INSTALLATION_OK 0x0010000 mean?`

Every time the smoke detector is mounted, it will do a new installation of the obstacle detection system.
Once successful done it will set this flag and update the bits for distance to obstacle

weetmuts, 2022-09-21:
I think we can close this one now!

closed 2022-09-21 by weetmuts