wmbusmeters/wmbusmeters #252: wmbusmeters for Qalcosonic W1 water meter?

joggele777, 2021-02-15:
Hello,

i want to use wmbusmeters for a QALCOSONIC W1 - SMART ULTRASONIC WATER METER - should it work with this one?

Thanks and regards
Jochen

weetmuts, 2021-02-16:
If you can collect a telegram and have a key (assuming its encrypted) then we can take a look!

joggele777, 2021-02-16:
Thanks for your answer. Sounds good.

Unfortunately, I do not receive any telegrams so far. But I am also not sure if I am doing the right program start?

I am on Pi OS:

pi@raspberrypi:/dev$ sudo snap run wmbusmeters auto:c1 --verbose --listmeters
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.

M-Bus Adapter: iM871A

looks good on Pi OS:

pi@raspberrypi:/dev$ usb-devices

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=12  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=10c4 ProdID=ea60 Rev=01.00
S:  Manufacturer=Silicon Labs
S:  Product=WiMOD iM871A-usb
S:  SerialNumber=020F9B22
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=100mA
I:  If#=0x0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=00 Prot=00 Driver=cp210x

Do you have a hint or tip for me?

BIBOLV, 2021-02-16:
I haven`t tested snap usage in that way, so I would suggest to use config file, without any additional parameters. Simple manual for snap usage can be found [here](https://github.com/weetmuts/wmbusmeters/blob/master/snap/README.md) . So add debug output in config and see if it works - telegrams should be in a log, or compile and run wmbusmeters directly on system.

joggele777, 2021-02-17:
cool - received some telegrams - excerpt from log:

No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 05158278
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07)
                   ver: 0x10
                device: im871a[00102463]
                  rssi: -56 dBm
                driver: q400
Received telegram from: 05158278
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07)
                   ver: 0x10
                device: im871a[00102463]
                  rssi: -56 dBm
                driver: q400
Received telegram from: 05158278
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07)
                   ver: 0x10
                device: im871a[00102463]
                  rssi: -57 dBm
                driver: q400

Received telegram from: 05158278 - Number matches with S/N printed on my water meter

What to do next?

BIBOLV, 2021-02-17:
You need to change in config file loglevel to debug and/or logtelegrams to true. Then you should see in logs actual telegrams sent by meter and you can post them here.

joggele777, 2021-02-17:
config and log

config:

loglevel=debug
device=auto:t1
# To use rtl_wmbus uncomment following line 
#device=rtlwmbus:LD_LIBRARY_PATH=/var/lib/snapd/lib/gl:/var/lib/snapd/lib/gl32:/var/lib/snapd/void:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu::/snap/wmbusmeters/current/lib:/snap/wmbusmeters/current/usr/lib:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu /snap/wmbusmeters/current/usr/bin/rtl_sdr -f 868.95M -s 1600000 - 2>/dev/null | /snap/wmbusmeters/current/usr/bin/rtl_wmbus
# To use rtl_433 uncomment following line 
#device=rtl_433:LD_LIBRARY_PATH=/var/lib/snapd/lib/gl:/var/lib/snapd/lib/gl32:/var/lib/snapd/void:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu::/snap/wmbusmeters/current/lib:/snap/wmbusmeters/current/usr/lib:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu /snap/wmbusmeters/current/usr/bin/rtl_433 -F csv -f 868.95M
listento=t1
logtelegrams=true
format=json
meterfiles=/var/snap/wmbusmeters/common/logs/meter_readings
meterfilesaction=overwrite
logfile=/var/snap/wmbusmeters/common/logs/wmbusmeters.log

log:

(wmbusmeters) version: 1.0.5
(config) store meter files in: "/var/snap/wmbusmeters/common/logs/meter_readings"
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32563241000010001FF070232000100000000E311"
(im871a) checkIM871AFrame "A5810617FF000300B32563241000010001FF070232000100000000E311"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 11e3 expected 11e3
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(device config bytes) "FF000300B32563241000010001FF070232000100000000"
(im871a) config: other link_mode=t1 id=00102463 media=00 version=01 c_field=00 auto_rssi=01
(im871a) are you there? yes 00102463
(main) opening t1
Started auto im871a[00102463] on /dev/ttyUSB0 listening on t1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32563241000010001FF0702320001000000D3F5F5"
(im871a) checkIM871AFrame "A5810617FF000300B32563241000010001FF0702320001000000D3F5F5"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 f5f5 expected f5f5
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00102463
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) set config to set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop

BIBOLV, 2021-02-17:
Those looks like encrypted, do you have decryption key?

joggele777, 2021-02-18:
I will try to get the AES key through my water supplier

BIBOLV, 2021-02-18:
Sorry, I reviewed your last log once more and I do not think that there are actual telegrams from meter. Could you run longer and see if there will colected anything more. Also check if mode is correct - initially you commented c1 usage but in config t1 is used.

joggele777, 2021-02-19:
config running with c1 - new log:

(wmbusmeters) version: 1.0.5
(config) store meter files in: "/var/snap/wmbusmeters/common/logs/meter_readings"
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32563241000010001FF07023200010000009D8F5E"
(im871a) checkIM871AFrame "A5810617FF000300B32563241000010001FF07023200010000009D8F5E"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 5e8f expected 5e8f
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(device config bytes) "FF000300B32563241000010001FF07023200010000009D"
(im871a) config: other link_mode=t1 id=00102463 media=00 version=01 c_field=00 auto_rssi=01
(im871a) are you there? yes 00102463
(main) opening c1
Started auto im871a[00102463] on /dev/ttyUSB0 listening on c1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32563241000010001FF070232000100000000E311"
(im871a) checkIM871AFrame "A5810617FF000300B32563241000010001FF070232000100000000E311"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 11e3 expected 11e3
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00102463
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) set config to set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
[... 65 lines cut ...]
(serial) received binary "A5810617FF000600B32563241000010001FF07023200010000009D53B4"
(im871a) checkIM871AFrame "A5810617FF000600B32563241000010001FF07023200010000009D53B4"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 b453 expected b453
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00102463
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) set config to set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop

BIBOLV, 2021-02-19:
Still not it. You need to wait for line starting with "telegram". We are interested only in those lines. Something like this:
`# cat /var/snap/wmbusmeters/common/logs/wmbusmeters.log | grep telegram=
telegram=||2E4433300230020014007ADA002025ACBB061BD56F1AB1E0E9B00377832B347E4B31B7720C53C8C4E1C460176336CA|+51`

joggele777, 2021-02-20:
What could be the reason that I do not receive telegrams?

joggele777, 2021-02-20:
Does this need to be turned on somehow? I am also trying to clarify this with my water supplier.

BIBOLV, 2021-02-20:
You received them based on first log you provided, so maybe try to revert config and see if that helps? Confirm that correct mode is used - t1 or c1.

joggele777, 2021-02-20:
strange, I don't get any LOG entries in the form with telegram=........ neither with c1 nor t1

BIBOLV, 2021-02-20:
Well try to revert config to initial one that you used then you received
`Received telegram from: 05158278....`
Then set log to debug and wait for same line, then post log here.

weetmuts, 2021-02-21:
It seems like the water meter is supported using the q400 driver. Nice!
Also the q400 trasmitted using t1 mode, so make sure you run with 
auto:t1 for example.

weetmuts, 2021-02-21:
Also, some meters do turn off radio transmissions during the weekend and nights, in the default configuration.

joggele777, 2021-02-21:
Interesting info! This also explains why I receive yesterday and today no telegrams (weekend). Configuration is set to "t1". Now I wait times whether tomorrow (weekdays) telegrams are sent again.

joggele777, 2021-02-22:
now log looks good:

Received telegram from: 05158278
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07)
                   ver: 0x10
                device: im871a[00102463]
                  rssi: -52 dBm
                driver: q400
(wmbus) 00: 2e length (46 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 0907 dll-mfct (AXI)
(wmbus) 04: 78821505 dll-id (05158278)
(wmbus) 08: 10 dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: a0 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
telegram=||2E4409077882150510077AA00020054532DA0D4470BB42703AEBE43A4B22AF5642FD5BDAAE81CE2AF05E1D78A96DD1|+5908
(serial) received binary "A5C2032E4409077882150510077AA1002005DBBEA2F8C4D56CDFB4738B0550AF6712705C1FFF754346AAE79A6C8E81B3FED68D800C"
(im871a) checkIM871AFrame "A5C2032E4409077882150510077AA1002005DBBEA2F8C4D56CDFB4738B0550AF6712705C1FFF754346AAE79A6C8E81B3FED68D800C"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 141 (-52 dBm)
(im871a) got crc16 0c80 expected 0c80
(im871a) received full frame
(wmbus) parseDLL @0 47
(telegram) DLL L=2e C=44 (from meter SND_NR) M=0709 (AXI) A=05158278 VER=10 TYPE=07 (Water meter) (driver q400) DEV=im871a[00102463] RSSI=-52
(wmbus) parseELL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(telegram) TPL CI=7a ACC=a1 STS=00 CFG=0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
Received telegram from: 05158278
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07)
                   ver: 0x10
                device: im871a[00102463]
                  rssi: -52 dBm
                driver: q400
(wmbus) 00: 2e length (46 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 0907 dll-mfct (AXI)
(wmbus) 04: 78821505 dll-id (05158278)
(wmbus) 08: 10 dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: a1 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
telegram=||2E4409077882150510077AA1002005DBBEA2F8C4D56CDFB4738B0550AF6712705C1FFF754346AAE79A6C8E81B3FED6|+5925
(serial) received binary "A5C2032E4409077882150510077AA2002005840F0372996F054397F213C24D53C68EC27FEDCE3330C386435C56482432ABA08DD6AB"
(im871a) checkIM871AFrame "A5C2032E4409077882150510077AA2002005840F0372996F054397F213C24D53C68EC27FEDCE3330C386435C56482432ABA08DD6AB"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 141 (-52 dBm)
(im871a) got crc16 abd6 expected abd6
(im871a) received full frame
(wmbus) parseDLL @0 47
(telegram) DLL L=2e C=44 (from meter SND_NR) M=0709 (AXI) A=05158278 VER=10 TYPE=07 (Water meter) (driver q400) DEV=im871a[00102463] RSSI=-52
(wmbus) parseELL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(telegram) TPL CI=7a ACC=a2 STS=00 CFG=0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
Received telegram from: 05158278
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07)
                   ver: 0x10
                device: im871a[00102463]
                  rssi: -52 dBm
                driver: q400
(wmbus) 00: 2e length (46 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 0907 dll-mfct (AXI)
(wmbus) 04: 78821505 dll-id (05158278)
(wmbus) 08: 10 dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: a2 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
telegram=||2E4409077882150510077AA2002005840F0372996F054397F213C24D53C68EC27FEDCE3330C386435C56482432ABA0|+5941
(serial) received binary "A5C2032E4409077882150510077AA3002005D62EE65F7E34C284CC7717FA7C54CF87FF1A3306C81F64F7888F8761F46474D08DD53C"
(im871a) checkIM871AFrame "A5C2032E4409077882150510077AA3002005D62EE65F7E34C284CC7717FA7C54CF87FF1A3306C81F64F7888F8761F46474D08DD53C"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 141 (-52 dBm)
(im871a) got crc16 3cd5 expected 3cd5
(im871a) received full frame
(wmbus) parseDLL @0 47
(telegram) DLL L=2e C=44 (from meter SND_NR) M=0709 (AXI) A=05158278 VER=10 TYPE=07 (Water meter) (driver q400) DEV=im871a[00102463] RSSI=-52
(wmbus) parseELL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(telegram) TPL CI=7a ACC=a3 STS=00 CFG=0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
Received telegram from: 05158278
          manufacturer: (AXI) UAB Axis Industries, Lithuania (0x709)
                  type: Water meter (0x07)
                   ver: 0x10
                device: im871a[00102463]
                  rssi: -52 dBm
                driver: q400
(wmbus) 00: 2e length (46 bytes)
(wmbus) 01: 44 dll-c (from meter SND_NR)
(wmbus) 02: 0907 dll-mfct (AXI)
(wmbus) 04: 78821505 dll-id (05158278)
(wmbus) 08: 10 dll-version
(wmbus) 09: 07 dll-type (Water meter)
(wmbus) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 0b: a3 tpl-acc-field
(wmbus) 0c: 00 tpl-sts-field
(wmbus) 0d: 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
telegram=||2E4409077882150510077AA3002005D62EE65F7E34C284CC7717FA7C54CF87FF1A3306C81F64F7888F8761F46474D0|+5958

BIBOLV, 2021-02-22:
Cool, now you need decryption key to decrypt those telegrams.

joggele777, 2021-02-23:
Talked to my water supplier on the phone. I should get the key in the next few days :-) Next step would be the implementation as follows:

Then add a meter file in /etc/wmbusmeters.d/MyTapWater

name=MyTapWater
type=multical21:c1
id=12345678
key=00112233445566778899AABBCCDDEEFF

Right?

After that I want a forwarding via MQTT to iobroker.

joggele777, 2021-02-23:
type should be q400:t1

weetmuts, 2021-02-23:
Yep. And you can read about iobroker+wmbusmeter here: https://weetmuts.github.io/wmbusmeterswiki/IOBROKER.html

weetmuts, 2021-02-23:
If you learn more about how to integrate with iobroker, please tell us, so that we can update the wiki. :-)

joggele777, 2021-02-23:
got the key :-)

now configured MyTapWater file (in my case PI with snap) under /var/snap/......

pi@raspberrypi:/var/snap/wmbusmeters/common/etc/wmbusmeters.d$ cat MyTapWater 
name=MyTapWater
type=q400:t1
id=05158278
key=1F3190D8F405CD0641D24B7BFFEE352E

I think there should be a meter reading file now at:

pi@raspberrypi:/var/snap/wmbusmeters/common/logs/meter_readings$

but nothing?

joggele777, 2021-02-23:
daemon restarted

sudo systemctl restart snap.wmbusmeters.wmbusmeters.service

weetmuts, 2021-02-23:
What is in the wmbusmeters.log file?

BIBOLV, 2021-02-23:
Maybe transmission window is closed for today!?

joggele777, 2021-02-23:
It must be. Now I actually stumbled across it again :-( Will look further tomorrow.

joggele777, 2021-02-24:
:-)

(serial) received binary "A5C2032E4409077882150510077ACC002005DDF6E4D38F4FE756A539D821993385B2B5013C5F64492BF44B388F0875C83BA78D8D23"
(im871a) checkIM871AFrame "A5C2032E4409077882150510077ACC002005DDF6E4D38F4FE756A539D821993385B2B5013C5F64492BF44B388F0875C83BA78D8D23"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 141 (-52 dBm)
(im871a) got crc16 238d expected 238d
(im871a) received full frame
(wmbus) parseDLL @0 47
(wmbus) parseELL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(meter) MyTapWater: for me? 05158278
(meter) MyTapWater: yes for me
(meter) MyTapWater q400 handling telegram from 05158278
(meter) MyTapWater 05158278 "2E4409077882150510077ACC002005DDF6E4D38F4FE756A539D821993385B2B5013C5F64492BF44B388F0875C83BA7"
(wmbus) parseDLL @0 47
(telegram) DLL L=2e C=44 (from meter SND_NR) M=0709 (AXI) A=05158278 VER=10 TYPE=07 (Water meter) (driver q400) DEV=im871a[00102463] RSSI=-52
(wmbus) parseELL @10 37
(wmbus) parseAFL @10 37
(wmbus) parseTPL @10 37
(TPL) decrypting "DDF6E4D38F4FE756A539D821993385B2B5013C5F64492BF44B388F0875C83BA7"
(TPL) num encrypted blocks 2 (32 bytes and remaining unencrypted 0 bytes)
(TPL) IV 0907788215051007CCCCCCCCCCCCCCCC
(TPL) decrypted  "2F2F046D2007B8220413C5A30000446D0000A1224413CE8600002F2F2F2F2F2F"
(telegram) TPL CI=7a ACC=cc STS=00 CFG=0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
telegram=|2E4409077882150510077ACC0020052F2F|046D2007B8220413C5A30000446D0000A1224413CE8600002F2F2F2F2F2F|+40464
(q400) 00: 2e length (46 bytes)
(q400) 01: 44 dll-c (from meter SND_NR)
(q400) 02: 0907 dll-mfct (AXI)
(q400) 04: 78821505 dll-id (05158278)
(q400) 08: 10 dll-version
(q400) 09: 07 dll-type (Water meter)
(q400) 0a: 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(q400) 0b: cc tpl-acc-field
(q400) 0c: 00 tpl-sts-field
(q400) 0d: 2005 tpl-cfg 0520 (AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
(q400) 0f: 2f2f decrypt check bytes
(q400) 11: 04 dif (32 Bit Integer/Binary Instantaneous value)
(q400) 12: 6D vif (Date and time type)
(q400) 13: 2007B822
(q400) 17: 04 dif (32 Bit Integer/Binary Instantaneous value)
(q400) 18: 13 vif (Volume l)
(q400) 19: * C5A30000 total consumption (41.925000 m3)
(q400) 1d: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(q400) 1e: 6D vif (Date and time type)
(q400) 1f: * 0000A122 set date (2021-02-01)
(q400) 23: 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
(q400) 24: 13 vif (Volume l)
(q400) 25: * CE860000 consumption at set date (34.510000 m3)
(q400) 29: 2F skip
(q400) 2a: 2F skip
(q400) 2b: 2F skip
(q400) 2c: 2F skip
(q400) 2d: 2F skip
(q400) 2e: 2F skip

pi@raspberrypi:/var/snap/wmbusmeters/common/logs/meter_readings$ cat MyTapWater 

{"media":"water","meter":"q400","name":"MyTapWater","id":"05158278","total_m3":41.925,"set_date":"2021-02-01","consumption_at_set_date_m3":34.51,"timestamp":"2021-02-24T05:34:02Z","device":"im871a[00102463]","rssi_dbm":-52}

joggele777, 2021-02-24:
next step - now trying MQTT, but return code: 127 ?  Also no objects found under: mqtt.0/wmbusmeters/[meter_id]

shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "METER_TOTAL_M3"

(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "METER_TOTAL_M3""
(shell) env "METER_JSON={"media":"water","meter":"q400","name":"MyTapWater","id":"05158278","total_m3":42.07,"set_date":"2021-02-01","consumption_at_set_date_m3":34.51,"timestamp":"2021-02-24T09:36:52Z","device":"im871a[00102463]","rssi_dbm":-51}"
(shell) env "METER_TYPE=q400"
(shell) env "METER_NAME=MyTapWater"
(shell) env "METER_ID=05158278"
(shell) env "METER_TOTAL_M3=42.07"
(shell) env "METER_SET_DATE=2021-02-01"
(shell) env "METER_CONSUMPTION_AT_SET_DATE_M3=34.51"
(shell) env "METER_TIMESTAMP=2021-02-24T09:36:52Z"
(shell) waiting for child 5058 to complete.
(shell) /bin/sh: return code 127
(shell) /bin/sh exited with non-zero return code: 127

BIBOLV, 2021-02-24:
Firstly - congrats on getting actual reading from meter :wink: . Can you confirm that reading from telegram corresponds with reading on meter? 
@weetmuts - do we need more telegrams in order to mark this meter as supported?
Secondly - please take a look at last line of [manual](https://github.com/weetmuts/wmbusmeters/blob/master/snap/README.md) and adjust "shell" command line accordingly. Also do you run MQTT server on same Pi? If not then change "localhost" to actual IP of MQTT server.

joggele777, 2021-02-24:
Thank you for the great support!!!

Telegram from meter corresponds 1:1 with reading on meter.

joggele777, 2021-02-24:
About MQTT

tried this: shell=LD_LIBRARY_PATH=/var/lib/snapd/lib/gl:/var/lib/snapd/lib/gl32:/var/lib/snapd/void:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu::/snap/wmbusmeters/current/lib:/snap/wmbusmeters/current/usr/lib:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu /snap/wmbusmeters/current/usr/bin/mosquitto_pub -h localhost -t water -m '$METER_JSON'

but still same problem.

I think thats my mosquitto-client path: /usr/bin/mosquitto_pub

MQTT server/broker (iobroker mqtt adapter) / client (mosquitto) and wmbusmeters are all on same PI host.

joggele777, 2021-02-24:
(shell) arg "LD_LIBRARY_PATH=/var/lib/snapd/lib/gl:/var/lib/snapd/lib/gl32:/var/lib/snapd/void:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu::/snap/wmbusmeters/current/lib:/snap/wmbusmeters/current/usr/lib:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu /snap/wmbusmeters/current/usr/bin/mosquitto_pub -h localhost -t water -m '$METER_JSON'"
(shell) env "METER_JSON={"media":"water","meter":"q400","name":"MyTapWater","id":"05158278","total_m3":42.116,"set_date":"2021-02-01","consumption_at_set_date_m3":34.51,"timestamp":"2021-02-24T12:53:46Z","device":"im871a[00102463]","rssi_dbm":-51}"
(shell) env "METER_TYPE=q400"
(shell) env "METER_NAME=MyTapWater"
(shell) env "METER_ID=05158278"
(shell) env "METER_TOTAL_M3=42.116"
(shell) env "METER_SET_DATE=2021-02-01"
(shell) env "METER_CONSUMPTION_AT_SET_DATE_M3=34.51"
(shell) env "METER_TIMESTAMP=2021-02-24T12:53:46Z"
(shell) waiting for child 2599 to complete.
(shell) /bin/sh: return code 127
(shell) /bin/sh exited with non-zero return code: 127

BIBOLV, 2021-02-24:
I will check/test and will get back to you - hopefully this week.

joggele777, 2021-02-24:
Super! And - please bear with me; I'm not really a Unix specialist ;-)

BIBOLV, 2021-02-24:
I tested snap and successfully sent mqtt message so it should work. Couple of things for you to try - look at system logs, maybe it will shed some light `journalctl --since=yesterday | less` or just wmbusmeters logs `journalctl --since=yesterday | grep wmbusmeters.wmbusmeters`. If there is nothing then add `-d` to mosquitto_pub command inside configuration, restart and look at system logs again when mqtt message is being sent.
snap is something similar to docker image or very simple vm - files that are needed for him to work is inside of snap, closest you can get to inside of snap is running `snap run --shell wmbusmeters` and then you can try to run again `mosquitto_pub -d -t testTopic -m "Hello world!"` and then mosquitto_pub from inside snap should be used. 
If non of it will help I will have to find and fire up pi to test on actual hw you are having issues on.

joggele777, 2021-02-25:
mosquitto_pub not found ?

Feb 25 07:54:02 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:54:18 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:54:18 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:54:35 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:54:35 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:54:51 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:54:51 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:55:08 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:55:08 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:55:24 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:55:24 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:55:41 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:55:41 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
pi@raspberrypi:/var/log$ 

Feb 25 07:55:41 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:55:57 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:55:57 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:56:14 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:56:14 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:56:30 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:56:30 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found
Feb 25 07:56:47 raspberrypi wmbusmeters.wmbusmeters[16479]: ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
Feb 25 07:56:47 raspberrypi wmbusmeters.wmbusmeters[16479]: /bin/sh: 1: /usr/bin/mosquitto_pub: not found

BIBOLV, 2021-02-25:
Tnx for info - will investigate.

joggele777, 2021-02-25:
I think I should install the snap package from mosquitto? But I can only find the package including broker but not only client. Since I also have the mqtt broker from iobroker running on the same pi, there might be problems. What do you think?

BIBOLV, 2021-02-25:
No, it's not the issue and/or solution in this case.

joggele777, 2021-02-25:
OK, it was just a suspicion. Then I'll wait and see if you find out anything else.

BIBOLV, 2021-02-26:
Please update mqtt line to `shell=LD_LIBRARY_PATH=/var/lib/snapd/lib/gl:/var/lib/snapd/lib/gl32:/var/lib/snapd/void:/snap/wmbusmeters/current/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/usr/lib/x86_64-linux-gnu:/snap/wmbusmeters/current/lib:/snap/wmbusmeters/current/usr/lib:/snap/wmbusmeters/current/usr/lib/arm-linux-gnueabihf /snap/wmbusmeters/current/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/$METER_ID -m "$METER_JSON"` restart service and check if it helps.
As for `ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.` - it is just a warning and there seams to be no fix for that other than commenting out `/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so` from `/etc/ld.so.preload`, so if it bothers you, just comment it out and error should disappear.

joggele777, 2021-02-27:
Thanks a lot!

Have configured the shell call in conf accordingly.

Since now again is weekend I have to wait until Monday. Will then inform again.

joggele777, 2021-03-01:
Works properly!  :-)

pi@raspberrypi:/var/snap/wmbusmeters/common/logs$ iobroker state get mqtt.0.wmbusmeters.05158278
{"val":"{\"media\":\"water\",\"meter\":\"q400\",\"name\":\"MyTapWater\",\"id\":\"05158278\",\"total_m3\":43.936,\"set_date\":\"2021-03-01\",\"consumption_at_set_date_m3\":43.932,\"timestamp\":\"2021-03-01T04:47:09Z\",\"device\":\"im871a[00102463]\",\"rssi_dbm\":-56}","ack":true,"ts":1614574029691,"q":0,"from":"system.adapter.mqtt.0","user":"system.user.admin","lc":1614574029691}

Thanks again!

weetmuts, 2021-03-06:
Great! Closing this now.

closed 2021-03-06 by weetmuts

joggele777, 2021-05-31:
Hello again,

unfortunately, something seems to have stopped working for me recently. Consciously I have not changed anything in the configuration. The following entries are in the LOG.

No wmbus device detected, waiting for a device to be plugged in.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Wmbusmeters daemon (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.

Could it be that the m-bus adapter is defective?

Regards
Jochen

BIBOLV, 2021-05-31:
Hi, it means that there are more than one instance of wmbusmeters running on system. Please check processes.

joggele777, 2021-05-31:
Wow - so fast :-)

With my modest Unix knowledge:

pi@raspberrypi:/var/snap/wmbusmeters/common/logs$ sudo ps -A | grep wmbus
12578 ?        00:00:14 wmbusmeters

-> only one instance?

BIBOLV, 2021-05-31:
Is there some other process that could use dongle? You could try to restart snap, reconnect dongle and if any from above does not help - restart pi.

joggele777, 2021-06-01:
Still same issue - iobroker is running on same device - but no m-bus adapter installed/configured at iobroker.

i think i should completely rebuild my pi. Currently I have PI-OS with snap. Would it be better to use Ubuntu or FreeBSD? Or would Fedora also work under PI-OS? What would be the recommendation?

BIBOLV, 2021-06-01:
I doubt that reinstall will help but you can try on another sdcard. Regarding OS - PiOS is fine if you do not want to try some other distro yourself. Wmbusmeter logs are same also after reboot? Can you post here system logs and wmbusmeter logs when you disconnected and reconnect dongle?

joggele777, 2021-06-01:
disconnect / reconnect dongle

SYSLOG:
Jun  1 13:40:45 raspberrypi kernel: [20696.541976] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
Jun  1 13:40:45 raspberrypi kernel: [20696.546833] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
Jun  1 13:40:45 raspberrypi mbusd[467]: 01 Jun 2021 13:40:45 tty: error in read() (port closed)
Jun  1 13:40:45 raspberrypi kernel: [20696.614945] usb 1-1.1: USB disconnect, device number 4
Jun  1 13:40:45 raspberrypi kernel: [20696.615360] cp210x ttyUSB0: failed set request 0x12 status: -19
Jun  1 13:40:45 raspberrypi kernel: [20696.615379] cp210x ttyUSB0: failed set request 0x0 status: -19
Jun  1 13:40:45 raspberrypi kernel: [20696.616268] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
Jun  1 13:40:45 raspberrypi kernel: [20696.623201] cp210x 1-1.1:1.0: device disconnected
Jun  1 13:40:45 raspberrypi mbusd[467]: 01 Jun 2021 13:40:45 tty_reopen(): can't open tty device /dev/ttyUSB0 (No such file or directory)
Jun  1 13:40:46 raspberrypi systemd-udevd[15121]: Process '/usr/lib/snapd/snap-device-helper unbind snap_wmbusmeters_wmbusmeters /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0 0:0' failed with exit code 1.
Jun  1 13:40:46 raspberrypi systemd-udevd[15121]: Process '/usr/lib/snapd/snap-device-helper unbind snap_wmbusmeters_wmbusmeters /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1 189:3' failed with exit code 1.
Jun  1 13:40:46 raspberrypi mbusd[467]: 01 Jun 2021 13:40:46 tty_reopen(): can't open tty device /dev/ttyUSB0 (No such file or directory)
Jun  1 13:40:46 raspberrypi mbusd[467]: 01 Jun 2021 13:40:46 tty_reopen(): can't open tty device /dev/ttyUSB0 (No such file or directory)
Jun  1 13:40:46 raspberrypi mbusd[467]: 01 Jun 2021 13:40:46 tty_reopen(): can't open tty device /dev/ttyUSB0 (No such file or directory)
Jun  1 13:40:47 raspberrypi mbusd[467]: 01 Jun 2021 13:40:47 tty_reopen(): can't open tty device /dev/ttyUSB0 (No such file or directory)
Jun  1 13:40:49 raspberrypi mbusd[467]: 01 Jun 2021 13:40:49 tty_reopen(): can't open tty device /dev/ttyUSB0 (No such file or directory)
Jun  1 13:40:52 raspberrypi mbusd[467]: 01 Jun 2021 13:40:52 tty_reopen(): can't open tty device /dev/ttyUSB0 (No such file or directory)
Jun  1 13:40:54 raspberrypi kernel: [20704.858279] usb 1-1.1: new full-speed USB device number 5 using xhci_hcd
Jun  1 13:40:54 raspberrypi kernel: [20704.996208] usb 1-1.1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
Jun  1 13:40:54 raspberrypi kernel: [20704.996227] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun  1 13:40:54 raspberrypi kernel: [20704.996244] usb 1-1.1: Product: WiMOD iM871A-usb
Jun  1 13:40:54 raspberrypi kernel: [20704.996261] usb 1-1.1: Manufacturer: Silicon Labs
Jun  1 13:40:54 raspberrypi kernel: [20704.996276] usb 1-1.1: SerialNumber: 020F9B22
Jun  1 13:40:54 raspberrypi kernel: [20705.002466] cp210x 1-1.1:1.0: cp210x converter detected
Jun  1 13:40:54 raspberrypi kernel: [20705.014958] usb 1-1.1: cp210x converter now attached to ttyUSB0
Jun  1 13:40:54 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Jun  1 13:40:54 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
Jun  1 13:40:54 raspberrypi snapd[484]: hotplug.go:199: hotplug device add event ignored, enable experimental.hotplug
Jun  1 13:40:54 raspberrypi systemd-udevd[15155]: Process '/usr/lib/snapd/snap-device-helper bind snap_wmbusmeters_wmbusmeters /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1/1-1.1:1.0 0:0' failed with exit code 1.
Jun  1 13:40:54 raspberrypi mtp-probe: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1"
Jun  1 13:40:54 raspberrypi mtp-probe: bus: 1, device: 5 was not an MTP device
Jun  1 13:40:54 raspberrypi systemd-udevd[15155]: Process '/usr/lib/snapd/snap-device-helper bind snap_wmbusmeters_wmbusmeters /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1 189:4' failed with exit code 1.
pi@raspberrypi:/var/log$ 

wmbusmeter still same log entries:
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.
Notice! Other wmbusmeters (pid 0) is running and it might hog any wmbus devices.

joggele777, 2021-06-02:
Were you able to find out anything else from the LOG?

BIBOLV, 2021-06-04:
Nothing really interesting in logs. You can also try to check are all snap interfaces still connected - `snap connections wmbusmeters`

joggele777, 2021-06-05:
pi@raspberrypi:/var/log$ snap connections wmbusmeters
Interface       Plug                        Slot             Notes
network         wmbusmeters:network         :network         -
network-bind    wmbusmeters:network-bind    :network-bind    -
raw-usb         wmbusmeters:raw-usb         :raw-usb         manual
system-observe  wmbusmeters:system-observe  :system-observe  manual
pi@raspberrypi:/var/log$

BIBOLV, 2021-06-06:
I would suggest to stop all services, restart snap and check if it helps. If not then disable and stop snap and compile from source - see if it helps, maybe something bad with snap after all. If it does not, then make new installation on different sdcard.

joggele777, 2021-06-09:
After a long search, I have now found the culprit. Modbus gateway service :-( - disabled - and wmbusmeters is running fine again :-)

[7 more replies]