wmbusmeters/wmbusmeters #253: iPerl water meter - mBus encryption not per meter(?)

konkrog, 2021-02-19:
This is not really an issue with the code, but rather with decrypting the data received. The vendor is claiming that this particular meter has the same encryption key for all devices (for mBus), so they cannot supply me the encryption key... The only option would be to switch off the encryption on my meter.

I have asked the vendor to supply some supporting documentation to this claim, but that will take some time...

Anyone have any experience with this particular device who can comment on this?

Including a snip from the log:

(telegram) DLL L=1e C=44 (from meter SND_NR) M=4cae (SEN) A=xxxxxxxx VER=68 TYPE=07 (Water meter) (driver iperl) DEV=rtlwmbus[00000001] RSSI=110
(telegram) TPL CI=7a ACC=c7 STS=90 CFG=0510 (AES_CBC_IV nb=1 cntn=0 ra=0 hc=0)
Received telegram from: 20651282
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07)
                   ver: 0x68
                device: rtlwmbus[00000001]
                  rssi: 110 dBm
                driver: iperl
telegram=||1E44AE4C8212652068077AC79010057EAFFDA9E682404DD90854D6094DFDD3|+45

Regards,
Knut

weetmuts, 2021-02-21:
Are you talking to the water vendor or the meter vendor?

A water-company can decide how their meters should be encrypted. If the company
wants a less secure solution with a single key for all meters, then so be it. :-)
I do not think it is a property of the iPerl meter itself.

If you are buying an iPerl meter from the meter-producer or a reseller, 
then for sure you should be able to have your own key. Anything else would be
very strange.

But Since they offer you a solution, take it. :-)

konkrog, 2021-02-21:
Hi Fredrik, thanks for the response. 

I have talked with both the water vendor as well as the meter vendor.  The meter vendor says it cannot provide the key.... but I think this is not correct.

I'm absolutely considering requesting encryption to be switched off.  However, it would be better to have the key... the data is after all considered sensitive information. 

It would be good to know the true capabilities of the device. I cannot believe that there is a single encryption key for all the devices... That would basically mean that the encryption cannot be trusted... as the key might already be leaked. 

I do believe that it should be possible to set the encryption key - like you can do on other devices. At the very least I think the manufacturer has a database with the encryption keys per device.   

If anyone has documentation on this device, it would be interesting to look into that...

Regards,
Knut

weetmuts, 2021-02-27:
I am certain that the meter itself can be bought with a unique key that you get when you buy the meter.
If they cannot give you that, then buy a meter from another company like Multical21 from Kamstrup.
.
However if the meter is already installed in your house, ie you did not buy it, it is part of the water company setup,
then for sure, you are limited to whatever the water company + the meter company are willing to offer you.

Go for the non-ecrypted approach, it seems unlikely that someone would both know that you have an 
un-encrypted meter and have a reason to eavesdrop. :-)

weetmuts, 2021-04-09:
Closing this for now. Please ping me again in this ticket or open a new ticket if you still have problems.

closed 2021-04-09 by weetmuts