wmbusmeters/wmbusmeters #254: SIGABORT after receiving first complete frame

rubdos, 2021-02-19:
Config:

```
loglevel=debug
device=rtlwmbus:868.9M
logtelegrams=true
format=json
shell=/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/"$METER_NAME" -m "$METER_JSON"
```

Meter:

```
name=4A12
type=multical302
id=67594...
key=...
```

After receiving the first full frame, which gets decrypted, decoded and sent to mosquitto and into Home Assistant (yay!), wmbusmeters crashes a short while after:

```
(serial) received ascii "C1;1;1;2021-02-19 20:26:08.000;135;109;67594030;0x27442d2c3040596730048d205cb0eba221213681fb46bf41c8893a0be1fdb88a6be6f99e341d<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2021-02-19 20:26:08.000;135;109;67594030;0x27442d2c3040596730048d205cb0eba221213681fb46bf41c8893a0be1fdb88a6be6f99e341d<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 38
(meter) 4A12: for me? 67594030
(meter) 4A12: yes for me
(meter) 4A12 67594030 "25442D2C3040596730048D205CB0EBA221213681FB46BF41C8893A0BE1FDB88A6BE6F99E341D"
(wmbus) parseDLL @0 38
(wmbus) parseELL @10 28
(ELL) decrypting "213681FB46BF41C8893A0BE1FDB88A6BE6F99E341D"
(ELL) IV 2D2C30405967300420B0EBA221000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "D4A779E7F159D4B50600950600914700"
(ELL) block 1 block_size 5 offset 16
(ELL) decrypted "BF21000000"
(ELL) decrypted "D4A779E7F159D4B50600950600914700BF21000000"
(wmbus) parseAFL @19 19
(wmbus) parseTPL @19 19
(wmbus) ignoring compressed telegram since format signature hash 0xf1e7 is yet unknown.
     this is not a problem, since you only need wait for at most 8 telegrams
     (8*16 seconds) until an full length telegram arrives and then we know
     the format giving this hash and start decoding the telegrams properly.
(wmbus) telegram ignored by all configured meters!
(serial) received ascii "C1;1;1;2021-02-19 20:26:08.000;135;136;67594029;0x27442d2c2940596730048d207200eca22124a8b7b0707092eadcb94fd59ec5786148cd718dad<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2021-02-19 20:26:08.000;135;136;67594029;0x27442d2c2940596730048d207200eca22124a8b7b0707092eadcb94fd59ec5786148cd718dad<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 38
(meter) 4A12: for me? 67594029
(meter) 4A12: not for me: not my id
(wmbus) telegram ignored by all configured meters!
(serial) received ascii "C1;1;1;2021-02-19 20:26:12.000;107;107;67594038;0x27442d2c3840596730048d208d60eca2213ca6a3b3340dc7857cc0a1968186d57f269cf6a6a9<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2021-02-19 20:26:12.000;107;107;67594038;0x27442d2c3840596730048d208d60eca2213ca6a3b3340dc7857cc0a1968186d57f269cf6a6a9<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 38
(meter) 4A12: for me? 67594038
(meter) 4A12: not for me: not my id
(wmbus) telegram ignored by all configured meters!
(serial) received ascii "C1;1;1;2021-02-19 20:26:13.000;135;117;67594026;0x27442d2c2640596730048d200421eca22102ad501e550d46b53f457b777ad7c8f77d191ad2fb<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2021-02-19 20:26:13.000;135;117;67594026;0x27442d2c2640596730048d200421eca22102ad501e550d46b53f457b777ad7c8f77d191ad2fb<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 38
(meter) 4A12: for me? 67594026
(meter) 4A12: not for me: not my id
(wmbus) telegram ignored by all configured meters!
(serial) received ascii "C1;0;1;2021-02-19 20:26:14.000;109;60;67594039;0x27442d2c3940596730048d20c160eca221b739ac2caf976728068b6b2e1dca0519e1669756bc<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;0;1;2021-02-19 20:26:14.000;109;60;67594039;0x27442d2c3940596730048d20c160eca221b739ac2caf976728068b6b2e1dca0519e1669756bc<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
[... 109 lines cut ...]
(meter) 4A12: for me? 67594030
(meter) 4A12: yes for me
(meter) 4A12 67594030 "25442D2C3040596730048D205EB2EBA22163B1293A4FE83A602A95FDFDE313519D8F5D6B096D"
(wmbus) parseDLL @0 38
(wmbus) parseELL @10 28
(ELL) decrypting "63B1293A4FE83A602A95FDFDE313519D8F5D6B096D"
(ELL) IV 2D2C30405967300420B2EBA221000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "D4A779E7F159D4B50600950600914700"
(ELL) block 1 block_size 5 offset 16
(ELL) decrypted "BF21000000"
(ELL) decrypted "D4A779E7F159D4B50600950600914700BF21000000"
(wmbus) parseAFL @19 19
(wmbus) parseTPL @19 19
(dvparser) found remembered format for hash f1e7
(dvparser) using format "030643060314426C022D01FF21"
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = unsigned char; _Alloc = std::allocator<unsigned char>; std::vector<_Tp, _Alloc>::reference = unsigned char&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Aborted (core dumped)
```

Context: I'm in an apartment complex of 69 apartments that all have been fitted with such a 302. It seems like wmbusmeters crashes on the next partial frame, but I'm not sure.

Please do let me know what you want me to try. I have `gdb` skills if needed, or I can provide you ssh access to the machine to try yourself. I tried on both the current Fedora 31 version (0.9.30) and on the Fedora 33 version (0.9.36) and on the current master; all with the same result.

Thanks for maintaining this software! It's going to be fun to play with :-)

weetmuts, 2021-02-21:
Can you build the latest master from github using
"make DEBUG=true"
and then run again? This will hopefully print more information!

rubdos, 2021-02-24:
The DEBUG version seems to work... 

```
(serial) received ascii "C1;1;1;2021-02-24 09:15:00.000;128;128;67594030;0x27442d2c3040596730048d2000c083a421ddcdc6b5411e22f96c34ce36cd76b690e33f5f3cfa<0A>"
(rtlwmbus) checkRTLWMBusFrame "C1;1;1;2021-02-24 09:15:00.000;128;128;67594030;0x27442d2c3040596730048d2000c083a421ddcdc6b5411e22f96c34ce36cd76b690e33f5f3cfa<0A>"
(rtlwmbus) received full frame
(wmbus) parseDLL @0 38
(wmbus) parseELL @10 28
(meter) 4A12: for me? 67594030
(meter) 4A12: yes for me
(meter) 4A12 multical302 handling telegram from 67594030
(meter) 4A12 67594030 "25442D2C3040596730048D2000C083A421DDCDC6B5411E22F96C34CE36CD76B690E33F5F3CFA"
(wmbus) parseDLL @0 38
(telegram) DLL L=25 C=44 (from meter SND_NR) M=2c2d (KAM) A=67594030 VER=30 TYPE=04 (Heat meter) (driver multical302) DEV=rtlwmbus[00000001] RSSI=128
(wmbus) parseELL @10 28
(ELL) decrypting "DDCDC6B5411E22F96C34CE36CD76B690E33F5F3CFA"
(ELL) IV 2D2C30405967300420C083A421000000
(ELL) block 0 block_size 16 offset 0
(ELL) decrypted "7A8479E7F13422BA06009506009F4700"
(ELL) block 1 block_size 5 offset 16
(ELL) decrypted "BF21000000"
(ELL) decrypted "7A8479E7F13422BA06009506009F4700BF21000000"
(telegram) ELL CI=8d CC=20 (slow_resp sync) ACC=00 SN=c083a421 (AES_CTR session=0 time=1722428) CRC=7a84
(wmbus) parseAFL @19 19
(wmbus) parseTPL @19 19
(dvparser) found remembered format for hash f1e7
(dvparser) using format "030643060314426C022D01FF21"
(telegram) TPL CI=79
telegram=|25442D2C3040596730048D2000C083A4217A8479E7F13422|BA06009506009F4700BF21000000|+88
(multical302) 00: 25 length (37 bytes)
(multical302) 01: 44 dll-c (from meter SND_NR)
(multical302) 02: 2d2c dll-mfct (KAM)
(multical302) 04: 30405967 dll-id (67594030)
(multical302) 08: 30 dll-version
(multical302) 09: 04 dll-type (Heat meter)
(multical302) 0a: 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
(multical302) 0b: 20 ell-cc (slow_resp sync)
(multical302) 0c: 00 ell-acc
(multical302) 0d: c083a421 sn (AES_CTR)
(multical302) 11: 7a84 payload crc (calculated 7a84 OK)
(multical302) 13: 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
(multical302) 14: e7f1 format signature
(multical302) 16: 3422 data crc
(multical302) 18: * BA0600 total energy consumption (1722.000000 kWh)
(multical302) 1b: * 950600 target energy consumption (1685.000000 kWh)
(multical302) 1e: * 9F4700 total volume (183.350000 m3)
(multical302) 21: * BF21 target date (2021-01-31 00:00)
(multical302) 23: * 0000 current power consumption (0.000000 kW)
(multical302) 25: * 00 info codes ()
(shell) exec "/bin/sh"
(shell) arg "-c"
(shell) arg "/usr/bin/mosquitto_pub -h localhost -t wmbusmeters/"$METER_NAME" -m "$METER_JSON""
(shell) env "METER_JSON={"media":"heat","meter":"multical302","name":"4A12","id":"67594030","total_energy_consumption_kwh":1722,"current_power_consumption_kw":0,"total_volume_m3":183.35,"at_date":"2021-01-31 00:00","total_energy_consumption_at_date_kwh":1685,"current_status":"","timestamp":"2021-02-24T09:15:00Z","device":"rtlwmbus[00000001]","rssi_dbm":128}"
(shell) env "METER_TYPE=multical302"
(shell) env "METER_NAME=4A12"
(shell) env "METER_ID=67594030"
(shell) env "METER_TOTAL_ENERGY_CONSUMPTION_KWH=1722"
(shell) env "METER_CURRENT_POWER_CONSUMPTION_KW=0"
(shell) env "METER_TOTAL_VOLUME_M3=183.35"
(shell) env "METER_AT_DATE=2021-01-31 00:00"
(shell) env "METER_TOTAL_ENERGY_CONSUMPTION_AT_DATE_KWH=1685"
(shell) env "METER_CURRENT_STATUS="
(shell) env "METER_TIMESTAMP=2021-02-24T09:15:00Z"
(shell) waiting for child 1907529 to complete.
(shell) /bin/sh: return code 0
```

... and the non-debug from git too. This is weird, I tested the git version when I posted. Maybe the upgrade from Fedora 31 to 33 did it (but I did run the git version...)

weetmuts, 2021-02-26:
The first error looks like something that happens when the build was a partial broken rebuild. make clean, then make again, is often useful to get rid of them.
Hmm, I have to take a look at the auto-generated dependencies in the makefile

rubdos, 2021-02-26:
FWIW, the git version (both release and debug) work. I've copied the wmbusmetersd release build to `/usr/local/bin/` and all is happy now. I suspect that I misremember really testing the git version...

weetmuts, 2021-02-27:
Cool! Closing this for now then.

closed 2021-02-27 by weetmuts