wmbusmeters/wmbusmeters #257: Izar RC868 Duplicate Telegrams and JSON Key Selection

theWagNerd, 2021-02-23:
Hi,

Apologies but I've got a couple of questions - one is possibly an "issue" / query, whereas the other is just a question:

Q1: Since I've done reinstall / relocation, I've noticed that I seem to get "duplicates":

> pi@raspberrypi:~ $ rtl_sdr -f 868.9M -s 1600000 - 2 | rtl_wmbus
> 
> Found 1 device(s):
>   0:  Realtek, RTL2838UHIDIR, SN: 00000001
> 
> Using device 0: Generic RTL2832U OEM
> Detached kernel driver
> Found Fitipower FC0012 tuner
> Sampling at 1600000 S/s.
> Tuned to 868900000 Hz.
> Tuner gain set to automatic.
> Reading samples in async mode...
> Allocating 15 zero-copy buffers
> T1;0;0;2021-02-23 21:26:31.000;56;54;A0C76986;0x1944304c8669c7a00604a211ff0013490b4a583f2ebf864d0ae5
> T1;1;1;2021-02-23 21:26:36.000;114;83;A0C74FD7;0x1944304cd74fc7a00604a2110f001300552209ec9d46afe360f5
> T1;1;1;2021-02-23 21:26:36.000;74;129;A0C74FD7;0x1944304cd74fc7a00604a2110f001300552209ec9d46afe360f5
> T1;1;1;2021-02-23 21:26:37.000;128;164;A0C7625F;0x1944304c5f62c7a00604a2210f001370f4b4ed766ec795d170e6
> T1;1;1;2021-02-23 21:26:37.000;125;162;A0C7625F;0x1944304c5f62c7a00604a2210f001370f4b4ed766ec795d170e6

There is only one tuner device attached so it's not like the telegram is being received by different tuners, or is it possible that I've got 2 tuners installed within one card? 

Q2: I'm MQTTing the output using the shell command within the config file.  I'm only interested in receiving the total_m3 and rssi_dbbm.

> shell=/usr/bin/mosquitto_pub -h 192.168.2.50 -p 1883 -u emonpi -P emonpimqtt2016 -t emon/$METER_NAME -m "$METER_JSON"
.. this provides every env variable for the meter type including the signal strength

I'm only interested in receiving total_m3 and signal strength. Interestingly, meter strength is not listed as a env variable for the izar meter yet it is passed through as a variable (tagged in emonCMS as rssi_dbm. I've tried modifying the shell statement as follows:
> shell=/usr/bin/mosquitto_pub -h 192.168.2.50 -p 1883 -u emonpi -P emonpimqtt2016 -t emon/$METER_NAME -m $METER_TOTAL_M3, $METER_RSSI_DBM
.. I get the volume passed through but not the signal strength - what should I be identifying the key as?

Thanks

BIBOLV, 2021-02-24:
Hi,

It is known that rtl_sdr sends duplicate messages that's why `ignoreduplicates=true` was introduced as default option by this [commit](https://github.com/weetmuts/wmbusmeters/commit/301c91ea68150234ea4df078bd0f3c5617f1988e). So if you are running latest version, it should not be a problem, if not, you can add `ignoreduplicates=true` in config and it should help.
As for "env variables", you can check what is provided by running `wmbusmeters --listenvs=izar` and if rssi will not be there (and I guess it will not be), then if you absolutely need it, you will have to forward whole JSON message to emon and do some filtering at app side and collect only data you need.

theWagNerd, 2021-02-24:
@BIBOLV - wasn't aware of the rtl_sdr duplication, thanks for that.

Re: "env variables" - yeah I had run the listenvs=izar option hence why I said it wasn't listed as a variable, but was surprised that it was output as part of the full JSON message. Appreciate that I could do some filtering / parsing, but thought given that it is contained within the JSON message, it should be extractable at source as a key. I'd assumed I was doing something wrong (not unheard of), or if it was possibly an issue?

weetmuts, 2021-02-24:
Yes, rtl_wmbus now performs two decode attempts using different algorithms, in parallel, where none, one or both can succeed. 
So often telegrams are duplicated. Fortunately there was already support for dropping duplicates, since this can happen
in other situations if you connect multiple dongles for better coverage or there are wmbus repeaters and you accidentally hear the original telegram and the repeated.

It was just a mistake that METER_DEVICE and METER_RSSI_DBM where not in the shell envs. I also noted
that METER_MEDIA was missing. I have added these now and pushed to github. Please test.

theWagNerd, 2021-02-24:
@weetmuts - Thanks for the looking into this and pushing it through. Still not working for me at the present. I've got the following line in my config but I'm still getting null values for the rssi_dbm key (meter_total_m3 is fine)

I performed git-pull and then make install. 

>shell=/usr/bin/mosquitto_pub -h 192.168.2.50 -p 1883 -u emonpi -P emonpimqtt2016 -t emon/$METER_NAME -m "$METER_TOTAL_M3, $METER_RSSI_DBM"

When I use the full JSON output, the RSSI_DBM outputs numerical values as you'd expect. Possibly something to do with my syntax in the config shell command though the meter_total_m3 outputs its value okay?

I'll call it a day for now and take another look tomorrow. Thanks again

theWagNerd, 2021-02-25:
Hi,

Still having issues...

Can confirm that METER_RSSI_DBM is now listed as an env variable when I run with --listenvs=izar

When I use:
> shell=/usr/bin/mosquitto_pub -h 192.168.2.50 -p 1883 -u emonpi -P emonpimqtt2016 -t emon/$METER_NAME -m "$METER_JSON"
I get numercial values for all the keys I expect to, including RSSI_DBM. For non-numerical values, I get null (I'm using emonCMS) so I'm okay with that as I suspect it is looking for a numerical value and if it doesn't see one, it returns null). 

When I use:
> shell=/usr/bin/mosquitto_pub -h 192.168.2.50 -p 1883 -u emonpi -P emonpimqtt2016 -t emon/$METER_NAME -m "$METER_TOTAL_M3, $METER_LAST_MONTH_TOTAL_M3"
I get both key values, e.g. 1663 and 1647

When I use the same syntax but try to extract just the TOTAL_M3 and RSSI_DBM:
> shell=/usr/bin/mosquitto_pub -h 192.168.2.50 -p 1883 -u emonpi -P emonpimqtt2016 -t emon/$METER_NAME -m "$METER_TOTAL_M3, $METER_RSSI_DBM"
I only get TOTAL_M3 value and null for RSSI_DBM (I'm guessing its telling me it is either there isn't a key RSSI_DBM, or it has no value, or it is of a different variable type to TOTAL_M3 and LAST_MONTH_TOTAL_M3?). Are there any logs I can run to see what data / variable type is being extracted from that key, or am I looking in the wrong direction?

Thanks

theWagNerd, 2021-02-25:
Sorted - Thanks!!

closed 2021-02-25 by theWagNerd