wmbusmeters/wmbusmeters #258: snap installation supported?

antosim, 2021-02-24:
I found a snap installation, [https://snapcraft.io/install/wmbusmeters/mint](https://snapcraft.io/install/wmbusmeters/mint)  but it seems like nothing was installed.

Is the snap install supported?

//Anders

BIBOLV, 2021-02-24:
Yes, there is experimental support for snaps - please refer to [simple manual](https://github.com/weetmuts/wmbusmeters/blob/master/snap/README.md) for installation. If something does not work - please let us know.

scroatamus, 2021-03-04:
I have installed & seen T1 packets with  rtl_sdr and rtl_wmbus.
T1;1;1;2021-03-04 08:55:42.000;15;14;A132257D;0x1944304c7d2532a1b201a2711d0013145810cf2de2183efea212
So I move on to wmbusmeters snap on an rpi3
I cant connect raw-usb. I execute command & receive error as below.
sudo snap connect wmbusmeters:raw-usb core:raw-usb
error: snap "core" has no slot named "raw-usb"

BIBOLV, 2021-03-04:
We have identified issues when using rtl_sdr dongles with snaps. We are working on a solution but it will not be fast, so I would suggest in meantime to use compiled version of wmbusmeters or use docker container. Once we will come up and implement fix I will let you know.

antosim, 2021-03-22:
I get this when I check the status after initiated a start:

 sudo systemctl status snap.wmbusmeters.wmbusmeters.service
● snap.wmbusmeters.wmbusmeters.service - Service for snap application wmbusmeters.wmbusmeters
     Loaded: loaded (/etc/systemd/system/snap.wmbusmeters.wmbusmeters.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2021-03-22 16:54:16 CET; 10s ago
    Process: 506927 ExecStart=/usr/bin/snap run wmbusmeters (code=exited, status=1/FAILURE)
   Main PID: 506927 (code=exited, status=1/FAILURE)

Mar 22 16:54:16 chieftech systemd[1]: snap.wmbusmeters.wmbusmeters.service: Scheduled restart job, restart counter is at 5.
Mar 22 16:54:16 chieftech systemd[1]: Stopped Service for snap application wmbusmeters.wmbusmeters.
Mar 22 16:54:16 chieftech systemd[1]: snap.wmbusmeters.wmbusmeters.service: Start request repeated too quickly.
Mar 22 16:54:16 chieftech systemd[1]: snap.wmbusmeters.wmbusmeters.service: Failed with result 'exit-code'.
Mar 22 16:54:16 chieftech systemd[1]: Failed to start Service for snap application wmbusmeters.wmbusmeters.

BIBOLV, 2021-03-22:
Tnx for info - yes, we are aware of this issue. It is the same issue I mentioned in previous comment.

Kosygor, 2021-05-23:
Hello.

Was there any progress with this iisue?
Today I have installed wmbusmeters via snap. I have successfully done everything systemctl process is running but in logs I have: 

```
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>/bin/sh: 1: rtl_sdr: not found
<<<
(serial) EVENT thread interrupted
(shell) return code 127
(shell) 15250 exited
(main) rtl_sdr NOT found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>>/bin/sh: 1: rtl_wmbus: not found
<<<
(serial) EVENT thread interrupted
(shell) return code 127
(shell) 15251 exited
(main) rtl_wmbus NOT found in path
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_sdr in the path!
Warning! Auto scan has found an rtl_sdr dongle, but you have no rtl_wmbus in the path!
```

I have tried original (was running in daemon directly on Linux) solution device=rtlwmbus:868.9M and auto:869.9M/auto:c1/rtl_wmbus:868.9M.

Is this issue related to mine, or t is something else?

BIBOLV, 2021-05-23:
Hi,
Yes - it is the same issue I mentioned in previous comment. We are working on fixing issues. There has been some progress, so you can use edge channel to test but it is not fully functional yet.

Kosygor, 2021-05-24:
@BIBOLV tested edge channel. Seems to be working ok with one exceptions there are a lot incomplete frames (this could be my setup - did not monitor it lately). It shares msgs via mqtt. My only concern is this 
`(wmbus) skipping already handled telegram.` because most of telegrams will be same when they come from smoke detector. But I assume that this will not affect situation when status of detector turns from `OK `to `SMOKE`.

BIBOLV, 2021-05-24:
Edge channel contains latest fixes and should be fully functional now. Regarding duplicates - you can add `ignoreduplicates=false` in config but it might result in showing one message twice, so you should test.

pswiatki, 2021-05-24:
I have a more general question: is snap maintained in a similar fashion as e.g. the linux kernel? While in principle it seems to be a good idea (to make it easy to install packages and that they have minimal number of dependencies) I am trying to find out if those snaps are safe/secure and trustworthy.

BIBOLV, 2021-05-25:
In short - I would say, yes. 
But as there are many moving parts and components, I will try to make some high level overview and comments without too deep tech details. So if we are looking from top-down - first and foremost you must [trust](https://snapcraft.io/blog/trust-and-security-in-the-snap-store) application and developers/community behind. Then there is question about possible attack vector - in wmbusmeters case there is no external listening interface currently, so only local attack is probable. To prevent any dangerous activities from applications inside snap towards underlaying OS there are safety measures in place both in userspace and kernel that is being uses for some time already, like AppArmor, Seccomp and cgroups. Basically only way how to get access to underlaying system is trough [interfaces](https://snapcraft.io/blog/security-corner-snap-interface-snap-connections) and those are under end-user control. 
Now about updates - in normal scenario updates are being done automatically in background for snap stuff, so again looking from top-down:
-  application snap - should contain minimal set of additional packages/binaries/libraries, but those are being updates each time when new snap is being built. If package that is used in snap has some new security fixes then snap developer receives notification about it and must initiate snap rebuild (or was it in case of docker images? I might be wrong here 😉) And as I wrote before - in normal scenario updates are being done automatically in background.
- snapd components, like core - provided by snapd developers and also updates automatically.
- underlaying OS - end-user`s responsibility to keep up-to-date.

I hope it was clear and not too long 😄 but if this question is of interest then there are many [resources](https://snapcraft.io/blog/where-eagles-snap-snap-security-overview) to read and understand this technology in more details.