wmbusmeters/wmbusmeters #259: Assertion 'month >= 0' failed

ursf, 2021-02-27:
I always end up with this error whenever one of my evo868 meters is received (or run in simulation mode):

```
(telegram) DLL L=ac C=44 (from meter SND_NR) M=3424 (MAD) A=20008373 VER=50 TYPE=07 (Water meter) (driver evo868) DEV=im871a[00102457] RSSI=-71
(wmbus) parseELL @10 163
(wmbus) parseAFL @10 163
(wmbus) parseTPL @10 163
(telegram) TPL CI=7a ACC=8e STS=00 CFG=2000 (synchronous)
telegram=|AC4424347383002050077A8E000020|2F2F0413A3330000046D0B2DBB2204FD17000000000E780000000000004413A3230000426C9F2C840113412C000082016CBF21D3013B320300C4016D0B299B288104FD280182046CBF21840413412C0000C40413A3230000840513F31A0000C405131D130000840613850A0000C406134A02000084071300000000C407130000000084081300000000C408130000000084091300000000C4091300000000|+18
wmbusmeters: src/util.cc:1264: int get_days_in_month(int, int): Assertion `month >= 0' failed.
```

weetmuts, 2021-02-27:
Oups. Fix pushed!

weetmuts, 2021-03-06:
Closing this now. If it does not work please open a new issue!

closed 2021-03-06 by weetmuts