wmbusmeters/wmbusmeters #260: Techem Radio 4 water meter unknown driver

KeelOli, 2021-03-01:
Dear WMBus Community,

I got two Techem Radio 4 WMBus water meters which are configured to OMS Standart. At the moment the program outputs driver unknown. Therefore I copied here the information of the meters and the telegram:

1. Meter
Serialnumber meter: 05152676 Serialnumber transmiter: 04060299 Key: F9213142F537196E5A9C061CEA2E6A56

`Received telegram from: 04060299
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
                device: im871a[00102489]
                  rssi: -75 dBm
                driver: unknown!
telegram=||494468509902060495377276261505A85CFE07FB003005D2CB895D36363F75C7E7BE88ADCB7105B856FF8912705A7B0DE4AC2852FBF334E15DF201F1421E656938C1200AAC9E360F0542|+84
`
2. Meter
Serialnumber meter: 94123586 Serialnumber transmiter: 94104494 Key: F417C018B5295D90BF4013A21F12573D

`Received telegram from: 94104494
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
                device: im871a[00102489]
                  rssi: -80 dBm
                driver: unknown!
telegram=||494468509444109495377286351294A85CFE07A9003005896145CAA35B3BF8F3940F862500752F089A1628AAFB0B496A5D7D60DF363BE1027060A0A4C7132225E8E4880C18016E0F31FB|+98
`
I saw in meters.h this type of meter has a T1_bit even though its operated on C1. Could that be the reason?

Thank you for your help!

Cheers,
Oliver
`

weetmuts, 2021-03-06:
Thanks! What is the branding on the meter? When you use the latest wmbusmeters the telegram is reported like this:

weetmuts, 2021-03-06:
Received telegram from: 94104494
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
      Concerning meter: 94123586
          manufacturer: (WEH) E Wehrle, Germany (0x5ca8)
                  type: Water meter (0x07)
                   ver: 0xfe
                driver: unknown!

weetmuts, 2021-03-06:
So it looks like a Wehrle meter, and they do have water meters for sale.
The telegrams contain a single total consumption value! You can listen to this value using
wmbusmeters auto:c1 MyWater2 supercom587 94123586 F417C018B5295D90BF4013A21F12573D
MyWater2	94123586	0.016 m3	2021-03-06 21:27.32

But, Techem, cannot help themselves, a single OMS compliant value, then immediatetly followed
by a lot of vendor proprietary data. Wonder why they are allowed to say that it is OMS compliant then....

(multical21) 19: 04 dif (32 Bit Integer/Binary Instantaneous value)
(multical21) 1a: 13 vif (Volume l)
(multical21) 1b: * 10000000 total consumption (0.016000 m3)
(multical21) 1f: 0F manufacturer specific data 52FCF6A52A90A8D83CA8F7FEAE86990502323D0C70EFF49833C7C1696F75BCABC1E52E6305308D0F31FB

weetmuts, 2021-03-06:
Does your meter have any other product/version/brand name that can be used to name the driver?

KeelOli, 2021-03-07:
I ordered the meters directly from the Techems headquarter in Switzerland. The technical salesperson told me that it is configured to OMS Standart. It is very confusing to me that it appears to be a Wehrle device. When I received the meter the first time I have sent it back because I received the SN from the radio and not the meter. He told me then that he fixed it and send me two meters back. But it seems to be the same again. He sent me a print screen of his configuration and there it indeed says at "Herstellerkennung": WEH which could stand for Wehrle. 

I attached a picture of the meter itself and his print screen. 
[image]
[image]

weetmuts, 2021-03-07:
If you use the latest wmbusmeters from github you will get the proper serial number in the json and you can use 94123586 to select the meter in your config file.

The wmbus standard is pretty complicated, what we see in the telegram is that the radio converter is from Techem,
but the meter is Wehrle. They have two different ids. Older versions of wmbusmeters could not discern this, but
it can now.

There is nothing wrong with the meter, its just that I am thinking what to do with this combination, which driver should it have, and under which name....  Hmmm.

weetmuts, 2021-03-07:
Ok! A decision has been made! I have realized that it is hopeless to name a driver so that there is a correlation that can be made between what is printed on the meter and the driver. For some meters its easy, Multical21 -> multical21 driver,
but as the wmbus products gets more and more advanced it seems that meters are sometimes a combination
from different manufacturers and sold under different brands and product names.

So, from now on. The driver is going to be named the transport layer MANUFACTURER_mediahex (and if necessary _versionhex) Your meter combo therefore gets the meter driver name: weh_07

I have just pushed this driver to github. Please test!

Eventually the driver name is going to be irrelevant. Just specifying serial_nr + key should be enough
for wmbusmeters to decode the telegrams. The driver is then always auto-detected from the telegram meta-data,
ie mfct+media+version.

KeelOli, 2021-03-07:
Thanks for your great work! Just tested it with the new version. Here the message for the two meters:

```
Received telegram from: 04060299
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
      Concerning meter: 05152676
          manufacturer: (WEH) E Wehrle, Germany (0x5ca8)
                  type: Water meter (0x07)
                   ver: 0xfe
                device: im871a[00102489]
                  rssi: -70 dBm
                driver: weh_07
telegram=||494468509902060495377276261505A85CFE07AF003005FE083C19D980053CE6D504966554432AB6D3E0F2B37DEFB6B91812F5B36A391D7F4C21B3BEF1C28319FEF031D33288220F59A6|+19

```

Reading out works perfectly too:

```
{
    "media": "water",
    "meter": "weh_07",
    "name": "weh_07_05152676",
    "id": "05152676",
    "total_m3": 0.013,
    "timestamp": "2021-03-07T15:41:08Z",
    "device": "im871a[00102489]",
    "rssi_dbm": -67
}
```

```
Received telegram from: 94104494
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x95
      Concerning meter: 94123586
          manufacturer: (WEH) E Wehrle, Germany (0x5ca8)
                  type: Water meter (0x07)
                   ver: 0xfe
                device: im871a[00102489]
                  rssi: -75 dBm
                driver: weh_07
telegram=||494468509444109495377286351294A85CFE075A003005C5E7C199F539307180143B0C556E9839135CC2FA68953F67409FADEC0BEA47D7F7FFB1A9A8926408DC09C2EE4B3EF22C0F8C4A|+14

```
```
{
    "media": "water",
    "meter": "weh_07",
    "name": "weh_07_94123586",
    "id": "94123586",
    "total_m3": 0.016,
    "timestamp": "2021-03-07T15:40:21Z",
    "device": "im871a[00102489]",
    "rssi_dbm": -73
}
```

Thank you Frederik for the great support!

weetmuts, 2021-04-09:
Nice! Closing this for now. 

There is now support for specifying the driver "auto" and it will use weh_07 automatically based on the telegram contents.

closed 2021-04-09 by weetmuts