wmbusmeters/wmbusmeters #261: Maddalena RadioEVO 868 a im871a usb

wajta, 2021-03-02:
Hello,
I apologize in advance if my question is too beginner.
I have a usb dongle im871a and two water meters, equipped with a RadioEVO 868 modules. According to the datasheet, 
communication is primarily set to T1. Unfortunately, no telegram is returned to me.

Could anyone help me, please.

Here is the debug statement 
`wmbusmeters --debug auto:t1
(wmbusmeters) version: 1.1.0-10-g4ae4010 with local changes
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 7 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 7 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(device config bytes) "FF000300B32552231000010001FF070232000100000000"
(serial) EVENT thread interrupted
(im871a) config: other link_mode=t1 id=00102352 media=00 version=01 c_field=00 auto_rssi=01
(im871a) are you there? yes 00102352
(main) opening t1
Started auto im871a[00102352] on /dev/ttyUSB0 listening on t1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 7 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00102352
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) set config to set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
`

BIBOLV, 2021-03-02:
Hi,
Specs says that T1 is factory setting, but C1 also is supported, so firstly I would suggest to listen to both modes for extended periods - ~1-2h. If it does not help then contact owner of meter (water company or landlord, etc) and ask what is configuration of module as you will need to contact them either way to get encryption key as message should be encrypted.

merbanan, 2021-03-03:
Or take your rtl-sdr device and use rtl_433 to identify the signals.

weetmuts, 2021-03-06:
Some meters also transmit only on weekdays during working hours.
How far away are you from the meters?

wajta, 2021-03-31:
> Some meters also transmit only on weekdays during working hours.
> How far away are you from the meters?

It is about 6 meters and 2 ytong (10cm) walls.

ursf, 2021-04-03:
TL;DR: The im871a shouldn't be the source of the problem - it definitely can receive signals from RadioEvo868.

I run same setup here (im871a and RadioEvo868 modules) and it works really great. From a relatively central position within the building, I'm able to receive 32 of 35 modules from up to 10 meters away through several massive walls and up to two concrete floors, even through the basement ceiling. Most of the modules are set to C1, only four (which have been installed at a later date) are set to T1. They all send a telegram every two minutes from 5am to 9pm (now 6am to 10pm because of DST). Some do send telegrams on sundays, some don't (although they are installed by the same service provider).

wajta, 2021-04-06:
Thank you for your assurance about im871a.
I tried to move usb dongle to a clean latest installation Raspbian Pi OS on Raspbery PI 4. 
Then I installed snap and wmbusmeters via https://github.com/weetmuts/wmbusmeters/blob/master/snap/README.md, but could not run a command sudo snap connect wmbusmeters:system-observe core:system-observe 
I got an error message error: snap "wmbusmeters" has no plug named "system-observe" so however service is running.
If I want  to run not like a service I get error wmbusmeters: command not found.

Is it possible to run wmbusmeters like an application, not like a service via snap?

BIBOLV, 2021-04-08:
Please try using snap from edge channel as stable channel snap has issues currently. 
Intended snap usage is as service, so I can`t confirm that it would work as application.

weetmuts, 2021-05-22:
We are still working on snap. The guaranteed way to get it working is by building from source.
clone; ./configure; make; sudo make install

Then you can stop any daemon "sudo systemctl stop wmbusmeters" "sudo systemctl disable wmbusmeters"
and run from the command line: wmbusmeters

wajta, 2021-05-26:
So finally I installed new clean RaspianOS Lite and build from source. 
I also changed the usb dongle, instead of im871a I used original Maddalena UBS.

I tried to run following:
wmbusmeters --debug --logfile=/home/pi/my-log.txt auto:t1

Here is the debug statement:
`
(wmbusmeters) version: 1.3.0-15-g193b89b
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary "00FC"
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary ""
(im871a/im170a) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 8 (detect rc1180)
(serial /dev/ttyUSB0) sent "00"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial /dev/ttyUSB0) sent "30"
(serial) EVENT thread interrupted
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect rc1180)
(serial) EVENT thread interrupted
(rc1180) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B19200) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 5 (detect cul)
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary "2970868C10D32AA061F008A02A30E211FD00140042204719DC1061F8281440A022E828281C40E011028432784082401120E2332850400E02"
(cul) probe response ")p<86><8C><10><D3>*<A0>a<F0><08><A0>*0<E2><11><FD><00><14><00>B G<19><DC><10>a<F8>(<14>@<A0>"<E8>((<1C>@<E0><11><02><84>2x@<82>@<11> <E2>3(P@<0E><02>"
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serial /dev/ttyUSB0) sent "560A0D"
(serial) EVENT thread interrupted
(serial) received binary ""
(cul) probe response ""
(serialtty) closed /dev/ttyUSB0 (detect cul)
(serial) EVENT thread interrupted
(cul) are you there? no
(main) ignoring /dev/ttyUSB0, it does not respond as any of the supported wmbus devices.
No wmbus device detected, waiting for a device to be plugged in.
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmetersd"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 21111 exited
(shell) exec (capture output) "/bin/pidof"
(shell) arg "wmbusmeters"
(shell) output: >>>21107 21030
<<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 21112 exited
Notice! Other wmbusmeters (pid 21030) is running and it might hog any wmbus devices.
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
`

BIBOLV, 2021-05-26:
Can you switch back and try again with `im871a`? Judging from logs `original Maddalena USB` is not supported by wmbusmeters.
BTW, edge channel snap was fixed and you can try it if want. It is intended to be used as service, so it will not work as command.

BIBOLV, 2021-05-26:
Did you try C1 mode, as @ursf commented that more of devices in his vicinity is C1?

wajta, 2021-05-27:
Yes, I did, here is the debug:

(wmbusmeters) version: 1.3.0-15-g193b89b
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 6 (detect amb8465)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial /dev/ttyUSB0) sent "FF0A02008077"
(serial) EVENT thread interrupted
(serial) received binary ""
(serial) received binary ""
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(amb8465) are you there? no.
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 6 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F23200005C3A"
(im871a) checkIM871AFrame "A581100833001301F23200005C3A"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 3a5c expected 3a5c
(im871a) received full frame
(device info bytes) "33001301F2320000"
(im871a/im170a) info: im871a other firmware=13 hci=01 uid=000032f2
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(device config bytes) "FF000300B32552231000010001FF070232000100000000"
(im871a/im170a) config: other link_mode=t1 id=00102352 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 00102352 im871a firmware: 13
(main) opening c1
Started auto im871a[00102352] on /dev/ttyUSB0 listening on c1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 8 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00102352
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F23200005C3A"
(im871a) checkIM871AFrame "A581100833001301F23200005C3A"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 3a5c expected 3a5c
(im871a) received full frame
(im871a) device info completed
(im871a) device info: im871a other firmware=13 hci=01 uid=000032f2
(im871a) set config to set link mode 06
(serial /dev/ttyUSB0) sent "A5010306000206300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) EVENT thread interrupted
(serial) closing 5 devices
(serial) event loop stopped!
(serialtty) closed /dev/ttyAMA0 (detect amb8465)
(serialtty) closed /dev/ttyAMA0 (detect im871a)
(serialtty) closed /dev/ttyAMA0 (detect rc1180)
(serialtty) closed /dev/ttyAMA0 (detect cul)
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 im871a
(serialtty) closed /dev/ttyUSB0 (im871a)
(wmbus) deleted im871a
(serial) closing 5 devices
(wmbusmeters) version: 1.3.0-15-g193b89b
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(serial) before config: /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serial) EVENT thread interrupted
(serial) after config:  /dev/ttyAMA0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS|CTS)
(serialtty) opened /dev/ttyAMA0 fd 3 (detect amb8465)
(serial) received binary ""

BIBOLV, 2021-05-27:
Well if your meters are alive then you should have received somethig if you did listen long enough. There is application that comes with im871a, you could try to use that to see if you can hear something afterall. Maddalena USB maybe also has some app that you could try to see if something is being transmited. You could also try to contact company that provided meters and ask them some guidance. I have no other ideas how to proceed.

weetmuts, 2021-09-25:
Are you running on an raspberry pi? Can you add: --donotprobe=/dev/ttyAMA0
to avoid upsetting the raspberry.

wajta, 2021-09-30:
Yes, on raspberry pi 4.
Here is the debug of :
wmbusmeters --debug --logfile=/home/pi/mylog-t1.txt auto:t1 --donotprobe=/dev/ttyAMA0

(wmbusmeters) version: 1.4.1-33-g2c0accb
(config) using device: --donotprobe=/dev/ttyAMA0
(config) number of meters: 0
(main) device /dev/ttyAMA0 not currently used, detect contents...
(amb8465) could not open tty for detection
(serial) EVENT thread interrupted
(main) ignoring /dev/ttyAMA0, it does not respond as any of the supported wmbus devices.
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F23200005C3A"
(im871a) checkIM871AFrame "A581100833001301F23200005C3A"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 3a5c expected 3a5c
(im871a) received full frame
(device info bytes) "33001301F2320000"
(im871a/im170a) info: im871a other firmware=13 hci=01 uid=000032f2
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(device config bytes) "FF000300B32552231000010001FF070232000100000000"
(im871a/im170a) config: other link_mode=t1 id=00102352 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 00102352 im871a firmware: 13
(main) opening t1
Started auto im871a[00102352] on /dev/ttyUSB0 listening on t1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 8 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00102352
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F23200005C3A"
(im871a) checkIM871AFrame "A581100833001301F23200005C3A"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 3a5c expected 3a5c
(im871a) received full frame
(im871a) device info completed
(im871a) device info: im871a other firmware=13 hci=01 uid=000032f2
(im871a) set config to set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
[ALARM SpecifiedDeviceNotFound] the device --donotprobe=/dev/ttyAMA0 is not working
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
[ALARM SpecifiedDeviceNotFound] the device --donotprobe=/dev/ttyAMA0 is not working
[ALARM SpecifiedDeviceNotFound] the device --donotprobe=/dev/ttyAMA0 is not working
[ALARM SpecifiedDeviceNotFound] the device --donotprobe=/dev/ttyAMA0 is not working
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) EVENT thread interrupted
(serial) closing 5 devices
(serial) event loop stopped!
(serialtty) closed /dev/ttyAMA0 (detect amb8465)
(serialtty) closed /dev/ttyAMA0 (detect im871a)
(serialtty) closed /dev/ttyAMA0 (detect rc1180)
(serialtty) closed /dev/ttyAMA0 (detect cul)
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 im871a
(serialtty) closed /dev/ttyUSB0 (im871a)
(wmbus) deleted im871a
(serial) closing 5 devices

weetmuts, 2021-09-30:
Ah, try this instead:

wmbusmeters --debug --logfile=/home/pi/mylog-t1.txt --donotprobe=/dev/ttyAMA0 auto:t1 

I have added an error message when the options are placed after the device. The --donotprobe became
a device, which clearly was not working.

wajta, 2021-09-30:
Aha :), here is output of 
wmbusmeters --debug --logfile=/home/pi/mylog-t1.txt --donotprobe=/dev/ttyAMA0 auto:t1

(wmbusmeters) version: 1.4.1-33-g2c0accb
(config) number of meters: 0
(main) device /dev/ttyUSB0 not currently used, detect contents...
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect amb8465)
(serial) received binary ""
(amb8465) sending 6 bytes attempt 0
(serial /dev/ttyUSB0) sent "FF0A02008077"
(amb8465) sent 6 bytes OK
(serial) EVENT thread interrupted
(amb8465) reading response... 1
(serial) received binary ""
(amb8465) reading response... 2
(serial) received binary ""
(amb8465) reading response... 3
(serial) received binary ""
(amb8465) are you there? no.
(serialtty) closed /dev/ttyUSB0 (detect amb8465)
(serial) EVENT thread interrupted
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B9600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 3 (detect im871a)
(serial) received binary ""
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F23200005C3A"
(im871a) checkIM871AFrame "A581100833001301F23200005C3A"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 3a5c expected 3a5c
(im871a) received full frame
(device info bytes) "33001301F2320000"
(im871a/im170a) info: im871a other firmware=13 hci=01 uid=000032f2
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(serialtty) closed /dev/ttyUSB0 (detect im871a)
(serial) EVENT thread interrupted
(device config bytes) "FF000300B32552231000010001FF070232000100000000"
(im871a/im170a) config: other link_mode=t1 id=00102352 media=00 version=01 c_field=00 auto_rssi=01
(im871a/im170a) are you there? yes 00102352 im871a firmware: 13
(main) opening t1
Started auto im871a[00102352] on /dev/ttyUSB0 listening on t1
(im871a) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B57600) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 4 (im871a)
(im871a) get config
(serial /dev/ttyUSB0) sent "A5010500"
(serial) EVENT thread interrupted
(serial) received binary "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) checkIM871AFrame "A5810617FF000300B32552231000010001FF0702320001000000008CAA"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 6
(im871a) got crc16 aa8c expected aa8c
(im871a) received full frame
(im871a) get config completed
(im871a) got device id 00102352
(main) regular reset of im871a  on /dev/ttyUSB0 will happen every 82800 seconds
(im871a) get device info
(serial /dev/ttyUSB0) sent "A5010F00"
(serial) EVENT thread interrupted
(serial) received binary "A581100833001301F23200005C3A"
(im871a) checkIM871AFrame "A581100833001301F23200005C3A"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 16
(im871a) got crc16 3a5c expected 3a5c
(im871a) received full frame
(im871a) device info completed
(im871a) device info: im871a other firmware=13 hci=01 uid=000032f2
(im871a) set config to set link mode 03
(serial /dev/ttyUSB0) sent "A5010306000203300100"
(serial) EVENT thread interrupted
(serial) received binary "A58104009CF7"
(im871a) checkIM871AFrame "A58104009CF7"
(im871a) has_timestamp=0 has_rssi=0 has_crc16=1
(im871a) endpoint 1
(im871a) msgid 4
(im871a) got crc16 f79c expected f79c
(im871a) received full frame
(im871a) set config completed
(wmbus) no alarm (expected activity) for im871a
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) EVENT thread interrupted
(serial) closing 1 devices
(serial) event loop stopped!
(serial) TIMER thread interrupted
(wmbus) disconnected /dev/ttyUSB0 im871a
(serialtty) closed /dev/ttyUSB0 (im871a)
(wmbus) deleted im871a
(serial) closing 1 devices

weetmuts, 2022-02-26:
Hi! I have improved the speed of detection when you know what dongle you have! Please test test latest wmbusmeters
if you have the time.

wmbusmeters /dev/ttyUSB0:im871a:t1

So, the recommendation now is to use auto (potentially with --donotprobe) to find your dongle, then specify the dongle exactly to speed up startup and avoid unnecessary probing.

Can we close this now!

closed 2022-05-25 by weetmuts