wmbusmeters/wmbusmeters #262: AM550 electricity counter

ncortines, 2021-03-02:
Hi,

Sorry for submitting an issue for asking silly questions...

I noticed some Tauron Amiplus devices are supported:

> Tauron Amiplus (amiplus) (includes vendor apator and echelon)

However in my case the vendor seems to differ. It appears to be Skraemeco AM550.

Would be possible to add support for it?

Which dongle should I get to start hacking?

Thanks!
Juan

BIBOLV, 2021-03-02:
Hi,
You should confirm that there is actually a wmbus module inside of meter as based on [specs ](https://www.iskraemeco.hr/AM550.pdf) it is optional. 
Regarding dongles, there is short mention in documentation, somewhere around the bottom, but basically you can check prices and availability and get what is most convenient for you - im, amb or rtl dongles. If you are looking at rtl then get proper one, not some clone.
Also you will need to get decryption key and then when you have decrypted message, you can post them here and we can take a look and see if it can be supported.

ncortines, 2021-03-02:
> Hi,
> You should confirm that there is actually a wmbus module inside of meter as based on [specs ](https://www.iskraemeco.hr/AM550.pdf) it is optional.
> Regarding dongles, there is short mention in [...]

Thanks for the reply!

I'm actually waiting for the activation and meanwhile I'm looking at these dongles:
RTL-SDR RTL2832U V3 (seems the better option)
S300U v2 SDR RTL 2832 NEW GEN (slightly cheaper)

ncortines, 2021-03-03:
Unfortunately my provider will not be able to enable the wireless m-bus :(

Closing the issue. Thanks :)

closed 2021-03-03 by ncortines