wmbusmeters/wmbusmeters #263: New Techem/Kamstrup heat meter (0x39) unknowm driver

henkiejan1, 2021-03-04:
Hi,

I try also for a long time to get my measurement from the heating that i consume. I know now that it is send with WMBus. But i don't get clear which driver i need to use. Special thing is that the heatmeter is a Techem but with also the brand and sticker of Kamstrup on the calculator. That makes it for me very unclear what it use and how i can read the measurement.

I got the Operating and Installation manual and it say's it's a: ''Ultrasonic vario3 heat meter 3.2.1 (MID)"

The output is:

`Received telegram from: 67270***
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat meter (0xc3)
                   ver: 0x39
                device: rtlwmbus[00000001]
                  rssi: 67 dBm
                driver: unknown!
telegram=||374468500905276739C3A2109F292C9201988E28000001A20C9E8F173E248DFA95A6D50A00000000000034B0874053D56571A148DAABEA8A|+2
`

henkiejan1, 2021-03-04:
Hmm, i don't know if there goes anything ok. I set the thermostat higher but the telegram's are now just the same! All:

`374468500905276739C3A2109F292C9201988E28000001A20C9E8F173E248DFA95A6D50A00000000000034B0874053D56571A148DAABEA8A`

henkiejan1, 2021-03-06:
I found this information in Dutch about the Kampstrup Multical 302 who looks the same as the Techem Ultrasonic Vario3 heat meter 3.2.1 (MID). Also does it like it use the C1 standard.

https://docplayer.nl/128353667-Multical-302-data-sheet-allround-warmte-en-koudemeter-eenvoudig-te-installeren-en-eenvoudig-in-gebruik.html

weetmuts, 2021-03-06:
Thanks, it is a typical Techem meter with a proprietary protocol wrapped inside a wmbus frame. 
It does not seem to be supported by the vario451 driver. Also it is definitely not a multical302 meter.

These values are not the right ones I presume?
{"media":"heat","meter":"vario451","name":"Heat","id":"67270509","total_kwh":13278.333333,"current_kwh":2883.888889,"previous_kwh":10394.444444,"timestamp":"2021-03-06T20:00:12Z"}

I do not have time to work on non-oms-compliant meters, sorry, but feel free to try to decode the telegram yourself. You can look at meter_vario451.cc for how its done for that meter.

closed 2021-06-20 by weetmuts