wmbusmeters/wmbusmeters #264: rssi

rattinca, 2021-03-05:
is there a json variable for the rssi so i can put it in my database with the rest of the info?

weetmuts, 2021-03-06:
Yes, there should be an rssi_dbm field in the json. Which wmbus device are you using?

rattinca, 2021-03-06:
Ah, its without the METER_ in front?

weetmuts, 2021-03-06:
Yes, { "rssi_dbm" : -66 } and METER_RSSI_DBM=-66
 
wmbusmeters --listfields=ei6500
and you will get the json keys supplied.

wmbusmeters --listenvs=ei6500
and you will get the env variables.

weetmuts, 2021-03-06:
But it depends on the device if you actually get an rssi.

closed 2021-04-08 by weetmuts

Marooned-MB, 2022-02-04:
Am I missing something? I'm using `rtlwmbus` device however, I get error when passing it to `--listenvs`. However, `multical21` seems to be working (to be honest, not sure what that `multical21`  is)
```
pi@raspberrypi:~ $ wmbusmeters --listenvs=rtlwmbus
No such meter type ""
pi@raspberrypi:~ $ wmbusmeters --listenvs=multical21
METER_JSON
METER_TYPE
METER_NAME
METER_ID
METER_TOTAL_M3
METER_TARGET_M3
METER_MAX_FLOW_M3H
METER_FLOW_TEMPERATURE_C
METER_EXTERNAL_TEMPERATURE_C
METER_CURRENT_STATUS
METER_TIME_DRY
METER_TIME_REVERSED
METER_TIME_LEAKING
METER_TIME_BURSTING
METER_TIMESTAMP
```
Anyway, no RSSI_DBM here, however:
```
pi@raspberrypi:~ $ wmbusmeters --listfields=rtlwmbus
No such meter type ""
pi@raspberrypi:~ $ wmbusmeters --listfields=multical21
                    id  The meter id number.
                  name  Your name for the meter.
                 media  What does the meter measure?
                 meter  Meter driver.
             timestamp  Timestamp when wmbusmeters received the telegram.
                device  The wmbus device that received the telegram.
              rssi_dbm  The rssi for the received telegram as reported by the device.
              total_m3  The total water consumption recorded by this meter.
             target_m3  The total water consumption recorded at the beginning of this month.
          max_flow_m3h  The maxium flow recorded during previous period.
    flow_temperature_c  The water temperature.
external_temperature_c  The external temperature outside of the meter.
        current_status  Status of meter.
              time_dry  Amount of time the meter has been dry.
         time_reversed  Amount of time the meter has been reversed.
          time_leaking  Amount of time the meter has been leaking.
         time_bursting  Amount of time the meter has been bursting.
```
here we can see it's supported, also from meter file:
```js
{
    "media": "warm water",
    "meter": "mkradio4",
    "name": "Water",
    "id": "123",
    "total_m3": 63.5,
    "target_m3": 56.7,
    "timestamp": "2022-02-04T14:14:58Z",
    "device": "rtlwmbus[00000001]",
    "rssi_dbm": 75 //<---
}
```

So I added shell command to my config:
```
shell=/usr/bin/curl -i -XPOST 'http://localhost:8086/write?db=wmbusmeters' --data-binary 'meters,meter_id='$METER_ID' value='$METER_TOTAL_M3',rssi_dbm='$METER_RSSI_DBM''
```
and no data is being added. When I remove `,rssi_dbm='$METER_RSSI_DBM'` then data starts to appear in db. No error in log file so I just guessed there is no `$METER_RSSI_DBM` but still have no clue how to add it to influxdb.
Thanks

PS, also small offtopic but I've searched every possible corner and did not find the answer. What's the meaning of `target_m3` which is different from `total_m3`? I assume, `total_m3` is the current reading. Thanks