wmbusmeters/wmbusmeters #265: DVParser: DIB 0x00 with or without VIB?

micw, 2021-03-05:
Hello,
I tried to verify my own implementation of wmbus with you tool (along with some other parsers/tools). I figured out that vendors use the "00" DIB (No Data) without any VIBs.
IMO that's against EN 13757-3 but if I use it in this way, it just works. And a commercial mbus analyzing tool I use shows the same results. If I patch wmbusmeters not to expect a VIB after No Data, it has the same results too. 

Is there any evidence that No Data with VIB is used in real world meters?

weetmuts, 2021-03-06:
I have never seen a No Data DIB without VIB. It seems counterintuitive. There must always be a VIB.
I have seen No Data DIB with VIB. I do not remember where though.

I am guessing that NoData DIB+VIB is used when requesting VIB measurements from a meter in a two way communication?

micw, 2021-03-08:
I will search for some examples and add it here. Please leave it open for documenting it. Thank you.

weetmuts, 2021-03-08:
Great!

weetmuts, 2021-03-08:
I would expect 0x2f to be used as a filler and not 0x00.

micw, 2021-03-08:
I'd say, 2F ist more a technical filling byte (e.g. because of required padding for AES). 00 is a logical filling byte, it means "there's a record but it has no data".

micw, 2021-03-11:
It looks like my "occurences" of "no data" had been caused by either faulty meters or  otherwise invalid data. I'm closing this for now. If we find valid "no data" records, let's add it here for documentation.

closed 2021-03-11 by micw