wmbusmeters/wmbusmeters #266: BFW 240 radio heat cost allocator

ursf, 2021-03-06:
Hi, I have access to several BFW 240 radio heat cost allocators, which are fortunately not encrypted. But the manufacturer seems to use some proprietary protocol, so none of the existing drivers was able to catch any values.

I would think, the fhkvdataiii driver could serve as a approach to a new driver. At least I was able to get the correct previous and current consumption after I copied meter_fhkvdataiii.cc and changed lines 139/140 and 161/162 to
```
139: uchar prev_lo = content[5];
140: uchar prev_hi = content[4];
161: uchar curr_lo = content[7];
162: uchar curr_hi = content[6];
```

At this point, my programming skills have reached their limits by far - I do not dare to go any further. Of course, the date-values are ridiculously off target, because fhkvdataiii provides them, while BFW 240 radio doesn't (at least not at the bytes, which are read out for date calculation).  

The meter should be able to provide some more data. I analyzed it a bit and found the following:

- First payload byte after 2F2F seems to be the value of hours since the beginning of the month. I verified this by taking the value of the last telegram of 03/05/21 (0x78 = 120 hours). Exactly one hour later, the value changed to 0x79. But: What's the use of this value?
- Second payload byte is always 0x1F (for all meters) - first I thought, it could be the days of the current month, but it was 0x1F in February also, which had only 28 days. No clue so far.
-  Third and fourth byte are the previous consumption (verified).
-  Fifth and sixth byte are the current consumption (verified)
- The next two bytes are different on any meter and do never change - maybe some kind of checksum?
- The next 27 bytes seem serve as storage for consumption values of previous months - but I have no clue, how it may be encoded. Maybe there is something else there, because 27 bytes seem a bit strange for 12, 18 or 24 values (different tech specs, not exactly clear, how many values are actually stored here).
- The last three bytes of the payload are the current date - stored in a plain "hex as decimal" (or the other way round?) way: dd mm yy

Most of these assumptions and findings come from the telegrams of the following meter (id 00707788, prev. consumption 961, current consumption from 880 to 977). I append telegrams of two more meters (one with zero consumption) for better comparison of even and different values:

```
telegram=||3644D7088877700002087AF6C000002F2F6E1F03C10370152A00000000000000000000000000000204000404000EE2020AC1321D260221|
telegram=||3644D7088877700002087ADBC000002F2F9E1F03C10388152A00000000000000000000000000000204000404000EE2020AC1321D280221| 
telegram=||3644D7088877700002087A8BC000002F2F011F03C1038D152A0000000000000000000000000200040400040E00E20A23C11D238D010321|
telegram=||3644D7088877700002087A62C000002F2F481F03C103B3152A0000000000000000000000000200040400040E00E20A23C11D238D040321|
telegram=||3644D7088877700002087A29C000002F2F751F03C103C4152A0000000000000000000000000200040400040E00E20A23C11D238D050321|
telegram=||3644D7088877700002087A5DC000002F2F781F03C103C5152A0000000000000000000000000200040400040E00E20A23C11D238D060321|+9531 Note: First telegram of the day, first payload-byte is 0x78 = 120 hours, since beginning of the month
telegram=||3644D7088877700002087A71C000002F2F791F03C103C5152A0000000000000000000000000200040400040E00E20A23C11D238D060321|+13144 Note: One hour later, value of the first payload byte changed to 0x79 = 121 hours since the beginning of the month
telegram=||3644D7088877700002087A02C000002F2F8B1F03C103D1152A0000000000000000000000000200040400040E00E20A23C11D238D060321|

telegram=||3644D7087670700002087A9CC000002F2F6E1F000000000B36000000000000000000000000000000000000000000000000000000260221|
telegram=||3644D7087670700002087A27C000002F2F011F000000000B36000000000000000000000000000000000000000000000000000000010321|

telegram=||3644D7084774700002087A80C000002F2F6F1F01440100147000000000000000000000000000000000000000000037009B4410AC260221|
telegram=||3644D7084774700002087AE1C000002F2F481F0144010014700000000000000000000000000000000000000000379B0144AC0100040321|

```
I'm absolutely clueless, where the meter gets the end of the current period from. Somewhere it has to get the 31.12.21 from, which it displays. The consumption of the previous months seems to be shifted around by the end of every month, but I can't read anything meaningful there.

If it's not in common interest, I could live with my very basic driver approach (in fact, what more than current and previous consumption do you really need?) - but perhaps someone wants to develop it a bit further. It would be appreciated!

alalons, 2021-03-07:
3644D7084774700002087A80C000002F2F6F1F01440100147000000000000000000000000000000000000000000037009B4410AC260221
3644D7084774700002087AE1C000002F2F481F0144010014700000000000000000000000000000000000000000379B0144AC0100040321

The last part of the telegram seems to be historic monthly values. the values are being shifted from right to left.
But there is some shifting in byte ordering or some other logic
37009B4410AC          february telegram
00379B0144AC0100  march telegram

ursf, 2021-04-03:
Ok, I might have understood now whats going on with the shifting of the historic monthly values...
 
Telegrams of one meter/three different months (payload part only, blanks added for reading purposes):
```
date of telegram reception--------------------------------------------------------------------------------|
                                                                                                          |
18 historic monthly values (newest to the right, byte-reordering for 2.,4.,6.,...-oldest month)----|      |
                                                                                                   |      |
???------------------------|                                                                       |      |
                           |                                                                       |      |
current consumption---|    |                                                                       |      |
                      |    |                                                                       |      |
prev. cons.---vvvv vvvv vvvv vvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvvv vvvvvv
feb: 2F2F6F1F 0144 0100 1470 000 000 000 000 000 000 000 000 000 000 000 000 000 000 370 09B 441 0AC 260221
mar: 2F2F481F 0144 0100 1470 000 000 000 000 000 000 000 000 000 000 000 000 000 037 9B0 144 AC0 100 040321
apr: 2F2F871F 0144 013C 1470 000 000 000 000 000 000 000 000 000 000 000 000 370 09B 441 0AC 001 13C 030421

```
Each historic monthly value seems to be stored as 1,5 bytes/12 bits, but also in different order every month. New historic values are filled in from the right, earlier values are shifted to the left. There is space for 18 historic values. 

In the example above (february telegram) 0x0AC (=172) was the value at the end of january, while in the march telegram it was shifted 12 bits to the left and stored as 0xAC0. In the april telegram it is shifted 12 bits to the left and reordered once more, stored as 0x0AC again.
You also can see the historic value of december (seen in payload bytes 5 and 6: 0x0144 = 324) being shifted around: 0x441 in the february telegram, 0x144, in the march telegram and 0x441 again in the april telegram.
The newest monthly historic value 0x13C can also be seen as current consumption at byte 7 and 8 of the payload (0x013C = 316). This was verified by reading the displayed value at the meter itself.

alalons, 2021-04-03:
You got it.
What decision made them put the most significant nibble (4bit) in the middle of the 24 bit word remains unknown to me.

now you get enough info to decode properly.

weetmuts, 2021-04-09:
Neat! Looking forward to a pull request! :-) If you can, please add a test case to simulation_t1.txt and tests/test_t1_meters.sh. If not, I will add it for you.

weetmuts, 2022-05-23:
Howdy! Should we try to get this into wmbusmeters? 
If you can write some pseudo code here, then I can turn it into a proper driver.

weetmuts, 2022-09-28:
I have now implemented the driver. Here is an example of the decoded values, that you have figured out!

```
./build/wmbusmeters --format=json --ppjson 3644D7084774700002087AE1C000002F2F481F0144010014700000000000000000000000000000000000000000379B0144AC0100040321 Foo bfw240radio 00707447 NOKEY
{
    "media":"heat cost allocation",
    "meter":"bfw240radio",
    "name":"Foo",
    "id":"00707447",
    "current_hca":256,
    "prev_hca":324,
    "prev_01_hca":256,
    "prev_02_hca":172,
    "prev_03_hca":324,
    "prev_04_hca":155,
    "prev_05_hca":55,
    "prev_06_hca":0,
    "prev_07_hca":0,
    "prev_08_hca":0,
    "prev_09_hca":0,
    "prev_10_hca":0,
    "prev_11_hca":0,
    "prev_12_hca":0,
    "prev_13_hca":0,
    "prev_14_hca":0,
    "prev_15_hca":0,
    "prev_16_hca":0,
    "prev_17_hca":0,
    "prev_18_hca":0,
    "device_date":"2021-03-04",
    "timestamp":"2022-09-28T17:58:24Z"
}
```

BIBOLV, 2023-01-01:
Closing due to inactivity. Driver was implemented. Please reopen issue for driver improvements.

closed 2023-01-01 by BIBOLV