wmbusmeters/wmbusmeters #268: Invalid handling of VIF 7C/FC

micw, 2021-03-08:
Hello,

I think there's an issue with VIF 7C/FC. The spec says, it it followed by a length byte and $length number of ascii chars.

While the length is read and honored by mbusmeters,, the ascii is ignored and processed as the next dib:

```
(wmbus) 0f: 81 dif (8 Bit Integer/Binary Instantaneous value)
(wmbus) 10: 02 dife (subunit=0 tariff=0 storagenr=4)
(wmbus) 11: 7C vif (VIF in following string (length in first byte))
(wmbus) 12: 03
(wmbus) 13: 49 dif (2 digit BCD Instantaneous value storagenr=1)
(wmbus) 14: 55 vif (Mass 10² kg/h)
(wmbus) 15: 23
(wmbus) 16: 00 dif (No data)
```

12: 03 is the length. 13-15 are the VIF Custom-Unit as ASCII. The 00 at 16 is the value of the 8-Bit Integer.

If the VIF is FC, at 16 would follow the next VIFE

weetmuts, 2021-03-08:
Yes, indeed I have not implemented the ASCII VIF. Can you provide a full telegram perhaps?
What meter is this?

micw, 2021-03-11:
I sent you one per E-Mail, could you check it?

weetmuts, 2022-02-26:
Close, fixed here: https://github.com/weetmuts/wmbusmeters/commit/b5f3ccc152c44f1bc2588dcef4dbd43b004e35be

closed 2022-02-26 by weetmuts