wmbusmeters/wmbusmeters #269: Timestamps in wmbusmeters?

pswiatki, 2021-03-10:
It would be good to have timestamps (real world date/time in **wmbusmeters.log**
If not for every line, then at least at some critical points, such as a restart of the interface (in my case **rtl_sdr** / **rtlwmbus**.
It would allow for better tracking of potential issues, especially resets are cyclic.

weetmuts, 2021-03-13:
I added a new setting --logtimestamps=important (never, always)
when using important it will print timestamps in front of warnings, started, lost and reset wmbus devices
and the daily summary of memory usage.

The default from command line is never.
The default in daemon mode is important.

Grep the source for notice_timestamp to see where the timestamp is added for important events.

Please test!

pswiatki, 2021-03-14:
Tested. Thank you so much. This is the end of the old [systemd] service running and the beginning of the new one:

```
(wmbus) telegram from 4113bc80 ignored by all configured meters!
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2021-03-15 00:01:45.000;141;135;4113BC80;0x1944304c80bc1341cc01a23119001326f456bb1055a410f3c136<0A>"
(rtlwmbus) received full frame
(wmbus) skipping already handled telegram.
(serial) stopping manager
(serial) MAIN thread interrupted
(serial) closing 1 devices
(serial) EVENT thread interrupted
(shell) sending SIGTERM to process group 1205
(serial) event loop stopped!
(bgshell) sent sigterm, now waiting for child 1205 to exit.
(serial) TIMER thread interrupted
(bgshell) 1205 terminated due to signal 15
(wmbus) disconnected 0000616 rtlwmbus
(serialcmd) closed /bin/sh pid=1205 fd=6 (rtlwmbus)
(wmbusmeters) shutting down
(wmbus) deleted rtlwmbus
(serial) closing 1 devices
(wmbusmeters) logging started 2021-03-15 01:01:49 using 1.2.0-17-gecc9e7f
[2021-03-15_01:01:49] (wmbusmeters) version: 1.2.0-17-gecc9e7f
[2021-03-15_01:01:49] (config) store meter files in: "/var/log/wmbusmeters/meter_readings"
[2021-03-15_01:01:49] (config) using device: rtlwmbus[0000616] 
[2021-03-15_01:01:49] (config) number of meters: 2
[2021-03-15_01:01:49] (shell) exec (capture output) "/bin/sh"
[2021-03-15_01:01:49] (shell) arg "-c"
[2021-03-15_01:01:49] (shell) arg "rtl_sdr < /dev/null"
[2021-03-15_01:01:49] (shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	filename (a '-' dumps samples to stdout)

<<<
[2021-03-15_01:01:49] (shell) return code 1
[2021-03-15_01:01:49] (shell) 10763 exited
[2021-03-15_01:01:49] (serial) EVENT thread interrupted
[2021-03-15_01:01:49] (main) rtl_srd found in path
[2021-03-15_01:01:49] (shell) exec (capture output) "/bin/sh"
[2021-03-15_01:01:49] (shell) arg "-c"
[2021-03-15_01:01:49] (shell) arg "rtl_wmbus < /dev/null"
[2021-03-15_01:01:49] (shell) output: >>><<<
[2021-03-15_01:01:49] (shell) return code 0 

````

But there is a completely separate issue that I have started seeing - the **usage** from rtl_sdr executable.
Some option must be improper...  Shall I open a new issue for that?

weetmuts, 2021-03-15:
Good that the timestamps work. 

The output you see is printed because --debug mode will log the output from the shell invocations.
That output is part of the detection of a working rtl_sdr, likewise for the rtl_wmbus, however rtl_wmbus is silent.
So all good! No need to open a new issue.

pswiatki, 2021-03-16:
So, does it mean the detection of a working **rtl_sdr** intentionally calls it with some improper parameter to trigger it to produce the "usage" information?
Anyhow, I removed the --debug flag and now have only what I am most interested in. Good time to close this, I think.

closed 2021-03-16 by pswiatki